Koenkk/zigbee2mqtt #10178: Nortek z-stick not working

DJAMIRSAM, 2021-12-14:
What happened

when i press the open in WebUI button i get a 502 Bad Gateway error
and a error on logs
Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'

i also get this error in HA logs
21-11-02 21:33:29 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.4:8099 ssl:default [Connect call failed ('172.30.33.4', 8099)]

its trying to connect to 172.30.33.4 except my IP range is 192.168.1.XXX

What did you expect to happen

what open a ui

How to reproduce it (minimal and precise)

install the plugin

Debug info

Zigbee2MQTT version: 1.18.1-1
Adapter hardware: Norteck z-stick
Adapter firmware version: Dont Know

Please Help

Twit123, 2021-12-14:
Hi,

That sounds like..

1 - Something else is already using the Z-stick (ZHA perhaps) - I presume you tried a reboot.
2 - I presume you are running in a Docker container (or other container) hence the IP beginning with 172?

Simon.

DJAMIRSAM, 2021-12-14:
It still no work 
i tried the ZHA removal, and I don't understand the others

`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[10:45:08] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[10:45:08] INFO: MQTT available, fetching server detail ...
[10:45:09] INFO: MQTT server settings not configured, trying to auto-discovering ...
[10:45:09] INFO: Configuring 'mqtt://core-mosquitto:1883' mqtt server
[10:45:10] INFO: MQTT credentials not configured, trying to auto-discovering ...
[10:45:10] INFO: Configuring'addons' mqtt user
[10:45:10] INFO: Previous config file found, checking backup
[10:45:10] INFO: Creating backup config in '/config/zigbee2mqtt/.configuration.yaml.bk'
[10:45:10] INFO: Adjusting Zigbee2mqtt core yaml config with add-on quirks ...
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[10:45:11] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.1 start
> node index.js
Zigbee2MQTT:error 2021-12-14 10:45:35: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-12-14 10:45:35: Failed to start zigbee
Zigbee2MQTT:error 2021-12-14 10:45:35: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2021-12-14 10:45:35: Exiting...
Zigbee2MQTT:error 2021-12-14 10:45:35: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:97:5)
[10:45:36] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.1 start
> node index.js
Zigbee2MQTT:error 2021-12-14 10:46:00: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-12-14 10:46:00: Failed to start zigbee
Zigbee2MQTT:error 2021-12-14 10:46:00: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2021-12-14 10:46:00: Exiting...
Zigbee2MQTT:error 2021-12-14 10:46:00: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:97:5)
[10:46:01] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.1 start
> node index.js
Zigbee2MQTT:error 2021-12-14 10:46:25: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-12-14 10:46:25: Failed to start zigbee
Zigbee2MQTT:error 2021-12-14 10:46:25: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2021-12-14 10:46:25: Exiting...
Zigbee2MQTT:error 2021-12-14 10:46:25: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:97:5)
[10:46:26] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.1 start
> node index.js
Zigbee2MQTT:error 2021-12-14 10:46:50: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-12-14 10:46:50: Failed to start zigbee
Zigbee2MQTT:error 2021-12-14 10:46:50: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2021-12-14 10:46:50: Exiting...
Zigbee2MQTT:error 2021-12-14 10:46:51: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:97:5)
[10:46:51] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.1 start
> node index.js`

DJAMIRSAM, 2021-12-14:
I thinks its my dongle https://www.amazon.ca/GoControl-CECOMINOD016164-Linear-HUSBZB-1/dp/B01GJ826F8/ref=asc_df_B01GJ826F8/?tag=googleshopc0c-20&linkCode=df0&hvadid=292939046378&hvpos=&hvnetw=g&hvrand=10033120608258962695&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9001490&hvtargid=pla-312424172362&psc=1

Twit123, 2021-12-14:
What port have you put in the configuration for the USB stick? It looks like Z2M can't find it.

Go into supervisor then click system at the top and in the bottom corner of the host box is 3 small dots, If you click that, then hardware, it will list all your hardware, Scroll down untill you find your z-stick (usually ttyACM or ttyUSB), copy what is under that and put that in your configuration.

Mine looks like this (yours wont be the same as we have different sticks.

[image]

I then enter the below in my configuration:

> serial:
  port: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0

Si

Koenkk, 2021-12-14:
This stick is not supported by zigbee2mqtt: https://www.zigbee2mqtt.io/guide/adapters/#recommended

Twit123, 2021-12-14:
Well that would explain it then.

Si

MattWestb, 2021-12-15:
If updating its firmware it shall being possible running in EZSP V8 mode.

vandad51, 2021-12-16:
I've updated my firmware on the HUSBZB-1 stick with walthowd's guide. I'm now at version 6.7.8 using the ezsp adapter setting in Z2M. It works for me, half the time. I'm using hassos inside vbox on a Windows 10 pc. It didn't work before I updated it, so flashing the stick helped. I can get it to pair with an Aqara water leak sensor after a few attempts. Everything reads and gets sent into HA as it should. Only thing is, I have to re-pair it every few days so I set up a link quality automation to alert me when it drops to "None". What logs would you need to see to find out what's happening? I don't get any errors in Z2M, and with herdsman debug on

DJAMIRSAM, 2021-12-16:
> I've updated my firmware on the HUSBZB-1 stick with walthowd's guide. I'm now at version 6.7.8 using the ezsp adapter setting in Z2M. It works for me, half the time. I'm using hassos inside vbox on a Windows 10 pc. It didn't work before I updated it, so flashing the stick helped. I can get it to pair with an Aqara water leak sensor after a few attempts. Everything reads and gets sent into HA as it should. Only thing is, I have to re-pair it every few days so I set up a link quality automation to alert me when it drops to "None". What logs would you need to see to find out what's happening? I don't get any errors in Z2M, and with herdsman debug on. It just all of a sudden stops reporting, all the while sending "Assertion Failed" I think every hour, regardless if the sensor reported.

Thank you,
how do i flash it?

vandad51, 2021-12-16:
Follow this guide https://github.com/walthowd/husbzb-firmware

DJAMIRSAM, 2021-12-17:
Thanks for your help but when i try to use the docker thing i get a "docker: Error response from daemon: error gathering device information while adding custom device "/dev/ttyUSB0": no such file or directory." error and i thought its in the wrong port but i realized that the usb is just off, and when i unplug it and plug it back it back in it goes on then turns back off. What should i do?

DJAMIRSAM, 2021-12-17:
Bruh i just installed docker and how Vurtual box is giving me this
Call to NEMR0InitVMPart2 failed: VERR_NEM_INIT_FAILED (VERR_NEM_VM_CREATE_FAILED).

Result Code: 
E_FAIL (0x80004005)
Component: 
ConsoleWrap
Interface: 
IConsole {872da645-4a9b-1727-bee2-5585105b9eed}

vandad51, 2021-12-17:
> Thanks for your help but when i try to use the docker thing i get a "docker: Error response from daemon: error gathering device information while adding custom device "/dev/ttyUSB0": no such file or [...]

If you're on Windows, you'll need to disable power saving for the two devices under "device management"

vandad51, 2021-12-17:
I installed a fresh Ubuntu OS and used docker on that and it worked perfectly

DJAMIRSAM, 2022-01-05:
Hey thanks for your help and sorry i have been offline for a while
This will sound very dumb but how to i find the port the device is (/dev/ttyUSB#)

DJAMIRSAM, 2022-01-05:
> This stick is not supported by zigbee2mqtt: https://www.zigbee2mqtt.io/guide/adapters/#recommended
I mean its worth a try

Hedda, 2022-01-14:
By the way, strongly recommend that you at least try upgrading HUSBZB-1 to the very latest Silabs EmberZNet 6.7.10.x firmware made available by the community, see here -> https://github.com/grobasoz/zigbee-firmware/issues/17 and also here -> https://github.com/walthowd/husbzb-firmware/issues/36

And of course, also make sure to always use latest "dev" branch (a.k.a. 'edge' version) of Zigbee2MQTT for experimental features.

Note that with grobasoz EmberZNet 6.7.10.0 / EmberZNet 6.7.10.x firmware uses 115200 as its default baud rate for the serial port.

Understand that Silicon Labs EM35x and ETRX35x Zigbee chip hardware used in Nortek HUSBZB-1 have already been end-of-life for a while and should now almost be considered obsolete as Silabs have started that they are deprecated in firmware newer than EmberZNet 6.7.10.x

There may come a few updates bug-fix releases for EmberZNet 6.7.10.x but otherwise it will not get any new features or functions.

PS: Before buying a newer adapter know that even some of the specific models Silicon Labs EFR32 Mighty Gecko Series 1 modules (i.e. EFR32MG1x e.g. EFR32MG12/EFR32MG13/EFR32MG14) based chip which does not feature enough flash storage size to fit firmware built with the newly released EmberZNet 7.0.0.0 Zigbee release have also reached the end of their planned lifecycle and been scheduled for obsolescence and discontinued by the manufacturer (Silicon Labs). So only Silicon Labs EFR32 Mighty Gecko Series 2 (i.e. EFR32MG2x e.g. EFR32MG21/EFR32MG23/EFR32MG24) and some of the Silicon Labs EFR32 Mighty Gecko Series 1 module model variants which feature more flash storage size is be supported by EmberZNet 7.0 Zigbee.

closed 2022-02-22 by github-actions[bot]

Hedda, 2022-03-14:
Since the EZSP support is still experimental recommend trying to keep discussions to that issue-> https://github.com/Koenkk/zigbee-herdsman/issues/319