Koenkk/zigbee2mqtt #10179: TH1123ZB does not report current / energy / power corectly

nslythe, 2021-12-14:
Hi 
I ave a TH1123ZB thermostat from sinope.
All the expose value for electricty consumption are wrong.
current = 0
energy  = null
power = null
voltage = 2425 should be 242.5

Here is an exemple of the state of the device when heating:
{
    "current": 0,
    "local_temperature": 22.56,
    "occupied_heating_setpoint": 26,
    "operation": "heating",
    "pi_heating_demand": 100,
    "running_state": "heat",
    "system_mode": "heat",
    "voltage": 2425,
    "energy": null,
    "keypad_lockout": null,
    "linkquality": 239,
    "power": null
}

MrNickll, 2022-01-08:
What's your z2m version? I had a similar issue but then updated to latest release. I now get current, energy and voltage reported correctly. Still getting keypad lockout issue, no power report and running state isn't reporting right.

closed 2022-02-15 by github-actions[bot]

geniusled, 2024-09-27:
Hello,  bringing this back up as i am having the same problem (and more mentions). please note i only have approximately 6 hours worth of experience with z2m and zigbee so my terminology may not be fully correct

## using z2m 1.40.1-1 | thermostats Sinope TH1123ZB  FW 0.3.9 | Homeassistant

here is the exposed states (after a bit of tinkering):

>{
    "linkquality": 126,
    "local_temperature": 24.37,
    "occupied_heating_setpoint": 7,
    "pi_heating_demand": 0,
    "power": null,
    "running_state": "idle",
    "voltage": 2368,
    "backlight_auto_dim": null,
    "current": null,
    "enable_outdoor_temperature": null,
    "energy": null,
    "keypad_lockout": null,
    "main_cycle_output": null,
    "outdoor_temperature_timeout": null,
    "second_display_mode": null,
    "temperature_display_mode": null,
    "thermostat_occupancy": null,
    "thermostat_outdoor_temperature": null,
    "time_format": null
}

### user story:
1. first paired the thermostats 
2. noticed that i received some error prompts 
>z2m: Failed to configure 'thermostat', attempt 7 (Error: ZCL command 0x500b9140000105a6/1 seMetering.read(["multiplier","divisor"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:347:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:916:26) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:443:29) at Object.readMeteringMultiplierDivisor (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:37:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/sinope.ts:722:13) at Configure.configure (/app/lib/extension/configure.ts:125:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:56:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
3. noticed in Homeassistant, most values are empty or unkown until manually set
4. current, power, energy, voltage always stay at unknown/null  unless some fiddling is done.

### fiddling
1. i noticed that if i select the device in z2m, go in exposes and click refresh button on the "Power
Instantaneous measured power", then the value gets populated in both z2m and HA. it will not updated itself, pressing the same refresh button again will update the values. the value is in Watts and is displayed correctly.

2. if i go on the dev console tab, there are a few things that i noticed.
> reading the following values manually will get the states updated.
haElectricalMeasurement :
- activePower  > Value in Watts, correctly displayed
- rmsCurrent   > Value in A, but the number is 100x too large (3000A instead of 3A)
- rmsVoltage   > Value in Volts, but the number is 10x too large (2394V instead of 239.4V)

### Conclusion
i think that there are some multipler/divider values that are not properly handled, probably some error somewhere stopping these incorrect values to be polled correctly. it seems there are some values that can be read via zigbee for these multiplier/dividers rather then hardcoding them

####notes
since the values are wrong, it does not seem possible to setup reporting to get these values sent periodically.

geniusled, 2024-10-04:
update on my last message.
i updated the thermostats from firmware version 0.3.9 to 0.6.9 using the sinope GT130, they are now able to properly report power/current/voltage

reinux, 2024-12-20:
I wonder, is it necessary to buy a GT130 to do a firmware update? Any chance an update can be enabled through z2m?

geniusled, 2024-12-20:
Currently Sinope only provides updates through the GT130 hub for zigbee
devices.

reinux, 2024-12-26:
I got a GT130, which updated my TH1123ZB-G2 to 0.1.4. This version doesn't seem to work with Z2M at all.

geniusled, 2024-12-26:
> @geniusled <https://github.com/geniusled> You're still on 0.6.9 I'm
> assuming, right?

Yes im on 0.6.9, but my units are not g2.

According to this page, 0.1.4 is the latest fw for g2.

https://github.com/claudegel/sinope-zha/blob/master/firmware%20version.md

When you say it doesnt seem to work at all, what is happening/not working?