Koenkk/zigbee2mqtt #10180: model 'TS011F' manufacturer name '_TZ3000_oiymh3qu' reused for a wall plug

deiger, 2021-12-14:
The same chip used in LoraTap RR400ZB is also used for this plug, labeled SP400ZB: https://www.aliexpress.com/item/1005003021955773.html
Beyond the selected icon, in order for the plug to report status changes, one must set in Reporting:
Endpoint: 1
Cluster: OnOff
Attribute: onOff
Min rep interval: 0

How would you suggest to accommodate that?

jebabi, 2022-01-03:
@deiger Can you please provide the entry of this device from `data/database.db` after pairing it?

deiger, 2022-01-04:
Sure, see here:
```json
{"id":24,"type":"Router","ieeeAddr":"0x60a423fffeaeef61","nwkAddr":58405,"manufId":4098,"manufName":"_TZ3000_oiymh3qu","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_oiymh3qu","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b776ac","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":"0","maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1641284691767}
```

deiger, 2022-02-05:
Not stale.

gargamelonly, 2022-02-25:
Not stale.

Koenkk, 2022-02-25:
- Picture will now show both devices
- Updated the definition; added the whitelabel + reporting
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-02-25 by Koenkk