Koenkk/zigbee2mqtt #10181: Downgrading from Edge to regular version in HA leads to a crash on start

RubenKelevra, 2021-12-14:
## What happened

"Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel" on startup after downgrade.

Increasing the PanID solves the no-start situation.

Switching back to the edge-version also fixes the issue.

## What did you expect to happen

That it's working.

## How to reproduce it (minimal and precise)

Don't know.

## Debug info
Zigbee2MQTT add-on version: 1.18.1-1
Adapter hardware: slaesh's stick
Adapter firmware version: don't know where to find that.

## Log 

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[20:00:28] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[20:00:29] INFO: MQTT available, fetching server detail ...
[20:00:29] INFO: Previous config file found, checking backup
[20:00:29] INFO: Creating backup config in '/share/zigbee2mqtt/.configuration.yaml.bk'
[20:00:29] INFO: Adjusting Zigbee2mqtt core yaml config with add-on quirks ...
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[20:00:31] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.18.1 start /app
> node index.js
Zigbee2MQTT:error 2021-12-14 20:01:50: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-12-14 20:01:50: Failed to start zigbee
Zigbee2MQTT:error 2021-12-14 20:01:50: Exiting...
Zigbee2MQTT:error 2021-12-14 20:01:50: Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel
    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:172:23
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.18.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-12-14T19_01_50_706Z-debug.log
[20:01:51] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.18.1 start /app
> node index.js
```

Adding log: debug does not lead to any additional message about the issue.

Koenkk, 2021-12-14:
1.18.1 is very old, check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

RubenKelevra, 2021-12-15:
Well, I was referring to the Add-On version for HA which contains zigbee2mqtt. It is shown as the latest version in HA.

Anyway, I grabbed the backup I had from 3 days ago and have now the Edge version again.

But still no luck starting it:

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[01:02:20] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[01:02:23] INFO: MQTT available, fetching server detail ...
[01:02:24] INFO: Previous config file found, checking backup
[01:02:24] INFO: Creating backup config in '/share/zigbee2mqtt/.configuration.yaml.bk'
[01:02:24] INFO: Adjusting Zigbee2mqtt core yaml config with add-on quirks ...
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[01:02:27] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.1-dev start
> node index.js
Zigbee2MQTT:error 2021-12-15 01:03:59: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-12-15 01:03:59: Failed to start zigbee
Zigbee2MQTT:error 2021-12-15 01:03:59: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2021-12-15 01:03:59: Exiting...
Zigbee2MQTT:error 2021-12-15 01:04:00: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:97:5)
[01:04:00] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.1-dev start
> node index.js
Zigbee2MQTT:error 2021-12-15 01:05:19: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-12-15 01:05:19: Failed to start zigbee
Zigbee2MQTT:error 2021-12-15 01:05:19: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2021-12-15 01:05:19: Exiting...
Zigbee2MQTT:error 2021-12-15 01:05:19: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:97:5)
[01:05:20] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.1-dev start
> node index.js
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
Zigbee2MQTT:error 2021-12-15 01:05:45: Failed to call 'Frontend' 'stop' (TypeError: Cannot read property 'clients' of null
    at Frontend.stop (/app/lib/extension/frontend.ts:59:39)
    at Controller.callExtensions (/app/lib/controller.ts:310:40)
    at Controller.stop (/app/lib/controller.ts:190:9)
    at stop (/app/index.js:101:5)
    at process.handleQuit (/app/index.js:107:9))
Zigbee2MQTT:error 2021-12-15 01:05:45: Not connected to MQTT server!
Zigbee2MQTT:error 2021-12-15 01:05:45: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
(node:445) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'end' of undefined
    at MQTT.disconnect (/app/lib/mqtt.ts:104:21)
    at Controller.stop (/app/lib/controller.ts:195:9)
    at stop (/app/index.js:101:5)
    at process.handleQuit (/app/index.js:107:9)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:445) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
(node:445) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
s6-svwait: fatal: timed out
```

RubenKelevra, 2021-12-15:
Well a bit story how that started: Many values on my thermostats started showing "N/A" (or something similar and I couldn't turn them on/off.

Not that I needed to (no worries I'm not freezing 😁) but I'm currently testing out a custom component which adds some more intelligence to them.

So I tried to downgrade, which haven't worked, as you've seen.

Now I tried to upgrade the edge version in an effort to make my lights work again... Which failed. Only like 5 devices came back with very very poor Link Quality.

Everything else stayed offline, even after waiting 12 hours nothing came back.

So something is still off.

Then I wrote the ticket.

---

I recovered (mostly) from it by changing the PanID and the channel and re-pair all devices. (Despite some warnings I read) I was using channel 26 which worked fine in the past 3 month for me.

I'm now using 25 🤷‍♂️

Koenkk, 2021-12-15:
I believe you are using the old addon version, since the addon version matches the z2m version. Make sure to use the correct repo (not danielwelch)

RubenKelevra, 2021-12-15:
Oh, looks like I missed a letter there ... yeah I was using the old one.

So I'll try to migrate :)

closed 2021-12-16 by RubenKelevra