Koenkk/zigbee2mqtt #10184: Adapter disconnecting (CC2531)

joermke, 2021-12-14:
## What happened
I‘m running the Docker container on a Raspberry Pi. Since some time the adapter stops working after a while. This applies to both, latest and latest-dev, builds. I could not figure out why this happens. At least there is some clue in the logs (see snippets below). This may be coincidence but it seems to bail after either one of the GU10 bulbs requests an OTA or the router recognizes „linkquality“ being zero. I have tried different adapters (although all of them CC2531-based). Removing the router does not solve the problem. To restart, I need to unplug the adapter before I can start the container again.
 
```

==> /etc/docker/config/zigbee2mqtt/log/2021-12-11.13-35-15/log.txt <==
info  2021-12-11 13:35:46: MQTT publish: topic 'zigbee2mqtt/Router_1', payload '{"led":false,"linkquality":0}'
debug 2021-12-11 13:36:01: Received MQTT message on 'zigbee2mqtt/fhem/state' with data 'online'
debug 2021-12-11 13:36:31: Received MQTT message on 'zigbee2mqtt/fhem/state' with data 'online'
error 2021-12-11 13:36:46: Adapter disconnected, stopping

==> /etc/docker/config/zigbee2mqtt/log/2021-12-12.21-53-22/log.txt <==
debug 2021-12-13 16:37:11: Received MQTT message on 'zigbee2mqtt/fhem/state' with data 'online'
debug 2021-12-13 16:37:28: Received Zigbee message from 'DoorSpot_left', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":554000944,"imageType":20496,"manufacturerCode":4635}' from endpoint 1 with groupID 0
debug 2021-12-13 16:37:28: Device 'DoorSpot_left' requested OTA
error 2021-12-13 16:37:31: Adapter disconnected, stopping

==> /etc/docker/config/zigbee2mqtt/log/2021-12-13.21-24-46/log.txt <==
debug 2021-12-14 18:30:11: Received MQTT message on 'zigbee2mqtt/fhem/state' with data 'online'
debug 2021-12-14 18:30:16: Received Zigbee message from 'DoorSpot_right', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":554000944,"imageType":20496,"manufacturerCode":4635}' from endpoint 1 with groupID 0
debug 2021-12-14 18:30:16: Device 'DoorSpot_right' requested OTA
error 2021-12-14 18:30:18: Adapter disconnected, stopping
```

## What did you expect to happen
The adapter should not disconnect.

## How to reproduce it (minimal and precise)
Restart and wait (time varies).

## Debug info
Zigbee2MQTT version: 1.22.1 (commit #c2b5229)
Adapter hardware: CC2531
Adapter firmware rev: 20201127

joermke, 2022-01-02:
Update: Disregarding the recommendation I flashed the 3.x coordinator firmware. This seems to solve the issue for me, the network is stable. However, I also upgraded to a _Sonoff Zigbee 3.0 USB Dongle Plus_, which seems to work like a charm, too.

closed 2022-01-02 by joermke

philipforget, 2022-01-17:
I'm seeing this behavior as well with the following firmware flashed to a CC2531 based board, here's the firmware version string that's affected:

```
Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"}
```

I just flashed the following onto the same board to see if a newer firmware might solve the issue for anyone looking into this comment in the future. New version string:

```
Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20211115,"transportrev":2},"type":"zStack12"}'
```

For what it's worth, this adapter has been working just fine since March of 2020. 

One more clue here:

The disconnect seems to happen exactly after (same timestamp second) an "Accepting joining not in blacklist device xxx" message. EG:

```
zigbee2mqtt_1    | Zigbee2MQTT:debug 2022-01-15 09:03:07: Saving state to file /app/data/state.json
zigbee2mqtt_1    | Zigbee2MQTT:info  2022-01-15 09:07:51: Accepting joining not in blocklist device '0x----------------'
zigbee2mqtt_1    | Zigbee2MQTT:error 2022-01-15 09:07:52: Adapter disconnected, stopping
zigbee2mqtt_1    | Zigbee2MQTT:debug 2022-01-15 09:07:52: Saving state to file /app/data/state.json
zigbee2mqtt_1    | Zigbee2MQTT:info  2022-01-15 09:07:52: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
--
zigbee2mqtt_1    | Zigbee2MQTT:debug 2022-01-15 22:55:33: Saving state to file /app/data/state.json
zigbee2mqtt_1    | Zigbee2MQTT:info  2022-01-15 23:00:30: Accepting joining not in blocklist device '0x----------------'
zigbee2mqtt_1    | Zigbee2MQTT:error 2022-01-15 23:00:30: Adapter disconnected, stopping
zigbee2mqtt_1    | Zigbee2MQTT:debug 2022-01-15 23:00:30: Saving state to file /app/data/state.json
zigbee2mqtt_1    | Zigbee2MQTT:info  2022-01-15 23:00:30: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
--
zigbee2mqtt_1    | Zigbee2MQTT:debug 2022-01-16 15:42:58: Saving state to file /app/data/state.json
zigbee2mqtt_1    | Zigbee2MQTT:info  2022-01-16 15:43:41: Accepting joining not in blocklist device '0x8----------------
zigbee2mqtt_1    | Zigbee2MQTT:error 2022-01-16 15:43:41: Adapter disconnected, stopping
zigbee2mqtt_1    | Zigbee2MQTT:debug 2022-01-16 15:43:41: Saving state to file /app/data/state.json
zigbee2mqtt_1    | Zigbee2MQTT:info  2022-01-16 15:43:41: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
--
zigbee2mqtt_1    | Zigbee2MQTT:debug 2022-01-16 15:55:37: Saving state to file /app/data/state.json
zigbee2mqtt_1    | Zigbee2MQTT:debug 2022-01-16 16:00:37: Saving state to file /app/data/state.json
zigbee2mqtt_1    | Zigbee2MQTT:error 2022-01-16 16:03:07: Adapter disconnected, stopping
zigbee2mqtt_1    | Zigbee2MQTT:debug 2022-01-16 16:03:07: Saving state to file /app/data/state.json
zigbee2mqtt_1    | Zigbee2MQTT:info  2022-01-16 16:03:07: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
```

But the last time it happened, this wasn't the case. Still there might be something to look into from this.

philipforget, 2022-02-07:
An update for anyone who happens to stumble upon this:

Flashing newer firmware seems to have fixed this for me on my cheap cc2531 usb stick. It's been 3 weeks now and it hasn't disconnected once.

```
Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20211115,"transportrev":2},"type":"zStack12"}'
```