Koenkk/zigbee2mqtt #10185: Danfoss Ally 014G2461 new firmware introduces "Radiator Covered" boolean

davidramiro, 2021-12-14:
## Is your feature request related to a problem? Please describe
Currently the `external_measured_room_sensor` works fairly well, but digging a bit deeper, several users reported that it rather uses an average of the `local_temperature` and the external one instead of the external temp as an absolute value.

Danfoss recently released version 1.18 of the firmware, and judging from their docs, they introduced a boolean called "Radiator Covered" at `0x4016`. They describe it as follows:

"FALSE = Auto Offset Mode for Exposed Radiators
TRUE = Room Sensor Mode (allows Covered Radiators)"

They also updated the external sensor description as follows:

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

lux73, 2021-12-15:
And please add:

- 0x404B Regulation SetPoint Offset
- 0x404C Adaptation run control
- 0x404D Adaptation run status
- 0x404E Adaptation run settings

i bought a few of them to replace my buggy Eurotronics - the Eurotronics overheated extremely

the new Danfoss otherwise needs to much time to heat up from 18.7°C to 21°C - it needs 5-6 Hours...  very disappointed actually :-/

if anybody point me in the right Direction (which Files to edit) i'm willing to implement this share for merge

codingcatgirl, 2022-01-01:
@lux73 this was added in https://github.com/Koenkk/zigbee-herdsman/pull/439 already :)

lux73, 2022-01-02:
that's correct - but there are no entrys in 'zigbee-herdsman-converters'

i had added these missing entry's & used the above PR as "template" - but no luck after z2m restart...

**i'm not sure if a simple restart is enough after editing the converter file?!?** i even had made some additions to 'fromZigbee.js' & 'toZigbee.js' & 'danfoss.js' -but could not seen the additional Topic. When i try to publish it gave me a warning that the converter is not found/available

a few Weeks ago i made some Changes under devices/miboxer.js for testing - these changes are immediatly online after z2m restart

codingcatgirl, 2022-01-03:
It sounds like a good idea to make your own ticket about this, since this one, based on its title, should probably be closed.

davidramiro, 2022-01-03:
Indeed, this got merged in 1.22.2 and I've successfully tested it. I'll close this here :)

closed 2022-01-03 by davidramiro