Koenkk/zigbee2mqtt #10186: Support for US aqara wall switches additional features (operation_mode,action,power_outage_memory)

jusxon, 2021-12-15:
Currently, Aqara US wall switches (WS-USC01, WS-USC02, WS-USC03, WS-USC04) only support switch on/off state.

These switches have the same functionality as the EU version, but what is needed to be fully supported in Z2M? Do I need to provide sniff logs via a CC2531 for the switches while they are connected to my Aqara hub?

The database.db listing for my WS-USC03 looks much different than the EU counterpart;

My WS-USC03 entry;
`
{
   "id":9,
   "type":"Router",
   "ieeeAddr":"0x04cf8cdf3c761f8e",
   "nwkAddr":36512,
   "manufId":4447,
   "manufName":"LUMI",
   "powerSource":"Mains (single phase)",
   "modelId":"lumi.switch.b1naus01",
   "epList":[
      1,
      242
   ],
   "endpoints":{
      "1":{
         "profId":260,
         "epId":1,
         "devId":256,
         "inClusterList":[
            0,
            2,
            3,
            4,
            5,
            6,
            9,
            1794,
            2820
         ],
         "outClusterList":[
            10,
            25
         ],
         "clusters":{
            "genBasic":{
               "attributes":{
                  "modelId":"lumi.switch.b1naus01",
                  "manufacturerName":"LUMI",
                  "powerSource":1,
                  "zclVersion":3,
                  "appVersion":25,
                  "stackVersion":2,
                  "hwVersion":1,
                  "dateCode":"09-06-2019"
               }
            },
            "genOnOff":{
               "attributes":{
                  "onOff":0
               }
            },
            "genScenes":{
               "attributes":{
                  "count":0
               }
            },
            "genIdentify":{
               "attributes":{
                  "identifyTime":0
               }
            },
            "haElectricalMeasurement":{
               "attributes":{
                  "measurementType":8,
                  "activePower":0
               }
            }
         },
         "binds":[
            {
               "cluster":6,
               "type":"endpoint",
               "deviceIeeeAddress":"0x00124b00237dfbd3",
               "endpointID":242
            },
            {
               "cluster":6,
               "type":"endpoint",
               "deviceIeeeAddress":"0x00124b00237dfbd3",
               "endpointID":1
            }
         ],
         "configuredReportings":[
            {
               "cluster":6,
               "attrId":0,
               "minRepIntval":0,
               "maxRepIntval":3600,
               "repChange":0
            }
         ],
         "meta":{
         }
      },
      "242":{
         "profId":41440,
         "epId":242,
         "devId":97,
         "inClusterList":[            
         ],
         "outClusterList":[
            33
         ],
         "clusters":{
         },
         "binds":[
         ],
         "configuredReportings":[
         ],
         "meta":{
         }
      }
   },
   "appVersion":25,
   "stackVersion":2,
   "hwVersion":1,
   "dateCode":"09-06-2019",
   "zclVersion":3,
   "interviewCompleted":true,
   "meta":{
      "configured":1
   },
   "lastSeen":1639203129752
}
`

Compared to a simular EU counterpart QBKG24LM;
`{"id":14,"type":"Router","ieeeAddr":"0x00158d0006d5b605","nwkAddr":48859,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.b2nacn02","epList":[1,2,3,4,5,6,7],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,3,6,16,5,10,1,2],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65281":{"3":30,"5":32,"8":4159,"9":1793,"100":0,"101":0,"149":1.0116310119628906,"152":0,"154":0},"65314":18,"65520":{"type":"Buffer","data":[170,16,5,65,135,1,1,16,0]},"modelId":"lumi.switch.b2nacn02","appVersion":1,"manufacturerName":"LUMI","powerSource":1,"zclVersion":1,"stackVersion":2,"hwVersion":18,"dateCode":"10-22-2020"}},"genOnOff":{"attributes":{"61440":62839552,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[6,16],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"61440":117440557,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":9,"inClusterList":[12],"outClusterList":[12,4],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":83,"inClusterList":[12],"outClusterList":[12],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":0,"inClusterList":[16,18],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":260,"epId":6,"devId":0,"inClusterList":[18,16],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"7":{"profId":260,"epId":7,"devId":0,"inClusterList":[18,16],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":18,"dateCode":"10-22-2020","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1639251571781}`

Noticeably the US switches are missing a lot of data under genBasic attributes. Will the sniffing help get these values?

jusxon, 2021-12-17:
Here is the current devices.js entry that only exposes switching on/off;
```
    {
        zigbeeModel: ['lumi.switch.b1naus01'],
        model: 'WS-USC03',
        vendor: 'Xiaomi',
        description: 'Aqara smart wall switch (neutral, single rocker)',
        extend: extend.switch(),
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint);
        },
    },
```

This is what the devices.js entry for QBKG23LM looks like;
```
    {
        zigbeeModel: ['lumi.switch.b1nacn02'],
        model: 'QBKG23LM',
        vendor: 'Xiaomi',
        description: 'Aqara D1 1 gang smart wall switch (with neutral wire)',
        fromZigbee: [fz.on_off, fz.xiaomi_power, fz.xiaomi_switch_basic, fz.xiaomi_multistate_action],
        toZigbee: [tz.on_off, tz.xiaomi_power, tz.xiaomi_switch_operation_mode_basic],
        endpoint: (device) => {
            return {'system': 1};
        },
        onEvent: preventReset,
        exposes: [
            e.switch(), e.power().withAccess(ea.STATE_GET),
            e.energy(), e.temperature().withAccess(ea.STATE),
            e.voltage().withAccess(ea.STATE), e.action(['single', 'release']),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode'),
        ],
    },
```

In the Aqara app for my WS-USC03, I have access to operation_mode, power, and energy. However the attributes the function xiaomi_switch_operation_mode_basic looks for aren't available in the database.db entry.

Here is the function;
```
xiaomi_operation_mode_basic: {
	cluster: 'genBasic',
	type: ['attributeReport', 'readResponse'],
	convert: (model, msg, publish, options, meta) => {
		const payload = {};
		if (model.meta && !model.meta.multiEndpoint) {
			const mappingMode = {0x12: 'control_relay', 0xFE: 'decoupled'};
			const key = 0xFF22;
			if (msg.data.hasOwnProperty(key)) {
				payload.operation_mode = mappingMode[msg.data[key]];
			}
		} else {
			const mappingButton = {0xFF22: 'left', 0xFF23: 'right'};
			const mappingMode = {0x12: 'control_left_relay', 0x22: 'control_right_relay', 0xFE: 'decoupled'};
			for (const key in mappingButton) {
				if (msg.data.hasOwnProperty(key)) {
					payload['operation_mode_${mappingButton[key]}'] = mappingMode[msg.data[key]];
				}
			}
		}
		return payload;
	},
},
```

closed 2022-01-25 by github-actions[bot]

lcsjandrey, 2022-04-16:
Any updates? I know that this feature is possible now when using the Aqaba hub.

boojew, 2022-07-04:
I just submitted a PR for this for the WS-USC01 https://github.com/Koenkk/zigbee-herdsman-converters/pull/4419

jusxon, 2022-07-17:
I tried implementing this for the WS-USC03 (same as WS-USC01 except with neutral) with the below external converter file and I got operation mode working, but the USC03 does not report the action after turning the operation_mode to decoupled.

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const constants = require('zigbee-herdsman-converters/lib/constants');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const preventReset = async (type, data, device) => {
    if (
        // options.allow_reset ||
        type !== 'message' ||
        data.type !== 'attributeReport' ||
        data.cluster !== 'genBasic' ||
        !data.data[0xfff0] ||
        // eg: [0xaa, 0x10, 0x05, 0x41, 0x87, 0x01, 0x01, 0x10, 0x00]
        !data.data[0xFFF0].slice(0, 5).equals(Buffer.from([0xaa, 0x10, 0x05, 0x41, 0x87]))
    ) {
        return;
    }
    const options = {manufacturerCode: 0x115f};
    const payload = {[0xfff0]: {
        value: [0xaa, 0x10, 0x05, 0x41, 0x47, 0x01, 0x01, 0x10, 0x01],
        type: 0x41,
    }};
    await device.getEndpoint(1).write('genBasic', payload, options);
};

const definition = {
    zigbeeModel: ['lumi.switch.b1naus01'],
    model: 'WS-USC03',
    vendor: 'Xiaomi',
    description: 'Aqara smart wall switch (with neutral, single rocker)',
    fromZigbee: [fz.on_off, fz.xiaomi_multistate_action, fz.aqara_opple],
    toZigbee: [tz.on_off, tz.xiaomi_switch_operation_mode_opple,
        tz.xiaomi_flip_indicator_light, tz.aqara_switch_mode_switch],
    exposes: [e.switch(), e.action(['single', 'double']), e.flip_indicator_light(),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled']).withDescription('Decoupled mode')],
    onEvent: preventReset,
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        // set "event" mode
        await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
    },
};

module.exports = definition;
```

@boojew Any idea why the USC03 version wouldn't be reporting actions once decoupled mode is enabled? Could it be the action names?

lcsjandrey, 2022-08-23:
> I just submitted a PR for this for the WS-USC01 https://github.com/Koenkk/zigbee-herdsman-converters/pull/4419

I tried adapting for the double rock with neutral but no luck.