Koenkk/zigbee2mqtt #10190: Sorting option in the frontend is broken

the-cyberspace, 2021-12-15:
## What happened

The sorting option in the frontend is broken.  See following screenshot:

Begin:
[image]

In this screenshot the name table is sorted alphabetically but i`m having a sensor for a 3dprinter (see screenshot below) and as far as i know numbers appear first when you sort alphabetically. 

End:
[image]

## What did you expect to happen

See info above..

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:  1.22.1-dev commit: d875b29
Adapter hardware: ConBee2/RaspBee2
Adapter firmware version: 0x26720700

tteck, 2021-12-15:
May want to open an issue here
https://github.com/nurikk/zigbee2mqtt-frontend/issues

the-cyberspace, 2021-12-15:
> May want to open an issue here https://github.com/nurikk/zigbee2mqtt-frontend/issues

Done that, i didn`t now that`s the right place for bugs in the frontend. Will close this issue :)

closed 2021-12-15 by the-cyberspace