Koenkk/zigbee2mqtt #10195: Philips Hue white A60 bulb B22 1055lm with Bluetooth (9290024693) - not supported

SkyLightNow, 2021-12-15:
I just bought this Philips Hue bulb (part number 9290024693), but it appears to be unsupported. I tried following the instructions listed on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html, but got a bit stuck.

I'm new to Home Assistant and Zigbee devices, but have several Zigbee bulbs, switches, PIR sensors and various others connected using Zigbee2MQTT. This was my first attempt at adding an unsupported device.

I'm not really sure I understand the instructions for adding unsupported devices, because they are not specific enough. It feels like I'm guessing a lot of it.

1. First I created a new file called "**LWA018.js**" (see below) and dropped this into this directory on my Home Assistant installation: **/root/config/zigbee2mqtt**
2. Edited this file: **/config/zigbee2mqtt/configuration.yaml** and added the following section:

```
external_converters: [
   LWA018.js
]
```

3. Restarted the zigbee2mqtt add-on.
4. The Hue bulb is discovered, but the following error get repeated several times in the zigbee2mqtt log on start-up. 
```
Zigbee2MQTT:warn  2021-12-15 18:42:35: Received message from unsupported device with Zigbee model 'LWA018' and manufacturer name 'Signify Netherlands B.V.'
Zigbee2MQTT:warn  2021-12-15 18:42:35: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```
5. I also noted that the new entry I added for LWA018.js in the **/config/zigbee2mqtt/configuration.yaml** file under "external_converters" gets removed. This is after I saved the change and restarted the Zigbee2MQTT add-on.

## Information about the device + link
The device is called "Philips Hue white 1100" and this is the link to the exact bulb:
https://www.philips-hue.com/en-gb/p/hue-white-1-pack-b22/8719514288270

Bought from Amazon a few days ago:
https://www.amazon.co.uk/gp/product/B099NRM744/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

## data/database.db entry of the device
`{"id":19,"type":"Router","ieeeAddr":"0x001788010b314647","nwkAddr":52150,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LWA018","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64514],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LWA018","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20201012","swBuildId":"1.76.10"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20201012","swBuildId":"1.76.10","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1639591406716,"useImplicitCheckin":true}`

## Contents of my new LWA018.js file
```
const exposes = require('../lib/exposes');
const fz = {...require('../converters/fromZigbee'), legacy: require('../lib/legacy').fromZigbee};
const tz = require('../converters/toZigbee');
const ota = require('../lib/ota');
const reporting = require('../lib/reporting');
const extend = require('../lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const hueExtend = {
    light_onoff_brightness: (options={}) => ({
        ...extend.light_onoff_brightness(options),
        toZigbee: extend.light_onoff_brightness(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp(options),
        toZigbee: extend.light_onoff_brightness_colortemp(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_color: (options={}) => ({
        ...extend.light_onoff_brightness_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_color({supportsHS: true, ...options}).toZigbee
            .concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp_color: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options})
            .toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
};

module.exports = [
    {
        zigbeeModel: ['LWA018'],
        model: '9290024693',
        vendor: 'Philips',
        description: 'Hue white A60 bulb B22 1055lm with Bluetooth',
        meta: {turnsOffAtBrightness1: true},
        extend: extend.light_onoff_brightness(),
        ota: ota.zigbeeOTA,
    }
];
```

danuals, 2022-01-12:
bulb still not supported?

danuals, 2022-01-12:
or did you get it going

Koenkk, 2022-01-12:
added. 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-12 by Koenkk

danuals, 2022-01-12:
thanks @Koenkk  I was just in there searching but you treat me to it

UhtredTheBold, 2022-01-29:
I've had a couple of these bulbs delivered today and although I can pair and control them via MQTT using the docker dev branch, the device isn't being discovered in home assistant. I know auto discover is working because I added a new device just a few days ago.

danuals, 2022-01-29:
> 

Download the lastest dev version of z2m. It’s supported and works.

UhtredTheBold, 2022-01-29:
I had set home assistant to false by mistake! Working perfectly now :)