Koenkk/zigbee2mqtt #10202: Supported Philips Hue Device marked as unsupported

mvdveek, 2021-12-16:
I have a Philips Hue Light Bulbs which is supported https://www.zigbee2mqtt.io/devices/92900244777.html

It is also present in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/philips.js
zigbeeModel: ['LTV001']

But when my controller (CC2652P) connects to it they are marked as Unsupported.

The log gives me the following info:
Warning Received message from unsupported device with Zigbee model 'LTV001' and manufacturer name 'Signify Netherlands B.V.'

I have already tried a restart of zigbee2mqtt and Home Assistant as well as replugging the controller.

Why are they not recognized?

version data:
Zigbee2Mqtt 1.18.1-1
core-2021.12.2
supervisor-2021.12.2
Home Assistant OS 7.0

I am running on a VM on my NAS.

Tonio16, 2021-12-16:
Hello

Your zigbee2mqtt version is not up to date that could be the reason of your issue.

Antoine

Koenkk, 2021-12-16:
If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0 , this device is supported in z2m 1.22.1

closed 2021-12-16 by Koenkk

mvdveek, 2021-12-16:
Z2M is still very new to me. I expected the auto-update to keep it updated.
Apparently I need to remove the plugin en reinstall it for it to be updated.
Working now.

Thx