Koenkk/zigbee2mqtt #10206: SONOFF SNZB-03: battery level shows 4% and is stuck there even with a fresh battery

danielbrunt57, 2021-12-17:
## What happened
Battery level dropped to 4% so I replaced the battery last week even though device was still working.  I've had this device since Sept 20 and it was okay until recently when it showed 4%.  Not sure if or which z2m update broke it.

## What did you expect to happen
Battery level should now be >90% if not 100% but it is stuck at 4%

## How to reproduce it (minimal and precise)
No idea

## Debug info
Zigbee2MQTT version: 1.22.1-1  (HA Add-on)
Coordinator type: zStack3x0
Coordinator revision: 20210120
Frontend version: 0.6.46
Adapter hardware: CC2652R
Adapter firmware version: 20210128

[image]

lux73, 2021-12-17:
my Aqara Devices needs at least 6-12 Hours before its Battery Status updated after replacing the Battery...

danielbrunt57, 2021-12-17:
It's been running with a new battery for weeks now...

ArtJames, 2021-12-17:
Zigbee2MQTT version: 1.22.1-1 Homeassistant
Coordinator type: zStack3x0
Coordinator revision: 20210708
Frontend version: 0.6.46
Adapter hardware: CC2652P
Adapter firmware version: 20210708

Similar problem with Tuya temperature & humidity sensor TS0201

[image]

Removed and repaired several times, but batterylevel never shows up.

danielbrunt57, 2022-01-17:
> It's been running with a new battery for weeks now...

I ended up deleting the device and re-pairing it and battery started updating again.

fabiobaldoibs, 2022-01-27:
I have a similar problem. I can't get the battery level for any of my sonoff devices. If I try to force, I receive this error below:

`Succesfully changed options
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"1jckd-1"}'
Debug Received MQTT message on 'zigbee2mqtt/Caixa Correio/1/set' with data '{"read":{"attributes":["batteryPercentageRemaining"],"cluster":"genPowerCfg","options":{}}}'
Debug Publishing 'set' 'read' to 'Caixa Correio'
Error Publish 'set' 'read' to 'Caixa Correio' failed: 'Error: Read 0x00124b0023584d1b/1 genPowerCfg(["batteryPercentageRemaining"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 8813 - 1 - 152 - 1 - 1 after 10000ms)'
Debug Error: Read 0x00124b0023584d1b/1 genPowerCfg(["batteryPercentageRemaining"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 8813 - 1 - 152 - 1 - 1 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7)
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'read' to 'Caixa Correio' failed: 'Error: Read 0x00124b0023584d1b/1 genPowerCfg([\"batteryPercentageRemaining\"], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 8813 - 1 - 152 - 1 - 1 after 10000ms)'","meta":{"friendly_name":"Caixa Correio"},"type":"zigbee_publish_error"}' `

danielbrunt57, 2022-01-27:
If they are battery powered then they usually sleep to save power and only connect and communicate when they wake up to do something so the publish command falls on deaf ears and fails as you see with Error Publish 'set' 'read' to 'Caixa Correio' failed.

I don't know about the inner workings of Zigbee but with Z-Wave, the controller queues has to command for most battery devices which sleep with earplugs in and only sends it when the device wakes up and is in a communicative state so it's just a matter of waiting for that to happen.  My battery powered Z-Wave blind controllers are different as they are always listening for commands to adjust the blinds.

Have you tried deleting a device and re-pairing it and then wait for normal battery reports to occur?
I recently had to do that with my Sonoff SNZB-03...

fabiobaldoibs, 2022-02-01:
I deleted the devices and I did a re-pairing and now it works

closed 2022-03-04 by danielbrunt57

litecross91, 2023-10-18:
This is also the case with me. It stuck at 7%. I'm going to try the re-pairing.