Koenkk/zigbee2mqtt #10209: Error: network commissioning timed out

Koli96, 2021-12-17:
## What happened
Hi.
I'm trying to run `zigbee2mtqq` with SONOFF Zigbee 3.0 USB Dongle Plus on Raspberry Pi4 with Raspberry Pi Os x86. I'm newbie with zigbee and it's my first contact either with mosqutto and zigbee itself. I have set up `mosquitto` broker and now I'm trying to run `zigbee2mtqq` in docker.
I have my dongle mounted at `/dev/ttyUSB0`.
After running container I'm still getting in logs:
```
 Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:97:5)
```
I tried to change `panId` and `extendedPanId` in `configuration.yaml`, but didn't help.

## Debug info
Zigbee2MQTT version: latest docker image
Adapter hardware: SONOFF Zigbee 3.0 USB Dongle Plus
Adapter firmware version: preflashed from itead.cc

Full logs:
```
Using '/app/data' as data directory
Zigbee2MQTT:debug 2021-12-17 18:58:04: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2021-12-17 18:58:04: Logging to console and directory: '/app/data/log/2021-12-17.18-58-02' filename: log.txt
Zigbee2MQTT:debug 2021-12-17 18:58:04: Removing old log directory '/app/data/log/2021-12-17.18-46-11'
Zigbee2MQTT:info  2021-12-17 18:58:04: Starting Zigbee2MQTT version 1.22.1 (commit #c2b5229)
Zigbee2MQTT:info  2021-12-17 18:58:04: Starting zigbee-herdsman (0.13.176)
Zigbee2MQTT:debug 2021-12-17 18:58:04: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,13,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6452},"serialPort":{"path":"/dev/ttyACM0"}}'
Zigbee2MQTT:error 2021-12-17 18:59:09: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-12-17 18:59:09: Failed to start zigbee
Zigbee2MQTT:error 2021-12-17 18:59:09: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2021-12-17 18:59:09: Exiting...
Zigbee2MQTT:error 2021-12-17 18:59:09: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:97:5)
```

`docker-compose.yaml`:
```
version: '3'
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    restart: unless-stopped
    volumes:
      - ./data:/app/data
      - /run/udev:/run/udev:ro
    ports:
      # Frontend port
      - 8080:8080
    environment:
      - TZ=Europe/Warsaw
    devices:
      # Make sure this matched your adapter location
      - /dev/ttyUSB0:/dev/ttyACM0
```

`configuration.yaml`:
```
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost
  user: XXX
  password: XXX
serial:
  port: /dev/ttyACM0
advanced:
  ext_pan_id:
    - 221
    - 13
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
  pan_id: 6452
  network_key: xxx
  log_level: debug
```

Thank you for your time!

Koli96, 2021-12-18:
Problem solved after reflashing dongle soft.

closed 2021-12-18 by Koli96

tvbkhanh, 2021-12-18:
@Koli96 may I know which firmware did you reflash? I also got them same issue, but reflashing did not help.

Koli96, 2021-12-19:
@tvbkhanh
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

crbn60, 2021-12-30:
I am also seeing this issue. I tried both the latest develop branch firmware and the latest stable. Is there any way to troubleshoot?

tvbkhanh, 2021-12-30:
> I am also seeing this issue. I tried both the latest develop branch firmware and the latest stable. Is there any way to troubleshoot?

@crbn60 in case you are upgrading from and old dongle, change pan_id in zigbee2mqtt config may help. At least this is what happened to my case.

crbn60, 2021-12-30:
I am switching from an old deconz to the new Sonoff @tvbkhanh . Doesn't changing the `pan_id` require repairing everything?

tvbkhanh, 2021-12-30:
@crbn60 changing dongle already require repairing all devices

crbn60, 2021-12-30:
Thanks for the help @tvbkhanh , I had incorrectly read the documentation! It worked, and I am now repairing all my devices…

dhlavaty, 2022-01-07:
@Koli96 I've had exactly same problem (SONOFF Zigbee 3.0 USB Dongle Plus on Raspberry Pi4), running in Docker. I'm new user - runnig **everything** from scratch - no previous zigbee2mqtt experience.

After hours of try-fails, I just unplugged dongle from rPi, and plugged it back. And voila, now it works. 🤦  _...but I have very bad feeling about it..._

lassivv, 2022-01-31:
Have same problems here. Try different things, firmware update with different pc, try webserial.slae.sh there i get this log or this log randomly:

`[10:48:53+0703] > port opened: {"usbProductId":60000,"usbVendorId":4292}
[10:48:53+0703] > trying to reset into bootloader!
[10:48:53+0723] > sending SYNC
[10:48:53+0727] > getting Chip-Id
[10:48:53+0743] > found: 0x3102f000
[10:48:53+0747] > reading IEEE address`

or

`[10:48:53+0703] > port opened: {"usbProductId":60000,"usbVendorId":4292}
[10:48:53+0703] > trying to reset into bootloader!
[10:48:53+0723] > sending SYNC
`

Try to change panid one or two numbers up --> not helped. Try to change channel 11 --> 15 not helped. Try to restart HA many times and try to unplug and plug CC2652BR stick --> not helped.

Hopefully someone can help with this problem now all zigbee devices are useless, because stick not working.

Firmware update i execute with Texas Instrument program. Couple times not get that programming mode working, but try couple times keep "next to chip" button hold and same time click other button fast and release this "next to chip" button. After that some times i get green connected mark next to my chip number: CC2652BR some times that say cannot get connection. When i get connection green, firmware update goes ok if believe that log file. After that i unplug stick and put that back to HA, but ha not working.
`>Reading file: C:/Users/lassi_koti/Desktop/CC2652RB_coordinator_20211217.hex.
>Start flash erase ...
>Erase finished successfully.
>Start flash programming ...
>Programming finished successfully.
>Start flash verify ...
>Skip verification of unassigned page: 21.
>Skip verification of unassigned page: 22.
>Skip verification of unassigned page: 23.
>Skip verification of unassigned page: 24.
>Skip verification of unassigned page: 25.
>Skip verification of unassigned page: 26.
>Skip verification of unassigned page: 27.
>Skip verification of unassigned page: 28.
>Skip verification of unassigned page: 29.
>Skip verification of unassigned page: 30.
>Skip verification of unassigned page: 31.
>Skip verification of unassigned page: 32.
>Skip verification of unassigned page: 33.
>Skip verification of unassigned page: 34.
>Skip verification of unassigned page: 35.
>Skip verification of unassigned page: 36.
>Skip verification of unassigned page: 37.
>Skip verification of unassigned page: 38.
>Skip verification of unassigned page: 39.
>Skip verification of unassigned page: 40.
>Skip verification of unassigned page: 41.
>Skip verification of unassigned page: 42.
>Page: 0 verified OK.
>Page: 1 verified OK.
>Page: 2 verified OK.
>Page: 3 verified OK.
>Page: 4 verified OK.
>Page: 5 verified OK.
>Page: 6 verified OK.
>Page: 7 verified OK.
>Page: 8 verified OK.
>Page: 9 verified OK.
>Page: 10 verified OK.
>Page: 11 verified OK.
>Page: 12 verified OK.
>Page: 13 verified OK.
>Page: 14 verified OK.
>Page: 15 verified OK.
>Page: 16 verified OK.
>Page: 17 verified OK.
>Page: 18 verified OK.
>Page: 19 verified OK.
>Page: 20 verified OK.
>Page: 43 verified OK.
>Verification finished successfully.
>Reset target ...
>Reset of target successful.`

Every single time i get this same error log on zigbee2mqtt. Allready try to install that add-on again and delete old backup files there, nothing helps. Here is log what zigbee2mqqt spits again and again in that log file: 

[image]

btittelbach, 2022-03-11:
I had the same problem and solved it.
- Zigbee2Mqtt-Edge as Homeassistant Plugin.
- 45df7312-zigbee2mqtt-edge
- zigbee2mqtt@1.24.0-dev

In my case, the issue was that zigbee2mqtt never saw the settings I set in the homeassistant configuration and so reverted to default values.

This became evident, when I subscribed to the mqtt broker and listened to `zigbee2mqtt/bridge/info`. The channel, pan_id, etc announced there, had nothing to do with my settings.

editing `share/zigbee2mqtt/configuration.yaml` directly via ssh on homeassistant solved the issue for me.

prashnts, 2022-03-22:
> This became evident, when I subscribed to the mqtt broker and listened to `zigbee2mqtt/bridge/info`. The channel, pan_id, etc announced there, had nothing to do with my settings.
> 
> editing `share/zigbee2mqtt/configuration.yaml` directly via ssh on homeassistant solved the issue for me.

I saw this too, and just before I was going to ssh in to change it, I decided to try something they [mention in the FAQ here](https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-coordinator-failed-to-start-probably-the-panid-is-already-in-use-try-a-different-panid-or-channel).

I plugged the dongle through a USB 3.0 hub (non-powered, TP-Link UH-304 if it matters), and restarted from HA gui. It... worked!

But... I don't understand why, though? It's as if there is some sort of interference (maybe being close to Ethernet)?

btittelbach, 2022-04-06:
I have it connected using a short USB extension cable. Yep, the interference directly next to the RPi USB plugs is actually significant.

th0str, 2022-05-03:
I have the same issue.

th0str, 2022-05-03:
I have the same issue.

It is a new SONOFF ZigBee 3.0 USB Dongle Plus with firmware CC1352P2_CC2652P_launchpad_coordinator_20220219.hex
I am running zigbee2mqtt version 1.25.1 in docker on unRAID 6.9.2

When starting docker I get following:
[image]

btittelbach, 2022-05-07:
@th0str try connecting it via a short USB extension cable or USB hub. Apparently the SONOFF ZigBee 3.0 USB Dongle gets pretty deaf if connected directly to USB plugs.

Would be good to know if that solves the issue for you as well or there are other problems around.

th0str, 2022-05-16:
It seemed to solve my issue while using an USB extension cable. Thanks

heapxor, 2022-05-26:
the zigbee2mqtt and sonoff stick its really nightmare.
I was everything working, i accidentally deleted sonoff from zigbee integration from Devices section ...
 i am not able to make it work anymore, even i reflashed deviced with previous firmware ... 

what is the solution is the stick DEAD?

csirek, 2022-09-19:
Just an update. Sonoff Zigbee 3.0 USB Dongle Plus with CC1352P2_CC2652P_launchpad_coordinator_20220219.hex firmware and Zigbee2MQTT version 1.27.2 on an oDroid XU4.

I got the same problem (Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby) i tried to change the panId or reflashing the firmware without any help and i used a short (10cm long) USB extension cable on the USB2 and the USB3 port too.

Zigbee2MQTT:error 2022-09-19 18:23:27: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
 

But that is forks for me: I change the extension cable to an USB3 hub (without power (Trendnet TU3-ETGH3) and looks like its works! I tried only on the USB2 port of the oDroid and its works!

Zigbee2MQTT:info  2022-09-19 18:26:22: Starting Zigbee2MQTT version 1.27.2 (commit #7dc48fb)
Zigbee2MQTT:info  2022-09-19 18:26:22: Starting zigbee-herdsman (0.14.53)
Zigbee2MQTT:info  2022-09-19 18:26:42: zigbee-herdsman started (reset)
Zigbee2MQTT:info  2022-09-19 18:26:42: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2022-09-19 18:26:43: Currently 0 devices are joined:

I tested it more... and i am confused...

Short 10cm USB extension cable: network error
Trendnet TU3 USB HUB: works
Longer 20cm USB extension cable: works
Another longer 20cm USB extension cable: works
BACK to the first short 10cm USB extension cable: WORKS
plug direct the dongle to the USB port: works (never tried yet, with the case there is no space for do this, now without the case i can plugged it just for the test

So looks like just need to try the cable as mentioned earlier or try to use a hub... don't give up!!!! :)

nicolasvila, 2024-09-19:
Switching from the USB3 to USB2 on the RPI4 seems enough to solve this tricky issue.
I'm copying my troubleshooting steps from this issue: https://github.com/Koenkk/Z-Stack-firmware/issues/518#issuecomment-2362035615

Just in case a few are facing the same issues as mine, maybe this could help someone.
I have not a clear answer of which of the points below solved my Zigbee network and make it back to normal.

    I reinstalled the following coordinators versions of:

    CC2652RB_coordinator_20210120.hex (originally installed on slaesh's stick)
    CC2652RB_coordinator_20221226.hex
    CC2652RB_coordinator_20230507.hex
    CC2652RB_coordinator_20240710.hex

    Network was complaining about existing network and pan_id was already assigned. So I did the following

    start Z2M without antenna (it started) when not being able to communicate with devices, but that's not solving the issue
    disconnected all the zigbee devices (removed batteries)
    the issue was still there so I erased, reflashed my CC2531 routers and still no luck. So I also put them offline too

    Maybe Z2M has a corrupted configuration or something like that

    I reinstalled Z2M from the latest repository. I was stuck with 1.18.1 as I never noticed the repository has moved. The latest version has the same behaviour as the old one...
    After a backup, I erased Z2M configuration from HomeAssistant. Still no luck

    Maybe a flash issue? I tried various options to reflash the 240710 firmware

    using JelmerT's serial bootload tool (cc2538-bsl.py script on Linux)
    I tried also ZigStar Multi Tool on Windows. And this time I totally erased the nvram of the CC2531RB coordinator, and reflashed the latest firmware. The pan_id conflict has gone

    With a working coordinator, the Zigbee network was still down. Unable to pair my devices with my clear coordinator...

    I tried different wiring and check that the USB cable was successfully plugged.
    The coordinator was on an USB3 port of my RaspberryPi4, so I decided to move it to an USB2 port... and TADAAAAAAAA !!! All my Zigbee devices were able to pair with the coordinator again !!!

This has been driving me crazy during 2 weeks (!!!!)
Maybe the issue was caused by USB3 on the RPi4 but why did it work during the past 2 years?
Maybe a subtle mix of different causes?

My advice:

    USB extension cable (20cm) is highly RECOMMENDED
    Use an USB2 slot and avoid USB3 !
    Use full path for the configuration ! /dev/serial/by-id/..... (not /dev/USB1)

Hope it helps.