Koenkk/zigbee2mqtt #10210: Failed installation on raspberry pi zero

Bartras, 2021-12-17:
## What happened

Has an error during installation.
At the npm ci step.

## What did you expect to happen

Successful installation and display of the number of packages.

## How to reproduce it (minimal and precise)

I installed "2021-12-02-raspios-buster-armhf-lite" on my raspberry pi zero.

I followed the recommanded installation procedure:

```
wget https://unofficial-builds.nodejs.org/download/release/v14.16.0/node-v14.16.0-linux-armv6l.tar.xz
tar xvfJ node-v14.16.0-linux-armv6l.tar.xz
sudo cp -R node-v14.16.0-linux-armv6l/* /usr/local
rm -rf node-*
sudo apt-get install -y git make g++ gcc
node --version
npm --version
sudo git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
sudo chown -R pi:pi /opt/zigbee2mqtt
cd /opt/zigbee2mqtt
npm ci
```

I got the following return:

```
npm ERR! code ENOENT
npm ERR! syscall open
npm ERR! path /opt/zigbee2mqtt/package.json
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, open '/opt/zigbee2mqtt/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-12-17T18_23_40_724Z-debug.log
```

Here the [complete log](https://www.toptal.com/developers/hastebin/ladodiqoso.sql)

francisp2, 2021-12-18:
Try this line again: 

sudo git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt

Bartras, 2021-12-18:
I did.
What I found wierd is there is nothing in the folder next...

Here what I have:

```
pi@raspberrypi:~ $ sudo git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
Clonage dans '/opt/zigbee2mqtt'...
pi@raspberrypi:~ $ cd /opt/zigbee2mqtt/
pi@raspberrypi:/opt/zigbee2mqtt $
pi@raspberrypi:/opt/zigbee2mqtt $ ls
pi@raspberrypi:/opt/zigbee2mqtt $
```

Bartras, 2021-12-18:
It is ok, I directly copy the git manually thanks WinSCP.
But now after npm ci, I have the following ERR:

```
npm ERR! cb() never called!

npm ERR! This is an error with npm itself. Please report this error at:
npm ERR!     <https://npm.community>

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-12-18T09_31_40_555Z-debug.log
```

Here the complete log.

pikciu, 2021-12-18:
I have the same issue with `git clone`. Any idea how to fix it?
```
pi@raspberrypi:~ $ git clone https://github.com/Koenkk/zigbee2mqtt.git
Cloning into 'zigbee2mqtt'...
pi@raspberrypi:~ $ git clone git@github.com:Koenkk/zigbee2mqtt.git
Cloning into 'zigbee2mqtt'...
error: index-pack died of signal 11
remote: Enumerating objects: 20114, done.
fatal: index-pack failed
pi@raspberrypi:~ $ git clone git@github.com:Koenkk/zigbee2mqtt.git
Cloning into 'zigbee2mqtt'...
error: index-pack died of signal 11
fatal: index-pack failed
```

flatsiedatsie, 2021-12-20:
You might want to increase the pagefile to 1Gb

closed 2022-01-27 by github-actions[bot]