Koenkk/zigbee2mqtt #10217: Aqara T1 without neutral (SSM-U02) not able to join the network

jd1900, 2021-12-18:
_NOTE: This is the same bug as described here (https://github.com/Koenkk/zigbee2mqtt/issues/7942), as this was closed and I can't reopen._

## What happened
I can't make the Aqara T1 without neutral (SSM-U02 - https://www.zigbee2mqtt.io/devices/SSM-U02.html) join. I first press 8 seconds the device button to be in pairing mode.
It "seems" to be joining, I can even control it for a few seconds, and then it is disconnected. 
According to the logs, the pairing is succesful, however, after a few seconds I can't control it anymore.
The blue light of the device keeps blinking all the time quickly. According to the documentation it means that "Zigbee network is connecting".
When I "Disable join" devices, the light changes from blue blinking to red blinking slowly (Zigbee network is not connected).

I've tried several times, and even with the device a few cm. away from the coordinator. Always the same.

I've tried removing the device, keeping it, reconfiguring...

The best I can get is to control it (turn on or off) right after pairing. 

According to the logs, the device seems to be keeping trying to connect announcing itself all the time.

Manually turning on or off when the device is in red works perfectly. 
When the device is in blue, it works, but after a few seconds it turns off back again.
Same when controlling remotely with Z2M when it allows me (for a few seconds).

I've seen that the DLKZMK12LM has recently been added (https://github.com/Koenkk/zigbee-herdsman-converters/pull/3368) by @DJTerentjev 

However, @julian-ladjani seems to have the same problem as I have with a different coordinator.

## What did you expect to happen
Join properly.

## How to reproduce it (minimal and precise)
Enable joining devices.
Press 8 seconds the device button.

## Logs
```
Zigbee2MQTT:debug 2021-07-04 21:36:12: Device '0x54ef441000076089' announced itself
Zigbee2MQTT:info  2021-07-04 21:36:12: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000076089","ieee_address":"0x54ef441000076089"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-07-04 21:36:12: Configuring '0x54ef441000076089'
Zigbee2MQTT:info  2021-07-04 21:36:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x54ef441000076089"},"type":"device_announced"}'
Zigbee2MQTT:info  2021-07-04 21:36:14: Successfully configured '0x54ef441000076089'

-----

Zigbee2MQTT:debug 2021-07-04 21:36:32: Device '0x54ef441000076089' announced itself
Zigbee2MQTT:info  2021-07-04 21:36:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000076089","ieee_address":"0x54ef441000076089"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-07-04 21:36:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x54ef441000076089"},"type":"device_announced"}'
Zigbee2MQTT:debug 2021-07-04 21:36:43: Received MQTT message on 'zigbee2mqtt/0x54ef441000076089/set' with data '{"state":"OFF"}'
Zigbee2MQTT:debug 2021-07-04 21:36:43: Publishing 'set' 'state' to '0x54ef441000076089'
Zigbee2MQTT:info  2021-07-04 21:36:44: MQTT publish: topic 'zigbee2mqtt/0x54ef441000076089', payload '{"last_seen":"2021-07-04T19:36:43.997Z","linkquality":81,"state":"OFF"}'
Zigbee2MQTT:debug 2021-07-04 21:36:45: Received MQTT message on 'zigbee2mqtt/0x54ef441000076089/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2021-07-04 21:36:45: Publishing 'set' 'state' to '0x54ef441000076089'
Zigbee2MQTT:info  2021-07-04 21:36:45: MQTT publish: topic 'zigbee2mqtt/0x54ef441000076089', payload '{"last_seen":"2021-07-04T19:36:45.484Z","linkquality":81,"state":"ON"}'

-----

Zigbee2MQTT:debug 2021-07-04 21:36:52: Device '0x54ef441000076089' announced itself
Zigbee2MQTT:info  2021-07-04 21:36:52: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000076089","ieee_address":"0x54ef441000076089"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-07-04 21:36:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x54ef441000076089"},"type":"device_announced"}'

-----

Zigbee2MQTT:debug 2021-07-04 21:37:12: Device '0x54ef441000076089' announced itself
Zigbee2MQTT:info  2021-07-04 21:37:12: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000076089","ieee_address":"0x54ef441000076089"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-07-04 21:37:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x54ef441000076089"},"type":"device_announced"}'

-----

Zigbee2MQTT:debug 2021-07-04 21:37:31: Device '0x54ef441000076089' announced itself
Zigbee2MQTT:info  2021-07-04 21:37:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000076089","ieee_address":"0x54ef441000076089"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-07-04 21:37:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x54ef441000076089"},"type":"device_announced"}'

-----

Zigbee2MQTT:debug 2021-07-04 21:37:50: Device '0x54ef441000076089' announced itself
Zigbee2MQTT:info  2021-07-04 21:37:50: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000076089","ieee_address":"0x54ef441000076089"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-07-04 21:37:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x54ef441000076089"},"type":"device_announced"}'

```

## Debug info
Zigbee2MQTT version: 1.22.1
Adapter hardware: zzh
Coordinator type: zStack3x0
Coordinator revision: 20210120
Frontend version: 0.6.46

denisdatsko, 2021-12-24:
Got exactly the same problem, the device is being remove right after the successful pair.
Here is the log from HA integration of xiaomi gateway 3:
`2021-12-24 01:45:27  DEBUG    gateway3      192.168.1.57 | Prepare Gateway
2021-12-24 01:45:27  DEBUG    gateway3      192.168.1.57 | Version: 1.5.0_0026
2021-12-24 01:45:27  DEBUG    gateway3      192.168.1.57 | Patch Bluetooth
2021-12-24 01:45:27  DEBUG    gateway3      192.168.1.57 | MQTT connected
2021-12-24 01:45:27  DEBUG    gateway3      192.168.1.57 | gateway <= True
2021-12-24 01:45:28  DEBUG    gateway3      192.168.1.57 | Gateway time offset: -0.6796672344207764
2021-12-24 01:45:48  DEBUG    gateway3      192.168.1.57 | 463152993 lumi.gateway.mgl03 <= {'pairing_start': 60} [0.02]
2021-12-24 01:45:52  DEBUG    gateway3      192.168.1.57 | gateway <= {'networkUp': True, 'networkPanId': '0x8958', 'radioTxPower': 7, 'radioChannel': 25}
2021-12-24 01:46:03  DEBUG    gateway3      192.168.1.57 | 463152993 lumi.gateway.mgl03 <= {'discovered_mac': '54ef441000075989'} [-0.02]
2021-12-24 01:46:03  DEBUG    gateway3      192.168.1.57 | 463152993 lumi.gateway.mgl03 <= {'8.0.2166': '{}'} [-0.01]
2021-12-24 01:46:03  DEBUG    gateway3      192.168.1.57 | 463152993 lumi.gateway.mgl03 <= {'pair_command': {'code': 0, 'install_code': '38bc2ab4d0ac3a0a31a36b2c86ef22c464c8', 'mac': '54ef441000075989'}} [-0.02]
2021-12-24 01:46:15  DEBUG    gateway3      192.168.1.57 | 463152993 lumi.gateway.mgl03 <= {'discovered_mac': '54ef441000075989'} [-0.02]
2021-12-24 01:46:15  DEBUG    gateway3      192.168.1.57 | 463152993 lumi.gateway.mgl03 <= {'8.0.2166': '{}'} [0.02]
2021-12-24 01:46:16  DEBUG    gateway3      192.168.1.57 | 463152993 lumi.gateway.mgl03 <= {'pair_command': {'code': 0, 'install_code': '38bc2ab4d0ac3a0a31a36b2c86ef22c464c8', 'mac': '54ef441000075989'}} [-0.02]
2021-12-24 01:46:17  DEBUG    gateway3      192.168.1.57 | 463152993 lumi.gateway.mgl03 <= {'added_device': {'did': 'lumi.54ef441000075989', 'mac': '54ef441000075989', 'model': 'lumi.switch.l0agl1', 'version': '22', 'zb_ver': '3.0', 'joined_type': 1}} [?]
2021-12-24 01:46:17  DEBUG    gateway3      192.168.1.57 | Setup zigbee device {'did': 'lumi.54ef441000075989', 'mac': '0x54ef441000075989', 'model': 'lumi.switch.l0agl1', 'version': '22', 'zb_ver': '3.0', 'joined_type': 1, 'type': 'zigbee', 'init': {'added_device': {...}}, 'nwk': '0xffff', 'fw_ver': 0, 'online': True}	
2021-12-24 01:46:17  DEBUG    gateway3      192.168.1.57 | 463152993 lumi.gateway.mgl03 <= {'device_model': 'lumi.switch.l0agl1'} [0.06]
2021-12-24 01:46:17  DEBUG    gateway3      192.168.1.57 | 463152993 lumi.gateway.mgl03 <= {'8.0.2104': {'model': 'lumi.switch.l0agl1', 'pdid': 69002, 'template_id': ''}} [-0.02]
2021-12-24 01:46:18  DEBUG    gateway3      192.168.1.57 | lumi.54ef441000075989 lumi.switch.l0agl1 <= {'8.0.2086': '0000000017b2a506'} [-0.02]
2021-12-24 01:46:18  DEBUG    gateway3      192.168.1.57 | 463152993 lumi.gateway.mgl03 <= {'pairing_start': 0} [-0.01]
2021-12-24 01:46:18  DEBUG    gateway3      192.168.1.57 | lumi.54ef441000075989 lumi.switch.l0agl1 <= {'8.0.2087': '05643CA3361B5CC68E085B11981CC293'} [-0.02]
2021-12-24 01:46:19  DEBUG    gateway3      192.168.1.57 | lumi.54ef441000075989 lumi.switch.l0agl1 <= {'8.0.2088': '5d1d456b708c51f9e8cd03b66d32be83'} [-0.02]
2021-12-24 01:46:19  DEBUG    gateway3      192.168.1.57 | lumi.54ef441000075989 lumi.switch.l0agl1 <= {'8.2': '64100003281c052101009a2010082116010a2100000b2000'} [-0.02]
2021-12-24 01:46:19  DEBUG    gateway3      192.168.1.57 | 463152993 lumi.gateway.mgl03 <= {'removed_did': 'lumi.54ef441000075989'} [-0.02]	
2021-12-24 01:46:19  DEBUG    gateway3      192.168.1.57 | remote.0x0c4314fffe865a98_pair | Handle removed_did: lumi.54ef441000075989s`

jd1900, 2022-01-01:
Seems to be fixed with the latest zzh! coordinator firmware (Z-Stack_Home_3.x.0_20211217) by @Koenkk

closed 2022-02-08 by github-actions[bot]