Koenkk/zigbee2mqtt #10231: segfault after changing configfile

viper539, 2021-12-18:
## What happened
My configfile is currently set to permit_join: true. when is set it to false, restart fails with:
`Dec 18 23:05:03 licht npm[25755]: #033[32mZigbee2MQTT:info #033[39m 2021-12-18 23:05:03: Zigbee: disabling joining new devices.
Dec 18 23:05:03 licht kernel: [987947.714056] node[25755]: segfault at 42 ip b636a891 sp bff4c6a4 error 4 in libnode.so.72[b6236000+15f7000]
Dec 18 23:05:03 licht kernel: [987947.714090] Code: 00 89 16 85 c0 74 16 8d 96 74 01 00 00 39 d0 74 0c 83 ec 0c 50 e8 4f 9b ee ff 83 c4 10 8b 86 58 01 00 00 85 c0 74 0c 83 ec 0c <8b> 10 50 ff 52 04 83 c4 10 8b 83 94 d9 01 00 8d 50 08 83 c0 54 89
Dec 18 23:05:03 licht npm[25754]: Segmentation fault
Dec 18 23:05:03 licht systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=139/n/a
Dec 18 23:05:03 licht systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
`

## What did you expect to happen
Same happens when i add something like frontend: blabla
When i change the friendly_name, it works fine.

## How to reproduce it (minimal and precise)
It seems an individual problem because i didn't found something about this

## Debug info
Zigbee2MQTT version: 1.22.1
zigbee-herdsman: 0.13.176
npm: 8.2.0
node: v12.22.5
OS: Debian 11
Hardware: Linux licht 5.10.0-9-686-pae #1 SMP Debian 5.10.70-1 (2021-09-30) i686 GNU/Linux

Adapter hardware: CC2652RB
Adapter firmware version: {"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}

viper539, 2021-12-20:
It didn't work very well for me.
I added a additional Device and now i'm in a segfault-loop.
I removed all devices an let add them zigbee2MQTT but it didn't help.

Log is showing this:
`Dec 20 14:05:11 licht systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=139/n/a
Dec 20 14:05:11 licht systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
`

viper539, 2021-12-20:
i deleted the database.db and coordinator_backup.json and removed everything from configuration.yaml except the Networkkey and now it's booting.

Logfile:
Dec 20 14:28:29 licht systemd[1]: Started zigbee2mqtt.
Dec 20 14:28:33 licht npm[15955]: > zigbee2mqtt@1.22.1 start
Dec 20 14:28:33 licht npm[15955]: > node index.js
Dec 20 14:28:44 licht npm[15970]: #033[32mZigbee2MQTT:info #033[39m 2021-12-20 14:28:44: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-12-20.14-28-39' filename: log.txt
Dec 20 14:28:44 licht npm[15970]: #033[32mZigbee2MQTT:info #033[39m 2021-12-20 14:28:44: Starting Zigbee2MQTT version 1.22.1 (commit #42569d6c)
Dec 20 14:28:44 licht npm[15970]: #033[32mZigbee2MQTT:info #033[39m 2021-12-20 14:28:44: Starting zigbee-herdsman (0.13.176)
Dec 20 14:28:58 licht npm[15970]: #033[32mZigbee2MQTT:info #033[39m 2021-12-20 14:28:58: zigbee-herdsman started (resumed)
Dec 20 14:28:58 licht npm[15970]: #033[32mZigbee2MQTT:info #033[39m 2021-12-20 14:28:58: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
Dec 20 14:28:58 licht npm[15970]: #033[32mZigbee2MQTT:info #033[39m 2021-12-20 14:28:58: Currently 0 devices are joined:
Dec 20 14:28:58 licht npm[15970]: #033[32mZigbee2MQTT:info #033[39m 2021-12-20 14:28:58: Zigbee: disabling joining new devices.
Dec 20 14:28:58 licht npm[15970]: #033[32mZigbee2MQTT:info #033[39m 2021-12-20 14:28:58: Connecting to MQTT server at mqtt://localhost
Dec 20 14:28:58 licht npm[15970]: #033[32mZigbee2MQTT:info #033[39m 2021-12-20 14:28:58: Connected to MQTT server
Dec 20 14:28:58 licht npm[15970]: #033[32mZigbee2MQTT:info #033[39m 2021-12-20 14:28:58: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Dec 20 14:28:59 licht npm[15970]: #033[33mZigbee2MQTT:warn #033[39m 2021-12-20 14:28:59: Using experimental new availability feature
Dec 20 14:28:59 licht npm[15970]: #033[32mZigbee2MQTT:info #033[39m 2021-12-20 14:28:59: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"42569d6c","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":11,"extendedPanID":"0x00124b002390de5f","panID":6754},"permit_join":false,"version":"1.22.1"}'

viper539, 2021-12-20:
when i change something in the config-file like 'permit_join: true' it fails again

viper539, 2021-12-20:
advanced:
  log_level: info
-> segfault
advanced:
  log_level: debug
-> working
advanced:
  log_level: info
-> segfault

It's crazy

viper539, 2021-12-20:
It's still unstable.
When i delete "state.json" it's running in the most times.

Koenkk, 2021-12-21:
Looks like something is wrong with your node.js installation, try updating to e.g. node 16

viper539, 2022-01-21:
Hi,
i switched to the webinterface, which makes it way easier to manage.
Great piece of code!
Thank you for your time.
This can be closed.

closed 2022-01-21 by viper539