Koenkk/zigbee2mqtt #10233: Philips Hue Switch V2

bachoo786, 2021-12-18:
## What happened
I have got the Philips Hue Switch V2 and have connected it to Zigbee2mqtt and it works fine.

However when I hold the up button I expect the action on mqtt payload to be up_hold. However this is not the case as it produces up_press and then up_hold.

## What did you expect to happen
I expected just the 'up_hold' payload to be produced, as I am pressing the button on the remote for long and not just for one second or a click.

## How to reproduce it (minimal and precise)
Keep holding any button on the remote and it will produce the same issue as described above.

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: 1.21.2

SHxKM, 2021-12-19:
I can confirm this is not the behavior with the `324131092621` dimmer switch, the events in case of a long press is `on-hold` → `on-hold-release`.

Ramalama2, 2022-01-16:
i have exactly the same issue and it drives me crazy.

the problem is, you want that your switches react fast, but thats impossible to differenciate in iobroker for example, between:
you press the on button short and it triggers:
on_press
on_press_release

and if you press the button long, it triggers:
on_press
on_hold
on_hold_release

So i could difference those 2 with "release", but this brings a delay.
Any help appriciated

closed 2022-02-23 by github-actions[bot]