Koenkk/zigbee2mqtt #10236: ROB_200-016-0 Zigbee 3-in1 remote not supported

FlexxFR, 2021-12-19:
Device '0x60a423fffe60f076' with Zigbee model 'ROB_200-016-0' and manufacturer name 'ROBB smarrt' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

after a quick search it appears to be exactly the same as:
https://www.zigbee2mqtt.io/devices/ZG2858A.html

I created a robb.js file based on the ZG2858A and it appears to be working :) :

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ROB_200-016-0'], 
    model: 'ROB_200-016-0', 
    vendor: 'ROBB smart', 
    description: 'RGB CCT DIM 3 in 1 Zigbee Remote', 
    fromZigbee: [fz.battery, fz.command_move_to_color, fz.command_move_to_color_temp, fz.command_move_hue,
            fz.command_step, fz.command_recall, fz.command_on, fz.command_off, fz.command_toggle, fz.command_stop,
            fz.command_move, fz.command_color_loop_set, fz.command_ehanced_move_to_hue_and_saturation], 
    toZigbee: [], 
    exposes: [e.battery(), e.action([
            'color_move', 'color_temperature_move', 'hue_move', 'brightness_step_up', 'brightness_step_down',
            'recall_*', 'on', 'off', 'toggle', 'brightness_stop', 'brightness_move_up', 'brightness_move_down',
            'color_loop_set', 'enhanced_move_to_hue_and_saturation', 'hue_stop'])], 
};

module.exports = definition;
```

## Information about the device + link
[..](https://www.robbshop.nl/robb-smarrt-rgbw-afstandsbediening-zigbee)
Which is just a white label.

## data/database.db entry of the device
..

I'm not sure how to proceed from here?
I believe nothing to do for steps 2&3 and it now to upload a picture & merge.

Is this also true for these kind of white label devices? I assume this will create a lot of repetitiveness.
Is there a way like if model = ROB_200-016-0 "use same as ZG2858A" ?

Koenkk, 2021-12-19:
Added it! 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-19 by Koenkk

mdbraber, 2023-01-19:
I've got this remote as a replacement for a multiEndpoint Sunricher Remote (similar to Namron 4512705) before. With that remote I got a message when switching on different channels (e.g. 1 and 2) for a `recall`. So pressing S1 with channel 1 and 2 enabled yielded `recall_1_l1` and `recall_1_l2`. Is it possible to have this behavior with this remote too? Is zigbee2mqtt filtering those message or are they not being issued by this device? @koenkk @FlexxFR

mdbraber, 2023-01-19:
When viewing the debug log I can see that the information needed is being received, not sure just how this can translate in multiple messages with payloads:

```
Zigbee2MQTT:debug 2023-01-19 18:42:42: Received Zigbee message from 'remote_01', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 1 with groupID 47361
Zigbee2MQTT:info  2023-01-19 18:42:42: MQTT publish: topic 'zigbee2mqtt/remote_01', payload '{"action":"recall_1_l1","action_group":47361,"battery":62.5,"linkquality":119}'
Zigbee2MQTT:info  2023-01-19 18:42:42: MQTT publish: topic 'zigbee2mqtt/remote_01', payload '{"action":"","battery":62.5,"linkquality":119}'
Zigbee2MQTT:info  2023-01-19 18:42:42: MQTT publish: topic 'zigbee2mqtt/remote_01/action', payload 'recall_1_l1'
Zigbee2MQTT:debug 2023-01-19 18:42:42: Received Zigbee message from 'remote_01', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 2 with groupID 47362
```

mdbraber, 2023-01-20:
The solution is to configure the device as `multiEndpoint` and set `publishDuplicateTransaction` to `true`:

```javascript
    {
        zigbeeModel: ['ROB_200-016-0'],
        model: 'ROB_200-016-0',
        vendor: 'ROBB smart',
        description: 'RGB CCT DIM 3 in 1 Zigbee Remote',
        fromZigbee: [fz.battery, fz.command_move_to_color, fz.command_move_to_color_temp, fz.command_move_hue,
            fz.command_step, fz.command_recall, fz.command_on, fz.command_off, fz.command_toggle, fz.command_stop,
            fz.command_move, fz.command_color_loop_set, fz.command_ehanced_move_to_hue_and_saturation],
        toZigbee: [],
        exposes: [e.battery(), e.action([
            'color_move', 'color_temperature_move', 'hue_move', 'brightness_step_up', 'brightness_step_down',
            'recall_*', 'on', 'off', 'toggle', 'brightness_stop', 'brightness_move_up', 'brightness_move_down',
            'color_loop_set', 'enhanced_move_to_hue_and_saturation', 'hue_stop'])],
        meta: {multiEndpoint: true, publishDuplicateTransaction: true},
        endpoint: (device) => {
            return {l1: 1, l2: 2, l3: 3};
        },
    },
```

As it seems logic that people using this device would use it as a multiEndpoint device I'll submit a PR to make this the default.

Kareltje2495, 2024-06-26:
The color wheel of my remote doesn't seem to be working. I have found similar issues with unricher ZG2858A see link below 
"unricher ZG2858A no action on color wheel #22467" [https://github.com/Koenkk/zigbee2mqtt/issues/22467](url)

Is there a possibility this is the same issue and therefore the same fix is needed?

Koenkk, 2024-06-27:
@Kareltje2495 do you see anything logged in the debug log when triggering this action?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Kareltje2495, 2024-06-27:
With all buttons, I see a reaction in the debug log. Only the color wheel is not reacting. I have two of the same remotes, and both are not working. It does not seem to be an issue with the remote itself. The blue LED on the remote also reacts when I turn the color wheel.

Koenkk, 2024-06-28:
Using the frontend -> device -> bind tab, can you bind `LColorCtrl` to coordinator endpoint 1 and see if it works after that? If it fails to bind, make sure to wakeup the device right before clicking the bind button.