Koenkk/zigbee2mqtt #10238: Some dependencies listed as "dev" are required for production

mbronk, 2021-12-19:
This is mostly related to typescript deps which are listed as "devDependencies", but are required for the app to build/run/.

### To reproduce
```
sudo mkdir -m 777 /location
git -c advice.detachedHead=false clone --depth=1 --branch 1.22.1 https://github.com/Koenkk/zigbee2mqtt.git /location
npm ci --production --prefix /location
npm run build --production --prefix /location
```

**My environment:** node v14.18.2 + npm v8.3.0 

### Actual results:

> > zigbee2mqtt@1.22.1 build
> tsc && node index.js writehash
> 
> lib/controller.ts:9:20 - error TS2307: Cannot find module 'assert' or its corresponding type declarations.
>  9 import assert from 'assert';
>                    ~~~~~~~~
> (...)
> 
> Found 182 errors.

### Workaround:
To run the following before `npm run build`
```
npm --prefix /location --production install typescript @types/node @types/debounce @types/finalhandler @types/humanize-duration @types/jest @types/js-yaml @types/object-assign-deep @types/rimraf @types/serve-static @types/ws --save-prod
```

### Expected results
These TS deps should(?) be listed in production dependencies and not devDependencies.
All in all, the build steps in "repro" should pass.

Koenkk, 2021-12-19:
This has been done on purpose, it prevents e.g. the installation of the typescript dependencies in the docker image (https://github.com/Koenkk/zigbee2mqtt/blob/master/docker/Dockerfile)

mbronk, 2021-12-19:
@Koenkk - By the way, is there a reason an old version of `npm-shrinkwrap.json` is commited into the repo?
Per [this](https://docs.npmjs.com/cli/v8/configuring-npm/package-lock-json#lockfileversion), v2 is fully backwards-compatible with v1... and npm5/6 are quire dated anyway. 
Reason I am asking is, each time I pull something, I get quite a lot of changes in npm-shrinkwrap.json and a slightly longer build time (it fetches the missing info). Have you been looking at upgrading npm by any chance? 
Thx!

mbronk, 2021-12-19:
> This has been done on purpose, it prevents e.g. the installation of the typescript dependencies in the docker image (https://github.com/Koenkk/zigbee2mqtt/blob/master/docker/Dockerfile)

I see. Thanks for explanation. It's been a few years since I coded anything "for real" in node, so this may be nonsense, but...
... shouldn't there be a way to both:
1. Keep the docker image clean of unnecessary deps, and
2. Allow to have a local install go through with (--production) all the way?

In other words, is the flow I am running as "repro steps" invalid from your POV?

nurikk, 2021-12-19:
if you're intended to use in production mode consider using zigbee2mqtt npm package

1. `npm install -g zigbee2mqtt`
2. `mkdir ~/.z2m`
3. put your config to `~/.z2m/configuration.yaml`
4. run `zigbee2mqtt`

mbronk, 2021-12-20:
@Koenkk , @nurikk - Thank you both for your replies. 

I guess it indeed makes sense to consider typescript a "dev" dependency, since - after all - I am _building_ from source. 
TBH, I've seen typescript included as both dev and prod deps, but using it as "dev" seems to be more typical indeed, so I am closing this issue as invalid.

----

A few extra thoughts which you may (or may not) want to consider :) 

1. What might be a bit misleading is that `zigbee2mqtt` has a self-build code in `index.js`. I.e., while my repro steps were explicit... someone could just `npm install --only=production` and then just **run the code**... which I would assume will fail with a rather cryptic reason. Perhaps the messaging there could be improved?

2. Maybe there's still a way to include these TS deps somewhere (e.g. as `optional`) so that they can be installed in one go? I.e. I don't really want to do UT/linting etc., but having source version of the package sometimes helps - e.g. if I one wants to run a dev branch build or quickly change a line or two on the spot.

3. The `ZIGBEE2MQTT_DATA` env var is documented [here](https://www.zigbee2mqtt.io/guide/configuration/), but I didn't find a note about fallback to home. Is it documented somewhere other than in code? If not, it might be good to add a note (e.g. in the NPM package description), to document the `~/.z2m` behaviour.

3. I would still appreciate upgrading `npm-shrinkwrap.json::lockfileVersion` to version `2` on master :)

closed 2021-12-20 by mbronk