Koenkk/zigbee2mqtt #10243: Aqara high precision motion sensor not working

Sun10568, 2021-12-19:
as title said.

Aqara high precision motion sensor (lumi.motion.agl04) occupancy is not working

lux73, 2021-12-19:
if you mean **RTCGQ13LM**? -> works with z2m 1.22.0

`info  2021-12-19 16:57:31: praesenz-flur (0x54ef441000023956): RTCGQ13LM - Xiaomi Aqara high precision motion sensor (EndDevice)

info  2021-12-19 17:04:21: MQTT publish: topic 'zigbee2mqtt/praesenz-flur', payload '{"battery":100,"last_seen":1639929861925,"linkquality":123,"motion_sensitivity":"high","occupancy":false,"occupancy_timeout":5,"voltage":3100}'`

Sun10568, 2021-12-19:
Error Failed to configure '0x54ef4410002684bd', attempt 2 (Error: Bind 0x54ef4410002684bd/1 genPowerCfg from '0x00124b00237dfd9a/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7))
Error Publish 'set' 'occupancy_timeout' to '0x54ef4410002684bd' failed: 'Error: Write 0x54ef4410002684bd/1 msOccupancySensing({"pirOToUDelay":3}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 2521 - 1 - 64 - 1030 - 4 after 10000ms)'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'occupancy_timeout' to '0x54ef4410002684bd' failed: 'Error: Write 0x54ef4410002684bd/1 msOccupancySensing({\"pirOToUDelay\":3}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 2521 - 1 - 64 - 1030 - 4 after 10000ms)'","meta":{"friendly_name":"0x54ef4410002684bd"},"type":"zigbee_publish_error"}'
Error Publish 'set' 'motion_sensitivity' to '0x54ef4410002684bd' failed: 'Error: Write 0x54ef4410002684bd/1 aqaraOpple({"268":{"value":2,"type":32}}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 2521 - 1 - 65 - 64704 - 4 after 10000ms)'

Error Publish 'set' 'motion_sensitivity' to '0x54ef4410002684bd' failed: 'Error: Write 0x54ef4410002684bd/1 aqaraOpple({"268":{"value":3,"type":32}}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 2521 - 1 - 66 - 64704 - 4 after 10000ms)'

serje73, 2021-12-21:
Have another problem.
{
    "linkquality": 136,
    "motion_sensitivity": "high",
    "occupancy_timeout": 30,
    "battery": null,
    "occupancy": null
}
Occupancy is null. Battery is null.
Zigbee2mqtt Current version: 1.22.1-1

Thanks

closed 2022-01-28 by github-actions[bot]