Koenkk/zigbee2mqtt #10245: Support for Owon THS317-ET, derivative of already supported OWON THS317

razmoket89, 2021-12-19:
## Information about the device + link
The device is a derivative of already supported device OWON THS317 https://www.zigbee2mqtt.io/devices/THS317.html

I think it only has an external temp probe instead of internal.

Clusters in Z2M : 
[image]

## data/database.db entry of the device
This should be working but couldn't add external converter on my installation, don't know why : 

module.exports = [
    {
        zigbeeModel: ['THS317-ET'],
        model: 'THS317-ET',
        vendor: 'OWON',
        description: 'Temperature and humidity sensor',
        fromZigbee: [fz.temperature, fz.humidity, fz.battery],
        toZigbee: [],
        exposes: [e.battery(), e.temperature(), e.humidity()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(2);
            await reporting.bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg']);
            await reporting.temperature(endpoint);
            await reporting.humidity(endpoint);
            await reporting.batteryVoltage(endpoint);
            await reporting.batteryPercentageRemaining(endpoint);
            device.powerSource = 'Battery';
            device.save();
        },
    },
];

Let me know if you need anything additional.

Koenkk, 2021-12-19:
added it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

jingquek, 2021-12-26:
the THS317-ET has some differences.
There is no humidity sensor and getEndpoint should be 3, not 2.
I modified the code and tested it with my device

       zigbeeModel: ['THS317-ET'],
        model: 'THS317-ET',
        vendor: 'OWON',
        description: 'Temperature sensor',
        fromZigbee: [fz.temperature, fz.battery],
        toZigbee: [],
        exposes: [e.battery(), e.temperature()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(3);
            await reporting.bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement','genPowerCfg']);
            await reporting.temperature(endpoint);
            await reporting.batteryVoltage(endpoint);
            await reporting.batteryPercentageRemaining(endpoint);
            device.powerSource = 'Battery';
            device.save();
        },
    },

Koenkk, 2021-12-26:
@jingquek please make a pr, this can be done by clicking the edit (pencil) icon here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/owon.js

miljbee, 2022-01-03:
I also have this device, I made the PR, I hope you don't mind jingquek

jingquek, 2022-01-04:
> I also have this device, I made the PR, I hope you don't mind jingquek

no problem! It's my first time commenting on github so I wasn't familiar with the processes involved for PR also

Koenkk, 2022-01-04:
Thanks, assuming this can be closed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-04 by Koenkk

86nikko, 2022-04-01:
Bonjour Koenk,
Je possède le capteur OWON THS317-ET, pris en charge sous Zigbee2mqtt cependant je n'arrive pas à réaliser l'appairage.

smoysan, 2022-07-22:
Bonjour,

J'ai le même problème d'appairage sous Zigbee2mqtt (home assistant)

jonathanD91, 2022-11-04:
Hello, I managed to integrate the probes on my zigbee2mqtt but I do not have the temperature and battery which goes back. only the lqi.
I tried several integrations but nothing comes back.
what manipulation to do to get the data?
thank you[image]

jonathanD91, 2022-11-05:
My message error on my Zigbee2mqtt ```Failed to configure '0x3c6a2cfffed2457e', attempt 1 (TypeError: Cannot read property 'bind' of undefined at Object.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:24) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/owon.js:185:29) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:115:37) at Configure.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/configure.ts:55:32) at EventEmitter.emit (events.js:327:22) at EventBus.emitMQTTMessage (/opt/zigbee2mqtt/lib/eventBus.ts:109:22) at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/opt/zigbee2mqtt/lib/extension/frontend.ts:94:27) at WebSocket.emit (events.js:315:20) at Receiver.receiverOnMessage (/opt/zigbee2mqtt/node_modules/ws/lib/websocket.js:1178:20))```

cikas, 2022-11-15:
Hello,
i have the same problem:

Device '0x3c6a2cfffed24662' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x3c6a2cfffed24662","ieee_address":"0x3c6a2cfffed24662"},"type":"device_joined"}'
Info Starting interview of '0x3c6a2cfffed24662'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x3c6a2cfffed24662","ieee_address":"0x3c6a2cfffed24662","status":"started"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x3c6a2cfffed24662"},"type":"device_connected"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x3c6a2cfffed24662"},"type":"pairing"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x3c6a2cfffed24662","ieee_address":"0x3c6a2cfffed24662"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x3c6a2cfffed24662"},"type":"device_announced"}'
Info MQTT publish: topic 'zigbee2mqtt/Control_pool_pump', payload '{"linkquality":57,"state":"OFF"}'
Info Successfully interviewed '0x3c6a2cfffed24662', device has successfully been paired
Warning Device '0x3c6a2cfffed24662' with Zigbee model 'THS317-ET' and manufacturer name 'OWON' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x3c6a2cfffed24662","ieee_address":"0x3c6a2cfffed24662","status":"successful","supported":false},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x3c6a2cfffed24662","supported":false},"type":"pairing"}'

cikas, 2022-11-15:
<html>
<body>

0x3c6a2cfffed24662 | 0x3c6a2cfffed24662 (0x1882) | Unsupported | THS317-ET | N/A |  
-- | -- | -- | -- | -- | --

<br class="Apple-interchange-newline">
</body>
</html>[0x3c6a2cfffed24662](http://10.0.1.70:8123/api/hassio_ingress/y0mroA2pzb2wnHeORYMLq7UanLqaxJhbJNi9fTaPJRU/#/device/0x3c6a2cfffed24662)	0x3c6a2cfffed24662 (0x1882)	Unsupported	[THS317-ET](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#how-to-support-new-devices)	N/A

Dewcal, 2024-01-23:
Have just added this device to Sonoff Zigbee USB Dongle Plus-P and find that it never updates once joined. Is there anything I can do to correct this?

razmoket89, 2024-01-23:
> Have just added this device to Sonoff Zigbee USB Dongle Plus-P and find that it never updates once joined. Is there anything I can do to correct this?

Yeah, he's a bit tricky. Had to include/exclude it a few times until it worked correctly. It doesn't update very often though, maybe once every hour. I think I recall it didn't report temperature immediately after pairing, had to wait for a couple of hours and monitored it for a few days back then.

Dewcal, 2024-01-23:
Thank you for your comments - they mirror what I am seeing even down to the one hour report interval (when it does)  Will try  to exclude / repair yet again before I throw it away!

razmoket89, 2024-01-23:
> Thank you for your comments - they mirror what I am seeing even down to the one hour report interval (when it does) Will try to exclude / repair yet again before I throw it away!

Just checked for today, got an updated temperature every hour except for a 2 hour period without any data sent. I use it to monitor my pool temp so I really don't care it doesn't update every 5 minutes but I guess I can be an issue depending on what you use it for. Feel free to share if you find a more reliable equivalent though, always good to share.

Dewcal, 2024-01-24:
Thanks for update. I am trying to use the device to measure the temperature of a tank of hot water as part of my heating system so would appreciate a more rapid response than hourly.
Yesterday at 19.54 hours, I force removed the device, reset it to factory default and re-paired it to the system without any issues.
It rapidly reported a fairly accurate temperature and then gave an update an hour later at 20.54hrs.
There has not been any update in the last 9 hours even though I have used hot water from the tank and the temperature is about 10C colder than it was.
All in all, this seems fairly useless for my needs.  I have not been able to see any ZB alternatives and may have to use a Wi-Fi version from another supplier.