Koenkk/zigbee2mqtt #10249: Scene activation unresponsive

fredguy, 2021-12-19:
## What happened
When a scene is called through MQTT, Z2M picks up the message from MQTT, but no the scene is not triggered. 
When the scene is recalled from the group tabs from GUI, the scenes visually "triggers" the switch but with no actual effect on the module. 
However, when the switch is triggerred directly on the module - manually - within the scene group, the triggers is effectively having an impact on the module. 

## What did you expect to happen
when the command sent to mqqt with the following payload : 
zigbee2mqtt/4/set{"scene_recall":1}
The scene number 1 from group 4 should be activated
When using the gui group tabs and selecting the choosen scene recall, it should enable the selected scene.

## How to reproduce it (minimal and precise)
Create a group with one device, save a scene, call a scene from the Gui : the Scene is stored. recall the selected scene, and check the actually status of the device vs the reported status of the device. 

## Debug info
Zigbee2MQTT version: 1.22.1
Adapter hardware: ZigStar CC2652P2 
Adapter firmware version: CC1352P2_CC2652P_launchpad_coordinator_20210708

lbt, 2022-01-20:
I seem to be experiencing this problem too.

fredguy, 2022-01-20:
The issue is indeed still there.

closed 2022-02-27 by github-actions[bot]

lbt, 2022-02-27:
Re-open?

smyers119, 2024-03-01:
I am having the same problem, any update?