Koenkk/zigbee2mqtt #10250: TS0601 don't update his state

fedor-smart, 2021-12-19:
## What happened
The Moes Switch 3 gang model ZS-EUB don't update his state. In Z2M its recognized as model Zigbee - "TS0601", Zigbee manufacturer - "_TZE200_tz32mtza" , model - "ZTS-EU_3gang"
When i turning it off or on with by my hands(mechanically) the state don't update. If it was "OFF" and I turn is ON mechanically, the light bulb lights up, but the state does not change. It seems to me that the topic is the state of some other. When I turn on and off from the application  HA, everything is fine, the states change. 

## What did you expect to happen
The state should change every time the switch state changes, but it only changes when controlled via HA.
## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531
This is was the first connection to Z2M, and I turned off and turned on the switch with my hands - mechanically, not from HA.

Failed to configure 'Switch_Kitchen-WC-Bath', attempt 1 (Error: Bind 0x60a423fffe31d495/1 genOnOff from '0x00124b0014d97ebe/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7))
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-14T23:34:30+03:00","linkquality":63,"state_l1":null,"state_l2":null,"state_l3":null}'
info Configuring 'Switch_Kitchen-WC-Bath'
info Successfully configured 'Switch_Kitchen-WC-Bath'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-14T23:34:50+03:00","linkquality":63,"state_l1":null,"state_l2":null,"state_l3":null}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-14T23:35:56+03:00","linkquality":42,"state_l1":null,"state_l2":null,"state_l3":null}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-14T23:35:56+03:00","linkquality":39,"state_l1":null,"state_l2":null,"state_l3":null}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-14T23:35:56+03:00","linkquality":31,"state_l1":null,"state_l2":null,"state_l3":null}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-14T23:35:57+03:00","linkquality":36,"state_l1":null,"state_l2":null,"state_l3":null}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-14T23:35:57+03:00","linkquality":31,"state_l1":null,"state_l2":null,"state_l3":null}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-14T23:35:58+03:00","linkquality":44,"state_l1":null,"state_l2":null,"state_l3":null}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-14T23:35:58+03:00","linkquality":42,"state_l1":null,"state_l2":null,"state_l3":null}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-14T23:35:58+03:00","linkquality":28,"state_l1":null,"state_l2":null,"state_l3":null}'

Koenkk, 2021-12-20:
This probably happens due to the `Failed to configure`, try the wakeup the device and right after that click the reconfigure button in the frontend might help the configure to succeed.

fedor-smart, 2021-12-20:
Did as you said. Unfortunately nothing has changed. After reconfigure I try to turn on and off (mechanically) several times, but the state always "OFF".
Log:
Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Switch_Kitchen-WC-Bath","transaction":"ynf5g-2"}'
info Configuring 'Switch_Kitchen-WC-Bath'
info Successfully configured 'Switch_Kitchen-WC-Bath'
info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Switch_Kitchen-WC-Bath"},"status":"ok","transaction":"ynf5g-2"}'
debug Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Switch_Kitchen-WC-Bath","transaction":"ynf5g-3"}'
info Configuring 'Switch_Kitchen-WC-Bath'
info Successfully configured 'Switch_Kitchen-WC-Bath'
info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Switch_Kitchen-WC-Bath"},"status":"ok","transaction":"ynf5g-3"}'
debug Saving state to file /config/zigbee2mqtt/state.json
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:44+03:00","linkquality":55,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:44+03:00","linkquality":49,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:44+03:00","linkquality":47,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:44+03:00","linkquality":47,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:44+03:00","linkquality":44,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:45+03:00","linkquality":39,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:45+03:00","linkquality":36,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:45+03:00","linkquality":26,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:45+03:00","linkquality":21,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:45+03:00","linkquality":21,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:46+03:00","linkquality":65,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:46+03:00","linkquality":65,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:47+03:00","linkquality":65,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:47+03:00","linkquality":65,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:47+03:00","linkquality":68,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:47+03:00","linkquality":65,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:47+03:00","linkquality":68,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:48+03:00","linkquality":70,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:48+03:00","linkquality":76,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:48+03:00","linkquality":68,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'

Koenkk, 2021-12-20:
Is anything logged in the debug log right after controlling the switch physically?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

fedor-smart, 2021-12-20:
In the previous message there was a log with the enabled debug. This log I copied from a file (the same):
debug 2021-12-20 22:31:12: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Switch_Kitchen-WC-Bath","transaction":"ynf5g-2"}'
info  2021-12-20 22:31:12: Configuring 'Switch_Kitchen-WC-Bath'
info  2021-12-20 22:31:12: Successfully configured 'Switch_Kitchen-WC-Bath'
info  2021-12-20 22:31:12: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Switch_Kitchen-WC-Bath"},"status":"ok","transaction":"ynf5g-2"}'
debug 2021-12-20 22:31:22: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Switch_Kitchen-WC-Bath","transaction":"ynf5g-3"}'
info  2021-12-20 22:31:22: Configuring 'Switch_Kitchen-WC-Bath'
info  2021-12-20 22:31:22: Successfully configured 'Switch_Kitchen-WC-Bath'
info  2021-12-20 22:31:22: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Switch_Kitchen-WC-Bath"},"status":"ok","transaction":"ynf5g-3"}'
debug 2021-12-20 22:31:42: Saving state to file /config/zigbee2mqtt/state.json
info  2021-12-20 22:31:44: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:44+03:00","linkquality":55,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:44: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:44+03:00","linkquality":49,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:44: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:44+03:00","linkquality":47,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:44: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:44+03:00","linkquality":47,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:44: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:44+03:00","linkquality":44,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:45: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:45+03:00","linkquality":39,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:45: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:45+03:00","linkquality":36,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:45: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:45+03:00","linkquality":26,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:45: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:45+03:00","linkquality":21,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:45: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:45+03:00","linkquality":21,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:46: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:46+03:00","linkquality":65,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:46: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:46+03:00","linkquality":65,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:47: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:47+03:00","linkquality":65,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:47: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:47+03:00","linkquality":65,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:47: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:47+03:00","linkquality":68,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:47: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:47+03:00","linkquality":65,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:47: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:47+03:00","linkquality":68,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:48: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:48+03:00","linkquality":70,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:48: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:48+03:00","linkquality":76,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info  2021-12-20 22:31:48: MQTT publish: topic 'zigbee2mqtt/Switch_Kitchen-WC-Bath', payload '{"last_seen":"2021-12-20T22:31:48+03:00","linkquality":68,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'

Koenkk, 2021-12-21:
@oleynikandrey since you also have this device, does this work for you?

fedor-smart, 2021-12-21:
No.This is my new device. This device work fine only in ZHA. In Z2M and Z2M_edge i have a problem that described above

oleynikandrey, 2021-12-22:
> @oleynikandrey since you also have this device, does this work for you?

I have 2 such devices and both work correctly.

fedor-smart, 2021-12-22:
> > @oleynikandrey since you also have this device, does this work for you?
> 
> I have 2 such devices and both work correctly.

OK! Tell me please, what add-on do you use? Z2M or Z2M_edge? What version? Maybe I'm doing something wrong with my crooked hands?

oleynikandrey, 2021-12-22:
I'm using Zigbee2MQTT version 1.22.1 commit: c2b5229f

michau83, 2021-12-25:
i have model Zigbee - "TS0601", Zigbee manufacturer - "_TZE200_g1ib5ldv" , model - "ZTS-EU_2gang". the problem i have is their availability become offline if i turn both switches on. the consequence is could not turn on and off with home assistant any more as it show offline. if i turn off manually by physical 1 of these switches then it will be showed as online and can be control via HA. the model i used is without neutral line and capacitor as well. any ideas how to fix?

fedor-smart, 2021-12-25:
I reinstalled the Z2M addon and deleted addon Z2M_edge. Current version is 1.18.1-1. And that's what I saw in the logs :

warning Device '0x5c0272fffe1dad57' with Zigbee model 'TS0601' and manufacturer name '_TZE200_tz32mtza' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_tz32mtza'
warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_tz32mtza'
warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Koenkk, 2021-12-25:
@fedor-smart 1.18.1 is very old. 
If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

fedor-smart, 2021-12-27:
> @fedor-smart 1.18.1 is very old. If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

I had a really problem with the update. After update all works fine. Sorry for wasting your time.

fedor-smart, 2021-12-27:
After update all works fine.

closed 2021-12-27 by fedor-smart