Koenkk/zigbee2mqtt #10251: After adding know device, it doesn't work (Schneider Electric WDE002334)

henrikrox, 2021-12-19:
## What happened
This is a follow up of #9005, after adding this Schneider dimmer, it doesn't turn off/on or dim in zigbee2mqtt. I have tried editing bindings, but I doesn't work.

## What did you expect to happen
The device to work.

## How to reproduce it (minimal and precise)
Get a hold of this device, https://www.zigbee2mqtt.io/devices/WDE002334.html add it to the network, try to use it, but it doesn't work.
[image]
[image]

## Debug info
Zigbee2MQTT version: 1.22.1
Adapter hardware: Conbee II
Adapter firmware version: August 2021

henrikrox, 2021-12-19:
Its working fine in ZHA and deconz.

Looking at the code on another platform it is set up like this
```
    "endpoints": {
      "3": {
        "clusters": [
          0,
          3,
          4,
          5,
          6,
          8,
          769
        ],
        "bindings": [
          6,
          8,
          769
        ]
```

I tried asking for help on discord

rindlerblabla, 2021-12-19:
[image]
This is what the binding settings looks like for me. Got two of those dimmers and they work just fine.

henrikrox, 2021-12-26:
> [image]
> This is what the reporting settings looks like for me. Got two of those dimmers and they work just fine.

2nd guy with this issue now. I Get that it works for someone but People Are also having issues. I cant Even set min/max values. I also dont see the reporting tab being important, when the light dosent Report to simple inn/off commands

henrikrox, 2021-12-27:
@rindlerblabla  @Koenkk 
[image]
[image]

I would love some help here, trough mqtt i can control the dimmer, and turn it off/on but the actually light doesnt turn off/on. How can i fix this? Do you have another firmware @rindlerblabla ? I have tried everything but even though i can turn it onn/off trough software the actually light doesnt respond :(

henrikrox, 2021-12-27:
If i try to make any changes in the exposes setting i get this
[image]

henrikrox, 2021-12-27:
Here is the payload
MQTT publish: topic 'zigbee2mqtt/Downlights Kjøkken', payload '{"ballast_maximum_level":null,"ballast_minimum_level":null,"linkquality":255,"state":"OFF"}'

I see when i physically adjust the dimmer the value gets updated in zigbee2mqtt. But the issue is that the light does not turn off/on

rindlerblabla, 2021-12-27:
If I remember right (am not at home) I cannot change ballast configuration from the UI but with a mqtt message it is working just fine. But as I wrote dimming/turning  on/off the device physically works just fine as well as when doing the same actions from the UI/HA. According to what is exposed to HA I'm on firmware: 001.000.001 V.

henrikrox, 2021-12-29:
@rindlerblabla i dont know why, but today the device work, i havnt done a single thing, so weird. Looks like it needs to think for a while. I change change the ballast now also from HA, and it works fine, go figure :D

henrikrox, 2021-12-29:
closing this with "glitch in the matrix, and 4 days lost due to frustration"

closed 2021-12-29 by henrikrox

HBir, 2022-11-07:
Bumping this closed topic, as it seems to still be an issue.

I'm having the same issue with my newly connected WDE002388, which is detected as WDE002334. It seems as the default bindings are trying to change it at the 21 endpoint, while it is the 3 endpoint that's working.

If I add endpoint 3 to a group and control that, it will work as expected.

Is there any straight forward way to change the default endpoint to be 3 rather than 21?

sopparus, 2022-11-14:
in the same boat as well, same problem on both my WDE002334 rotary dimmers

mandrean, 2022-12-09:
I ran into the same issue

What solved it for me (almost instantly) was restarting `zigbee2mqtt`:

```
$ sudo systemctl restart zigbee2mqtt.service
```