Koenkk/zigbee2mqtt #10253: Allow Debouncing on Leading Edge

scolby33, 2021-12-20:
## Is your feature request related to a problem? Please describe
If I haven't used it in a moderate period of time (I assume after a sleep threshold has elapsed), my Tuya TS0044 sends its first button press after wakeup twice:
```
Zigbee2MQTT:debug 2021-11-29 22:50:26: Received Zigbee message from 'button', type 'raw', cluster 'genOnOff', data '{"data":[1,89,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-11-29 22:50:26: MQTT publish: topic 'zigbee2mqtt/button', payload '{"action":"1_single","battery":null,"linkquality":87}'
Zigbee2MQTT:info  2021-11-29 22:50:26: MQTT publish: topic 'zigbee2mqtt/button/action', payload '1_single'
Zigbee2MQTT:debug 2021-11-29 22:50:27: Device 'button' announced itself
Zigbee2MQTT:info  2021-11-29 22:50:27: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"button","ieee_address":"0xsnip"},"type":"device_announce"}'
Zigbee2MQTT:debug 2021-11-29 22:50:29: Retrieving state of 'button' after reconnect
Zigbee2MQTT:debug 2021-11-29 22:50:29: Received Zigbee message from 'button', type 'raw', cluster 'genOnOff', data '{"data":[1,90,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-11-29 22:50:29: MQTT publish: topic 'zigbee2mqtt/button', payload '{"action":"1_single","battery":null,"linkquality":90}'
Zigbee2MQTT:info  2021-11-29 22:50:29: MQTT publish: topic 'zigbee2mqtt/button/action', payload '1_single'
Zigbee2MQTT:debug 2021-11-29 22:50:31: Received Zigbee message from 'button', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
```

This is 100% reproducible after waiting for some time between uses of the button.

## Describe the solution you'd like
Instead of debouncing on the trailing edge of the debounce period (send the message after the period has elapsed), allow debouncing on the leading edge (send the message immediately, then suppress further messages within the debounce period). The debounce library used by zigbee2mqtt supports this with a third argument: `debounce(..., immediate: boolean?)`.

As far as I can tell, the required changes would be:
- adding a new configuration value to devices, perhaps `debounce_leading_edge`: https://github.com/Koenkk/zigbee2mqtt/blob/42569d6c39295b6bcbde788a2db94e0262b434e8/lib/types/types.d.ts#L273-L287 and https://github.com/Koenkk/zigbee2mqtt/blob/42569d6c39295b6bcbde788a2db94e0262b434e8/lib/util/settings.schema.json#L693-L706
- adding a new boolean argument to `publishDebounce()` to enable this when it calls `debounce()`: https://github.com/Koenkk/zigbee2mqtt/blob/42569d6c39295b6bcbde788a2db94e0262b434e8/lib/extension/receive.ts#L34-L43
- modifying the call to `publishDebounce()` to pass in the new configuration variable: https://github.com/Koenkk/zigbee2mqtt/blob/42569d6c39295b6bcbde788a2db94e0262b434e8/lib/extension/receive.ts#L146-L148
- updating the frontend for the new setting: https://github.com/nurikk/zigbee2mqtt-frontend/blob/489f251047cbb29f79888b8a4bae48004cd90793/src/i18n/locales/en.json#L1348-L1349 and the other translations, perhaps other areas too (I have not looked deeply in to the frontend)

## Describe alternatives you've considered
I have tried setting the debounce configuration, which worked. However, I needed a 3-second debounce delay, and it is annoying to wait that long for a button press to be recognized.

## Additional context
I am happy to do this work, but only if I get confirmation that it is the right direction to go and I have not missed any areas where changes would need to be made. I'm looking forward to comments on this issue.

Koenkk, 2021-12-20:
> my Tuya TS0044 sends its first button press after wakeup twice:

Could you check if this also happens with the availability feature disabled?

scolby33, 2021-12-20:
The availability feature is disabled (not in my configuration.yaml, environment, and shows as unchecked in the UI)

Koenkk, 2021-12-20:
@Spirituss since you added support for this device, do you also have this issue?

Spirituss, 2021-12-21:
> @Spirituss since you added support for this device, do you also have this issue?

Do you mean IKEA dimmer?

Koenkk, 2021-12-21:
@Spirituss no the "Lonsonho Tuya ZigBee Wireless Smart Scene Switch 4 Gang"  aka TS0044 (https://github.com/Koenkk/zigbee2mqtt/issues/4468)

scolby33, 2021-12-21:
Oh, interesting: although mine is identified as that device in the UI, it in fact is this one https://www.loratap.com/products/ss6400zb. I assumed that it was the same thing in a different form factor, but given that we're looking explicitly at device support it's relevant.

Spirituss, 2021-12-23:
At the moment I see that after some time when the switch was inactive the first button press is ignored. So HA automation on the switch button works only from the second button press in this case.

Spirituss, 2022-01-24:
> @Spirituss since you added support for this device, do you also have this issue?

I have another problem with it. In case when I haven't used the switch for some time, it acts on key press only from the second time. The first press doesn't works.

scolby33, 2022-01-29:
I'm not sure what has changed, but I am no longer experiencing duplicate events from this device. Between my initial report and now, I've upgraded both Home Assistant and Zigbee2MQTT. The debounce setting for this device is disabled in Zigbee2MQTT and I'm not doing any debounce logic in Home Assistant.

I see logs like:
```
Zigbee2MQTT:debug 2022-01-29 21:58:23: Received Zigbee message from 'Desk Button', type 'raw', cluster 'genOnOff', data '{"data":[1,54,253,0],"type":"Buffer"}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2022-01-29 21:58:23: MQTT publish: topic 'zigbee2mqtt/Desk Button', payload '{"action":"2_single","battery":null,"linkquality":51}'
Zigbee2MQTT:info  2022-01-29 21:58:23: MQTT publish: topic 'zigbee2mqtt/Desk Button/action', payload '2_single'
Zigbee2MQTT:debug 2022-01-29 21:58:24: Received Zigbee message from 'Desk Button', type 'raw', cluster 'genOnOff', data '{"data":[1,54,253,0],"type":"Buffer"}' from endpoint 2 with groupID 0
```
The Zigbee message comes in twice, but is only sent out once on MQTT.

So, 🤷 I'm not sure what happened but I'm happy that it's working now!

closed 2022-01-29 by scolby33

BradNagel, 2022-07-06:
Like @Spirituss I too have the issue of the first button hit being ignored.

MarcelJ28, 2022-08-07:
I also have the same issue. The older three ([TS004F) don't have that problem.

papa-smurf, 2022-09-30:
> > @Spirituss since you added support for this device, do you also have this issue?
> 
> I have another problem with it. In case when I haven't used the switch for some time, it acts on key press only from the second time. The first press doesn't works.

I have this exact same issue. It's technically not the first press, but the first action, since the first double-click action also doesn't work.

I've noticed that, after a long period of inactivity (night), all 4 buttons light op briefly which makes me believe it goes in a standby or low power mode from that point onward to save battery life until you wake it up again

cmsj, 2022-10-12:
@papa-smurf interesting discovery. I wonder if it has an option somewhere to not go to sleep? It's super annoying having a quick-access button that doesn't work the first time ;)

papa-smurf, 2022-10-12:
I don't see one in zigbee2mqtt at least and I'm not familiar enough to do further research/debugging sadly. But yeah it's annoying.