Koenkk/zigbee2mqtt #10255: Wrong Firmware build date (dateCode) after OTA

alexwala, 2021-12-20:
## What happened
Recently updated my TRADFRI bulb E27 WW 806lm due [Release Version: 1.19.0](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html) via z2m-ui.

If i go to OTA section, it will display correct `Firmware version 2.3.086`, but i think wrong `Firmware build date 20181203`.

Using Dev console `genBasic.dateCode` `Read result of 'genBasic': {"dateCode":"20181203"}`

## What did you expect to happen
Checking file `10046695-1.1-TRADFRI-light-unified-w-2.3.086.ota.ota.signed` with a hex editor:

```
0002f490h: 1D 06 1F 08 18 01 10 01 08 01 31 02 30 01 39 04 ; ..........1.0.9.
0002f4a0h: F0 08 29 02 21 02 2A 03 22 03 2B 04 23 04 25 06 ; ð.).!.*.".+.#.%.
0002f4b0h: 26 07 28 01 20 01 F1 10 E2 04 04 01 E0 A1 01 01 ; &.(. .ñ.â...à¡..
0002f4c0h: 61 00 0E 49 4B 45 41 20 6F 66 20 53 77 65 64 65 ; a..IKEA of Swede
0002f4d0h: 6E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ; n...............
0002f4e0h: 00 00 00 0F 57 61 72 6D 20 57 68 69 74 65 20 42 ; ....Warm White B
0002f4f0h: 75 6C 62 00 00 00 00 00 00 00 00 00 00 00 00 00 ; ulb.............
0002f500h: 00 00 00 00 08 32 30 32 31 31 30 32 35 00 00 00 ; .....20211025...
0002f510h: 00 00 00 00 00 0F 4C 45 44 31 36 32 32 47 31 32 ; ......LED1622G12
0002f520h: 45 32 37 45 55 00 07 32 2E 33 2E 30 38 36 00 00 ; E27EU..2.3.086..
0002f530h: 00 00 00 00 00 00 00 FF FF FF FF FF FF FF FF FF ; .......ÿÿÿÿÿÿÿÿÿ
0002f540h: FF FF FF 31 66 08 23 2F AC 09 FF FF FF 5A 69 67 ; ÿÿÿ1f.#/¬.ÿÿÿZig
0002f550h: 42 65 65 41 6C 6C 69 61 6E 63 65 30 39 00 03 00 ; BeeAlliance09...
```

Shouldn't be `Firmware build date` `20211025` ... ?

## Debug info
Zigbee2MQTT version: 1.22.1-dev commit: 38f42bb3
Adapter hardware: CC2531
Adapter firmware version: 20201127

Koenkk, 2021-12-20:
It should be whatever the value of the `dateCode` read returns, so this looks to be correct and an issue on the bulb firmwares side.

alexwala, 2021-12-21:
Ok thanks, as the `Firmware version 2.3.086` is correct, I assume that the update was successful. Is it possible to update a device with the same version again somehow ?

Koenkk, 2021-12-21:
@alexwala you cannot update it again (makes no sense), but with the latest dev branch the dateCode/swBuildID will be updated after reading them (and restarting z2m afterwards).  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

alexwala, 2021-12-21:
ok thx

closed 2021-12-21 by alexwala