Koenkk/zigbee2mqtt #10256: Tuya Human Presence Detector manufacturerName: '_TZE200_lu01t0zl'

losip, 2021-12-20:
I ordered a radar human presence detector from the supported devices list but when trying to pair with Z2M, I get a continuous stream of messages saying it's unsupported. Connecting through Tuya gateway works (albeit slowly via cloud). Trying to follow the instructions for adding support for new devices goes well at first and I created Tuya dev account and extracted all 20 Data Points.

The example external converter for the Saswell thermostat was not helpful so I started looking for a more similar device and found that there was already support for a similar radar detector (manufacturerName: '_TZE200_vrfecyku'). It seems silly for me, as a non-programmer, to go any further when it seems that manufacturerName: '_TZE200_lu01t0zl' is a synonym for this one, or at least has all the functionality I need

It seems Tuya has produced a new model but it appears to have exactly the same functionality. Is it possible to clone the existing device to recognize the new manufacturerName?
-->
## Information about the device + link
The link doesn't help - it's the same as the existing supported model. The product name for this one includes Chinese characters and is: NEW_云帆主设备-跌倒雷达+zigbee   while the Zigbee model remains 'TS0601'. Device ID is bf4f51d531f5d41dd0evvv
..

## data/database.db entry of the device 
No entry because it had to be removed due to streaming continuous error messages when apired with Z2M

..

losip, 2021-12-21:
Further information: there is no model number on the outside of the sensor but I took it apart and the PCB is marked IFAR. This may correspond to the previous model AIR - but with the addition of the fall-down detection.

I struggled on trying to follow the instructions but when I got to the part where I had to add the data point definitions to a file on GitHub, I was too scared I'd muck something up, not knowing how to "drive" github. Never mind, the existing TZ and FZ files would provide most of what I want so I created the external filter and placed it in the correct directory. Just to edit configuration.yaml, I thought to include the external converter and set the logging level to debug. Done! But when restarting Z2M, it took no notice of the new configuration.yaml and reverted it to the previous version but with strange formatting that was not yaml (json?)

At this point I screamed and reached for a bottle of wine. Meanwhile, I've had to stop using the device through the new Tuya integration because it stops from time to time. I have got an automation restarting it every 3 hours but I need better reliability through local Zigbee connection rather than through the Tuya gateway

losip, 2021-12-26:
Data Points are:
1 - Presence State
2 - Sensitivity (numeric - withValueMin(1).withValueMax(10).withValueStep(1))
101 - Reset Flag
102 - Motion State (binary - 'human detected as peaceful or moving')
103 - Illumination
104 - Detection Data
105 - Tumble Switch (binary)
106 - Tumble Alarm Time (numeric - withValueMin(1).withValueMax(5).withValueStep(1))
107 - Radar Check End (time)
108 - Radar Check Start (time)
109 - HW Version
110 - SW Version
111 - Radar ID
112 - Scene (enum - 'default', 'area', 'toilet', 'bedroom', 'parlour', 'office', 'hotel')
113 - undefined
114 - Motion Direction (enum: 'none', 'close to', 'away from')
115 - Motion (binary)
116 - Fall Down Status (binary)
117 - Static_dwell_alarm (binary)
118 - Fall sensitivity (numeric - withValueMin(1).withValueMax(10).withValueStep(1))

norsoa, 2021-12-28:
I wounder if you have successfully integrated the sensor with a local ZigBee connection? Like you I am searching for a presence sensor with ZigBee and 230v. I am very interested in getting feedback how this sensor i working in a real life setup.

losip, 2021-12-28:
No, I've given up. I'm an engineer, not a developer and, although I tried, I was unable to make an external converter work. Device works with Zigbee using the new Tuya integration but that's too slow and unreliable for my purposes so I really wanted it to work with local Zigbee2mqtt. I can't do it on my own

losip, 2021-12-29:
This is the external converter I tried:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_lu01t0zl'}],
    model: 'NEW Cloud Sail Main Equipment-Falling Down Radar+zigbee',
    vendor: 'TuYa',
    description: 'Human presence sensor',
    fromZigbee: [fz.ignore_basic_report, fz.tuya_radar_sensor],
    toZigbee: [tz.tuya_radar_sensor],
    onEvent: tuya.setTime, // Add this if you are getting no converter for 'commandSetTimeRequest'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        e.illuminance_lux(), e.presence(),
        exposes.binary('motion', ea.STATE, [true, false]).withDescription('moving inside the range of the sensor'),
        exposes.binary('motion_state', ea.STATE, [true, false]).withDescription('human detected as peaceful or moving'),
        exposes.binary('fall_down_status', ea.STATE, [true, false]).withDescription('human fallen down detected'),
        exposes.binary('static_dwell_alarm', ea.STATE, [true, false]).withDescription('human has been still for too long'),
        exposes.binary('tumble_switch', ea.STATE_SET, [true, false]).withDescription('enables tumble detection'),
        exposes.numeric('motion_speed', ea.STATE).withDescription('Speed of movement'),
        exposes.enum('motion_direction', ea.STATE, ['none', 'close to', 'away from'])
            .withDescription('direction of movement from the point of view of the radar'),
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(10).withValueStep(1)
            .withDescription('sensitivity of the radar'),
        exposes.numeric('fall_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(10).withValueStep(1)
            .withDescription('sensitivity of the fall detection'),
        exposes.enum('radar_scene', ea.STATE_SET, ['default', 'area', 'toilet', 'bedroom', 'parlour', 'office', 'hotel'])
            .withDescription('presets for sensivity for presence and movement'),
        exposes.numeric('tumble_time', ea.STATE_SET).withValueMin(1).withValueMax(5).withValueStep(1)
            .withDescription('set time in minutes for detection of a tumble'),
    ],
},

module.exports = definition;

prokudin07, 2022-01-03:
> This is the external converter I tried: const fz = require('zigbee-herdsman-converters/converters/fromZigbee'); const tz = require('zigbee-herdsman-converters/converters/toZigbee'); const exposes = [...]

Hello, this converter will work?
I bought the same device and have the problem with z2m.

prokudin07, 2022-01-04:
Hello! I can to start this device in Z2M. Just replaced manufacturerName in file tuya.js  and it work.

prokudin07, 2022-01-04:
Have two way @losip :
1. wait while @Koenkk do it  in herdsman-converter and it will be available for all users of z2m
2. while we wait paragraph 1 - in your instance of z2m on you local computer (i have raspberry pi) find file tuya.js, in that file find string "vrfecyku" and replace it on "lu01t0zl", then restart your z2m and rejoin your device.  But remember when you will update your z2m  - you must repeat that operation one more time.

losip, 2022-01-04:
I think I'll have to wait for option 1. I can't find tuya.js anywhere on the computer that's running HA. Is it hidden somewhere?

prokudin07, 2022-01-04:
What kind of installation of HA do you have? HASOS, HA in docker? 
I have docker installation of HA and login via ssh on my raspberry pi. That file (tuya.js) isn't in directory of HA, other place. If you use Telegram can find me @prokudinvn

losip, 2022-01-04:
HASSOS on generic x64. I have Samba mapped to config directory but I'll try using my non-existent Linux skills to look elsewhere. Seems strange for it not to be somewhere under the zigbee2mqtt directory

prokudin07, 2022-01-04:
Maybe I wrong, but in HassOS it's impossible. But... Turn off your comp with HA, unplug disk (ssd, flash) and plug in orher comp and I think you will find that file)

losip, 2022-01-04:
Ah, therein lies a problem: it's a cheap mini PC with 64Gb emmc soldered to the circuit board! I guess I'll just have to wait...

prokudin07, 2022-01-04:
Ok. I soon will ask my friend who works with koenkk - add this device to z2m

prokudin07, 2022-01-04:
> Ah, therein lies a problem: it's a cheap mini PC with 64Gb emmc soldered to the circuit board! I guess I'll just have to wait...

I got consultation. You can use addon SSH and find that file.
[image]

prokudin07, 2022-01-05:
Hello @losip 
I did  [pull request](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3624)
then will wait

losip, 2022-01-05:
Thank you @prokudin07 . Noobs like me have no idea what a pull request is or how to 'operate' github so that is very useful

prokudin07, 2022-01-05:
@losip I did it first time too )) Other man helped me.

prokudin07, 2022-01-05:
@losip  pull request was accept. Try to install Z2MEdge - it's beta version of Z2M and try to join senser.

losip, 2022-01-05:
Yes, thanks @prokudin07 . That's working. It will do all that I need although I suppose that, in due course, the "fall down" and "tumble" support should be added using the data points I listed above. Would you know how to go about that?

prokudin07, 2022-01-05:
I don't know, but later i'll try do it.

prokudin07, 2022-01-14:
Hello @losip 
I doing but it's all new for me and so slow, but on screenshot - did 2 exposes. I hope soon will do it)
[image]

prokudin07, 2022-01-16:
Hello @losip 
I did it!!!  Now this device is avaible in z2mEdge
[Pull request](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3686)

losip, 2022-01-16:
Congratulations @prokudin07 !! I'll update Edge

closed 2022-01-16 by losip

prokudin07, 2022-01-19:
Hello @losip 
I updated  a little. 
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3701

losip, 2022-01-19:
Thank you. The only thing that seems to be missing is the picture. I get the generic device image.

Nothing to do with the integration or Z2M but I’m disappointed in the performance of the device. If I have the radar sensitivity set to reliably detect presence at 7 or 8, then it doesn’t switch back to ‘away’ unless I set the sensitivity down to 3 or 4. I may swap it out for a different detector.

Peter

Koenkk/zigbee-herdsman-converters#3701<https://github.com/Koenkk/zigbee-herdsman-converters/pull/3701>

losip, 2022-01-19:
> About picture: I know, bгt Koenk advised me add this device as new device and I don't know how do description that device
> on the page<https://www.zigbee2mqtt.io/supported-devices/>
>
> About "I’m disappointed": strange. My deviсe very accurately determines the presence or absence.
> From nobody to someone - 0.5 sec
> from someone to nobode - about 2 min.
>
> Here<https://buy-lcd.com/products/24ghz-wireless-bio-radar-sensor-for-fall-detection-ir24fda> is datasheet and user manual of this sensor
> in Library

Thanks. The right picture was there before your previous update but no matter – it doesn’t affect the operation.

I have mine in a bathroom to switch the lights on when presence is detected and switch them off after 2 minutes of nobody detected.  What settings do you have for sensitivity and scene?  I might try again – for now, I’ve installed the ‘AIR’ radar but it seems to be more like a motion detector, so I’m still not happy with the solution

Thanks for the light to the data sheet; I’ll study it and see if I’ve got the location or orientation wrong – but I was careful to read the instructions that came with the device

prokudin07, 2022-01-19:
[image]

haint126, 2022-07-08:
> Data Points are: 1 - Presence State 2 - Sensitivity (numeric - withValueMin(1).withValueMax(10).withValueStep(1)) 101 - Reset Flag 102 - Motion State (binary - 'human detected as peaceful or [...]

I Just bought a few of these devices and keep getting missing datapoint warnings. I found your thread very helpful.
May I ask where did you find these datapoint definitions and what are their functions?
Thank you very much @losip

losip, 2022-07-08:
@haint126 I got these datapoints by pairing the device with a Tuya hub and then using the Tuya IOT platform to follow the instructions provided by @Koenkk for adding a new Tuya device. I have no further information about their functionality

haint126, 2022-07-08:
Oh, I did not know there was instruction for that. Just found it.
For anyone missed that instruction like me, here is the link: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html
Thank you very much @losip

SeLLeRoNe, 2023-02-11:
I seem to be facing the same issue with model: _TZE200_jkbljri7

I was using this converter, but now suddently it returns errors (latest: tuya is not defined):

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_jkbljri7'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_jkbljri7'}],
        model: 'MIR-HE200-TY',
        vendor: 'TuYa',
        description: 'Human presence sensor with fall function',
        fromZigbee: [fz.tuya_radar_sensor_fall],
        toZigbee: [tz.tuya_radar_sensor_fall],
        exposes: [
            e.illuminance_lux(), e.presence(), e.occupancy(),
            exposes.numeric('motion_speed', ea.STATE).withDescription('Speed of movement'),
            exposes.enum('motion_direction', ea.STATE, Object.values(tuya.tuyaRadar.motionDirection))
                .withDescription('direction of movement from the point of view of the radar'),
            exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1)
                .withDescription('sensitivity of the radar'),
            exposes.enum('radar_scene', ea.STATE_SET, Object.values(tuya.tuyaRadar.radarScene))
                .withDescription('presets for sensitivity for presence and movement'),
            exposes.enum('tumble_switch', ea.STATE_SET, ['ON', 'OFF']).withDescription('Tumble status switch'),
            exposes.numeric('fall_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(10).withValueStep(1)
                .withDescription('fall sensitivity of the radar'),
            exposes.numeric('tumble_alarm_time', ea.STATE_SET).withValueMin(1).withValueMax(5).withValueStep(1)
                .withUnit('min').withDescription('tumble alarm time'),
            exposes.enum('fall_down_status', ea.STATE, Object.values(tuya.tuyaRadar.fallDown))
                .withDescription('fall down status'),
            exposes.text('static_dwell_alarm', ea.STATE).withDescription('static dwell alarm'),
        ],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await tuya.sendDataPointEnum(endpoint, tuya.dataPoints.trsfTumbleSwitch, false);
        }
    };

module.exports = definition;
`
Maybe someone can help me with this converter?

Thanks!

SeLLeRoNe, 2023-03-16:
Ok, so I got the above updated to work with latest updates (I think the template has changed), new one is:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_jkbljri7',
        },
    ],
    model: 'MIR-HE200-TY',
    vendor: 'TuYa',
    description: 'Human presence sensor with fall function',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        e.illuminance_lux(), e.presence(), e.occupancy(),
        exposes.numeric('motion_speed', ea.STATE).withDescription('Speed of movement'),
        exposes.enum('motion_direction', ea.STATE, Object.values(tuya.tuyaRadar.motionDirection))
            .withDescription('direction of movement from the point of view of the radar'),
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1)
            .withDescription('sensitivity of the radar'),
        exposes.enum('radar_scene', ea.STATE_SET, Object.values(tuya.tuyaRadar.radarScene))
            .withDescription('presets for sensitivity for presence and movement'),
        exposes.enum('tumble_switch', ea.STATE_SET, ['ON', 'OFF']).withDescription('Tumble status switch'),
        exposes.numeric('fall_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(10).withValueStep(1)
            .withDescription('fall sensitivity of the radar'),
        exposes.numeric('tumble_alarm_time', ea.STATE_SET).withValueMin(1).withValueMax(5).withValueStep(1)
            .withUnit('min').withDescription('tumble alarm time'),
        exposes.enum('fall_down_status', ea.STATE, Object.values(tuya.tuyaRadar.fallDown))
            .withDescription('fall down status'),
        exposes.text('static_dwell_alarm', ea.STATE).withDescription('static dwell alarm'),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
        ],
    },
};

module.exports = definition;
`

Only issue I have is that I don't have a Tuya hub, so I can't make the right thing to have it correctly paired :)
The only odd thing I have for now seems to be that occupancy return Null when there is none, but because I can't see exact codes, but beside that, this should be now be somewhat working :)

SeLLeRoNe, 2023-09-07:
Unfortunately the above no longer works, I think there was an update on how the template needs to be setup but I haven't yet had the time to try to update the above, so I had to remove this customization and that device is just a piece of unrecognized zigbee device :)

SeLLeRoNe, 2023-09-07:
Looking for the Manufacturer I see another github repo that seems to have had the update to manage that: https://github.com/ioBroker/ioBroker.zigbee/issues/1672

Maybe someone with more understanding on how to implement the code for it could throw some help in the mix :)
Would be much appreciated.

Thanks