Koenkk/zigbee2mqtt #10257: Expose switch as light doesn't do nothing in Home assistant

adizag, 2021-12-20:
## What happened
Trying to expose switch as light to home assistant, still exposed as switch.

## What did you expect to happen
Expected to have light entities instead of switches

## my config file:
```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://10.0.0.251'
serial:
  port: /dev/ttyACM0
advanced:
  network_key:
    - 18
    - 73
    - 40
    - 36
    - 202
    - 152
    - 137
    - 228
    - 72
    - 181
    - 149
    - 126
    - 178
    - 90
    - 66
    - 57
  log_level: debug
'\n\nadvanced:\n    network_key': GENERATE
frontend:
  port: 8080
  host: 10.0.0.102
devices:
  '0x00158d0006a25d21':
    friendly_name: '0x00158d0006a25d21'
  '0x00158d0006a157a6':
    friendly_name: '0x00158d0006a157a6'
  '0x00158d0006f25853':
    friendly_name: '0x00158d0006f25853'
  '0x00158d0006a261df':
    friendly_name: '0x00158d0006a261df'
  '0x00158d0006f7805b':
    friendly_name: '0x00158d0006f7805b'
  '0x00158d0006a25f66':
    friendly_name: Salon_Kitchen
    occupancy_timeout: 10
  '0xbc33acfffec80607':
    friendly_name: Stairs Up
    homeassistant:
      switch:
        type: light
        object_id: stairs_up_light
      light:
        name: Stairs Up
        value_template: null
        state_value_template: '{{ value_json.state }}'
  '0x5c0272fffe8ef707':
    friendly_name: Stairs Down
    homeassistant:
      switch:
        type: light
        object_id: stairs_down_light
      light:
        name: Stairs Down
[... 249 lines cut ...]
debug 2021-12-20 13:10:02: Received Zigbee message from 'Stairs Up', type 'attributeReport', cluster 'genBasic', data '{"65506":27,"appVersion":66}' from endpoint 1 with groupID 0
debug 2021-12-20 13:11:01: Received Zigbee message from 'Stairs Down', type 'attributeReport', cluster 'genBasic', data '{"65506":27,"appVersion":66}' from endpoint 1 with groupID 0
debug 2021-12-20 13:11:03: Received Zigbee message from 'mamad 2nd', type 'attributeReport', cluster 'genBasic', data '{"65506":19,"appVersion":128}' from endpoint 1 with groupID 0
debug 2021-12-20 13:11:13: Received Zigbee message from 'Mamad', type 'attributeReport', cluster 'genBasic', data '{"65506":27,"appVersion":66}' from endpoint 1 with groupID 0
debug 2021-12-20 13:11:39: Received MQTT message on 'zigbee2mqtt/mamad 2nd/l1/get' with data '{"state_l1":""}'
debug 2021-12-20 13:11:39: Publishing get 'get' 'state' to 'mamad 2nd'
debug 2021-12-20 13:11:39: Received Zigbee message from 'mamad 2nd', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2021-12-20 13:11:39: MQTT publish: topic 'zigbee2mqtt/mamad 2nd', payload '{"linkquality":84,"state_l1":"OFF","state_l2":"ON"}'
debug 2021-12-20 13:13:11: Received Zigbee message from 'Stairs Up', type 'attributeReport', cluster 'genBasic', data '{"65506":27,"appVersion":66}' from endpoint 1 with groupID 0
debug 2021-12-20 13:13:45: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2021-12-20 13:14:17: Received Zigbee message from 'Stairs Down', type 'attributeReport', cluster 'genBasic', data '{"65506":27,"appVersion":66}' from endpoint 1 with groupID 0
debug 2021-12-20 13:14:18: Received Zigbee message from 'mamad 2nd', type 'attributeReport', cluster 'genBasic', data '{"65506":19,"appVersion":128}' from endpoint 1 with groupID 0
debug 2021-12-20 13:14:20: Received Zigbee message from 'Mamad', type 'attributeReport', cluster 'genBasic', data '{"65506":27,"appVersion":66}' from endpoint 1 with groupID 0
debug 2021-12-20 13:16:25: Received Zigbee message from 'Stairs Up', type 'attributeReport', cluster 'genBasic', data '{"65506":27,"appVersion":66}' from endpoint 1 with groupID 0
debug 2021-12-20 13:17:23: Received Zigbee message from 'Mamad', type 'attributeReport', cluster 'genBasic', data '{"65506":27,"appVersion":66}' from endpoint 1 with groupID 0
debug 2021-12-20 13:17:33: Received Zigbee message from 'Stairs Down', type 'attributeReport', cluster 'genBasic', data '{"65506":27,"appVersion":66}' from endpoint 1 with groupID 0
debug 2021-12-20 13:17:34: Received Zigbee message from 'mamad 2nd', type 'attributeReport', cluster 'genBasic', data '{"65506":19,"appVersion":128}' from endpoint 1 with groupID 0
debug 2021-12-20 13:18:45: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2021-12-20 13:19:30: Received Zigbee message from 'Stairs Up', type 'attributeReport', cluster 'genBasic', data '{"65506":27,"appVersion":66}' from endpoint 1 with groupID 0
```

## Debug info
Zigbee2MQTT version: 1.22.1
Adapter hardware: CC2531
Adapter firmware version: zStack12

adizag, 2021-12-23:
@Koenkk ? any Idea?

adizag, 2022-01-03:
Please help??

paataD, 2022-01-04:
Just try to remove your device from HA then restart mqtt server for auto descovery new devices

adizag, 2022-01-04:
Still - discovered as switch! ahhhhhh

____________________________________________
"יש המטביעים את צרותיהם - ויש המלמדים אותן לשחות..."
"אין מקום שהוא רחוק מדי"

Cossey, 2022-01-06:
I think there might be a bug in the homeassistant auto discovery in zigbee2mqtt?

Looking at the documentation here https://www.home-assistant.io/docs/mqtt/discovery/#lighting it says that the autodiscovery should be at `<homeassistanttopic>/light/<devicename>/config` but I notice when using MQTT explorer on version `1.22.2 commit: 414c51f` that my Philips hue lights show up in `homeassistant/light/<devicename>/light/config`. 

[image]

adizag, 2022-02-06:
any idea why it is not discovered as light?
It drives me crazy...

brokenmass, 2022-02-08:
the functionality has been broken in a recent update

adizag, 2022-02-16:
My bad, didn't notice there is a different setup for multiple endpoints....

closed 2022-02-16 by adizag

jbytheway, 2023-08-07:
@adizag it's not clear from your last comment whether you got this working or decided it was not supported. If you got it working, can you share your resolution? I've been trying to figure out how to achieve the same thing (exposing switches as lights), but have found no documentation on it. This issue is the only hint I've found that it might be possible at all.

adizag, 2023-08-07:
@jbytheway 
here is my config for some different switches. 
please not that the conf must follow the switch endpoints names as seen in the Z2M GUI:
[image]

```
    friendly_name: Mamad 2nd
    homeassistant:
      switch_l1:
        type: light
        object_id: light_l1
      light_l1:
        name: stairs_down_light_1
        value_template: null
        state_value_template: '{{ value_json.state_l1 }}'
      switch_l2:
        type: light
        object_id: light_l2
      light_l2:
        name: stairs_down_light_2
        value_template: null
        state_value_template: '{{ value_json.state_l2 }}'
```

[image]

```
    friendly_name: Stairs Up
    homeassistant:
      switch_left:
        type: light
        object_id: light_left
      light_left:
        name: stairs_up_left
        value_template: null
        state_value_template: '{{ value_json.state_left }}'
      switch_center:
        type: light
        object_id: light_center
      light_center:
        name: stairs_up_center
        value_template: null
        state_value_template: '{{ value_json.state_center }}'
      switch_right:
        type: light
        object_id: light_right
      light_right:
        name: stairs_up_right
        value_template: null
        state_value_template: '{{ value_json.state_right }}'

```

The instructions aren't actually very clear, maybe better to clarify @Koenkk ?

GOOD LUCK!

jbytheway, 2023-08-07:
Thanks!  That was enough to enable me to find [the documentation I was previously missing](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#exposing-switch-as-a-light).