Koenkk/zigbee2mqtt #10259: SNZB-03 dies at 80% battery

elgerg, 2021-12-20:
## What happened
My SNZB-03 reported it's battery at 80%, 2900 voltage. It however stopped clearing the occupied state then stopped reporting in at all.

## What did you expect to happen
I expected it to stop working at between 0% and 10%

## How to reproduce it (minimal and precise)
Let the battery deplete.

## Debug info
Zigbee2MQTT version: 1.22.1
Adapter hardware: Slaesh's CC2652RB stick
Adapter firmware version:
Coordinator type zStack3x0
Coordinator revision 20210708

Koenkk, 2021-12-20:
What voltage does it report when you put in a new battery? Based on that I can change the battery % range.

elgerg, 2021-12-20:
3200 but it should be a 3v battery. very odd

Koenkk, 2021-12-20:
Accuracy should be better now (can take up to 1 day to report)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

elgerg, 2021-12-20:
Wow, nice one. Ill give it a go when I can. Thanks!

elgerg, 2021-12-20:
However looking at one of my other PIRs I have this:

{
    "battery": 65,
    "battery_low": false,
    "linkquality": 60,
    "occupancy": false,
    "tamper": false,
    "voltage": 2900
}

and

{
    "battery": 76.5,
    "battery_low": false,
    "linkquality": 84,
    "occupancy": false,
    "tamper": false,
    "voltage": 2900
}

Will that cause any issues?

Koenkk, 2021-12-20:
no, since we will now rely on the battery % reported by the device, not the one derived from the voltage.

elgerg, 2021-12-20:
Ah, cool. Thanks!

closed 2022-01-27 by github-actions[bot]