Koenkk/zigbee2mqtt #10261: No OTA update found for some IKEA GU10 bulbs

andrasg, 2021-12-20:
## What happened

I have a set of IKEA bulbs. Some are getting OTA updates to the latest version `2.3.087`, others can't find any updates and are stuck on `2.0.029`.

[image]

## What did you expect to happen
The bulbs should find the latest update `2.3.087`.

## How to reproduce it (minimal and precise)
Tried force removing the bulbs, z2m restart, readding. Still no OTA update found.

## Debug info
Zigbee2MQTT version: 1.22.1
Adapter hardware: CC2652R1
Adapter firmware version: 20210708 zStack 3x0

sjorge, 2021-12-20:
I'm guessing this is a old cashed version, even when removing and adding the bulb again it pulls it from the cache unless you restart z2m between removing and adding.

You can check if this is the case by manually reading the value using the dev console:
[image]

At least that should allow you to verify they are up to date, I assume they are as the check action does not use the info cached in the database.

andrasg, 2021-12-20:
> I'm guessing this is a old cashed version, even when removing and adding the bulb again it pulls it from the cache unless you restart z2m between removing and adding.

I already tried this approach (force remove, z2m restart, re-add), but the versions remained the same. 

Using the approach you have suggested, I see an old fw version:

[image]

andrasg, 2021-12-20:
> This is a known issue and koen is working on a fix: #10131

Hmmm, this issue seems to conclude that there is a case where the an old version is reported, but the actual version is up-to-date. In my case, it seems that the device is on an older firmware and the update is not found.

sjorge, 2021-12-20:
I think currently one 1 GU10 bulb doesn't have a firmware update yet, I you have that one. I see the model number mentiones 2 different GU10 that got grouped together.

Is the model number on the neck of the bulb the same between the ones with and without updated swBuildId?

andrasg, 2021-12-20:
That might very well be the case... Will climb up to the bulbs tomorrow to check. Will also try the dev branch to see if it's any different.

andrasg, 2021-12-20:
No update found on `latest-dev` either.

sjorge, 2021-12-20:
Looks like I have one of those in my test box...

LED1739R5, which I believe is the older of the 2 variants, I does not seem to have an update available and it's also reporting 2.0.029. It looks different from the picture, the plastic cover is smooth.

lukyjay, 2021-12-25:
Hi please try disabling the OTA test server. I did this and all of my bulbs updated to the latest firmware.

advanced:
  ikea_ota_use_test_url: false
in your configuration.yaml, or find the option on the 'advanced' tab in the settings and restart.

andrasg, 2021-12-26:
> Hi please try disabling the OTA test server. I did this and all of my bulbs updated to the latest firmware.

I did have the test server setting in there previously, it absolutely found no updates with the test server. I have removed the test setting, rebooted Z2M, that's how I ended up with only some of the lights updating.

I have now climbed to the bulbs. They are indeed different hardware. The ones updating are LED1537R6 ones (white body, slight protrusion around the lenses), while the ones not finding an update are LED1739R5's (transparent body, silver reflectors can be seen, flat front).

sjorge, 2021-12-26:
The 2 bulbs in my test box are both R5 and no update found for them here as mentioned earlier.

andrasg, 2021-12-26:
Thanks, mystery solved then, closing issue :)

closed 2021-12-26 by andrasg