Koenkk/zigbee2mqtt #10263: NAS-WR01B

totom80, 2021-12-20:
Hello,
I've buy some wall plug NAS-WR01B (Tuya like) and I can't use them.
When I paired, I've a warning log: 
Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_gjnozsaz'

Is it possible to add in a futur version ?

Thanks

Koenkk, 2021-12-20:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-20 by Koenkk

PrzemekSkw, 2024-11-02:
Hello, I use that plug for some time but now when I try to pair again I cannot. Nothing happen. I click button on my plug for 5 or 10 seconds but nothing happen. There was some error show in z2m dashboard but now when I try again there's no info in logs also. Can I do something or plug is trash?
Regard

zhuykovkb, 2025-01-06:
@PrzemekSkw i have the same problem. it looks like better to open new issue, because current is closed

PrzemekSkw, 2025-01-06:
Hi, I get rid of that plug. I think ZigBee module was broken.
Regards.