Koenkk/zigbee2mqtt #10280: Frontend should work without internet connection Remove call to (cloudflare all.min.css)

hppyworld, 2021-12-20:
# Bug Report
Frontend should work without internet connection.
Its better to not call internet for frontend load. it should work without calling external service.
https://cdnjs.cloudflare.com/ajax/libs/font-awesome/5.15.2/css/all.min.css

## What happened

## What did you expect to happen
Frontend should work without calling to internet service 
https://cdnjs.cloudflare.com/ajax/libs/font-awesome/5.15.2/css/all.min.css

## How to reproduce it (minimal and precise)
Open frontend in webbrowser without internet and you will see it tries to load External CSS and can not work as internet is needed to load CSS.

sjorge, 2021-12-20:
You might want to file this here: https://github.com/nurikk/zigbee2mqtt-frontend
As the frontend is a seperate repo/developer.

Koenkk, 2021-12-21:
Frontend issues should be created here indeed: https://github.com/nurikk/zigbee2mqtt-frontend

closed 2021-12-21 by Koenkk