Koenkk/zigbee2mqtt #10286: TuYa S-LUX-ZB

boklubre, 2021-12-21:
Hello,

I have tried to include an new TuYa S-LUX-ZB.  It´s listed as a supported device.
But when I try to include in Zigbee2MQTT (Home Assistant) the devices is marked as unsupportet.
Is my device meaybe a new version?
Cheers

Boris

Koenkk, 2021-12-24:
Make sure you are on 1.22.1 first. 
If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

If you are on 1.22.1, provide the data/database.db entry of this device

fautor, 2021-12-25:
I have the same problem and here is entry from data/database.db:

`{"id":6,"type":"EndDevice","ieeeAddr":"0x50325ffffe586dba","nwkAddr":19966,"manufId":4098,"manufName":"_TZE200_pisltm67","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_pisltm67","powerSource":3,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1640462766557,"useImplicitCheckin":true}`

boklubre, 2022-01-03:
After the last Update still not working.

hossface, 2022-01-05:
Mine also stopped working after the update

boklubre, 2022-02-07:
Unfortunately still not working.

deepcoder, 2022-02-18:
Any thought on getting to work. Tried one with zigbee2mqtt 1.23.0 and still says not supported, this unit seems to have different model and manufacture than is listed for the unit here:

Mine reports:
TS0601
_TZE200_pisltm67

Blakadder database reports:
https://zigbee.blakadder.com/Tuya_S-LUX-ZB.html
Zigbee ID: "isltm67\u0000"; "_TYST11_pisltm67"

deepcoder, 2022-02-18:
As a follow up to my post above, I created a external converter as shown below. This seems to be reporting the lux and link quality correctly, however it continues to report battery and battery_low reading, even though I believe commented these out. In the template, used to start from it appears the code does not include these values to be returned by this particular model/manufacture signature. I'm over my head here, so I am not sure why it is excluded and why it appears in my instance to be continuing to be reported. I have included the database.db entry from my zigbee2mqtt 1.23.0 setup as well below. 

herdsman code I started with:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/b24d133403fc5561ac5aa209672ebecc7445f53e/devices/tuya.js

test of lux reporting, seems to work correctly, reports about once per second when there is a change:
```
{"battery":"100.0","battery_low":"2.0","elapsed":120,"illuminance_lux":"776","last_seen":"2022-02-17T21:38:28-08:00","linkquality":134}
{"battery":"100.0","battery_low":"2.0","elapsed":6436,"illuminance_lux":"10","last_seen":"2022-02-17T21:38:34-08:00","linkquality":142}
```
```

// TuYa S-LUX-ZB Illuminance Detector
// 2022-Feb-17

// Press and hold to 'arrow' button on the front to put into pairing mode, hold for 5 seconds or so

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_pisltm67'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_pisltm67'}],
        model: 'S-LUX-ZB',
        vendor: 'TuYa',
        description: 'Light sensor',
        fromZigbee: [fz.SLUXZB],
        toZigbee: [],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
        },
        exposes: (device, options) => {
            return [e.illuminance_lux(), e.linkquality()];
            // if (device && device.manufacturerName === '_TZE200_pisltm67') {
            //     return [e.illuminance_lux(), e.linkquality()];
            // } else {
            //     return [e.battery(), e.illuminance_lux(), e.battery_low(), e.linkquality()];
            // }
        },
    };

module.exports = definition;

```

```
{
   "id":12,
   "type":"EndDevice",
   "ieeeAddr":"0xa4c13809ddcefc6e",
   "nwkAddr":50091,
   "manufId":4417,
   "manufName":"_TZE200_pisltm67",
   "powerSource":"Battery",
   "modelId":"TS0601",
   "epList":[
      1
   ],
   "endpoints":{
      "1":{
         "profId":260,
         "epId":1,
         "devId":81,
         "inClusterList":[
            4,
            5,
            61184,
            0
         ],
         "outClusterList":[
            25,
            10
         ],
         "clusters":{
            "genBasic":{
               "attributes":{
                  "modelId":"TS0601",
                  "manufacturerName":"_TZE200_pisltm67",
                  "powerSource":3,
                  "zclVersion":3,
                  "appVersion":67,
                  "stackVersion":0,
                  "hwVersion":1,
                  "dateCode":""
               }
            }
         },
         "binds":[
            {
               "cluster":0,
               "type":"endpoint",
               "deviceIeeeAddress":"0x00124b0024c16a25",
               "endpointID":1
            }
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      }
   },
   "appVersion":67,
   "stackVersion":0,
   "hwVersion":1,
   "dateCode":"",
   "zclVersion":3,
   "interviewCompleted":true,
   "meta":{
      "configured":821693351
   },
   "lastSeen":1645162367283,
   "defaultSendRequestWhen":"immediate"
}
```

closed 2022-03-29 by github-actions[bot]

willthong, 2022-07-19:
Thank you @deepcoder - this code worked well following [these instructions](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#instructions), aside from not returning battery status.