Koenkk/zigbee2mqtt #10287: 1 gang smart dimmer switch QS-Zigbee-D02-TRIAC-LN brightness doesn't work

Daniel-WW, 2021-12-21:
Hello, 

im using the aforementioned [dimmer switch](https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-LN.html) but the brightness setting does not work. 

I found out, that the problem has already be mentioned in other issues (#8834 #8575 #9878) but I couldn't found a solution how to let it work. Can anyone help me how to adjust my converter file?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS110E'],
    model: 'QS-Zigbee-D02-TRIAC-LN',
    vendor: 'RTX',
    description: '1 gang smart dimmer switch module with neutral',
    // Note that fromZigbee, toZigbee and exposes are missing here since we use extend here.
    // Extend contains a default set of fromZigbee/toZigbee converters and expose for common device types.
    // The following extends are available:
    // - extend.switch
    // - extend.light_onoff_brightness
    // - extend.light_onoff_brightness_colortemp
    // - extend.light_onoff_brightness_color
    // - extend.light_onoff_brightness_colortemp_color
    extend: extend.light_onoff_brightness(),
};

module.exports = definition;
```

Sorry, I am quite a newbie in this field and would be grateful for any kind of help.
What do I have to change?

Thanks in advance and best regards - Daniel

Daniel-WW, 2021-12-22:
Hello, 
i guess I have found a solution: I used the converter file from this [post](https://github.com/Koenkk/zigbee2mqtt/issues/9878#issuecomment-981736573) and now the brightnes works fine. Best regards - Daniel

closed 2021-12-22 by Daniel-WW

Koenkk, 2021-12-22:
@pavax could you create a pr for this?