Koenkk/zigbee2mqtt #10297: TuYa TS1101_dimmer_module _TZ3000_xfs39dbf

ovcze, 2021-12-21:
## What happened
Dimmer TuYa TS1101_dimmer_module _TZ3000_xfs39dbf is only 1 channel dimmer, but it is exposing as 2 channnel dimmer (_TZ3000_7ysdnebc).

AliExpress link: https://www.aliexpress.com/item/1005003012298844.html

## What did you expect to happen
Dimmer TuYa TS1101_dimmer_module _TZ3000_xfs39dbf should have only one endpoint not two.

## How to reproduce it (minimal and precise)
After pairing the TuYa TS1101_dimmer_module _TZ3000_xfs39dbf is created with 2 endpoints. 

## Debug info
Zigbee2MQTT version: 1.22.1 
Adapter hardware: CC2530
Adapter firmware version: 20190608

Koenkk, 2021-12-24:
@PshemoSK since you added this in https://github.com/Koenkk/zigbee-herdsman-converters/pull/3322, can you confirm `_TZ3000_xfs39dbf` is indeed a one channel dimmer?

marthubner, 2021-12-29:
@Koenkk `_TZ3000_xfs39dbf` really is a one channel dimmer and mine just got recognized as a two channel.
It is this device in the picture 
[image]
[image]

Koenkk, 2021-12-29:
I see, fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-29 by Koenkk