Koenkk/zigbee2mqtt #10299: "payload is not iterable" when configuring reporting

6cUbi57z, 2021-12-21:
## What happened

Devices fail to configure reporting on startup, reconfiguration or manually. This results in hue devices not updating their state when updated from bound devices. Logs show a "payload is not iterable" error.

https://pastebin.com/JSRkeQAV

## What did you expect to happen

No errors in logs. When lights are turned off via bound switch, zigbee2mqtt and HA state should update automatically.

## How to reproduce it (minimal and precise)

This seems to happen on all of my Philips Hue and Gledopto devices. Ikea and SmartThings devices seem to be fine.

The error happens on restart, reconfiguration, re-adding of the device and configuring reporting (onOff with default options).

## Debug info
Zigbee2MQTT version: 1.22.1
Adapter hardware: Conbee II
Adapter firmware version:
```
"coordinator": {
        "meta": {
            "maintrel": 0,
            "majorrel": 38,
            "minorrel": 102,
            "product": 0,
            "revision": "0x26660700",
            "transportrev": 0
        },
        "type": "ConBee2/RaspBee2"
    }
```

Koenkk, 2021-12-24:
This means the device is not configurable, however you should get a more friendly error message for this. Please provide the herdsman debug logging.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

6cUbi57z, 2021-12-25:
Thanks for the quick response.

I've enabled herdsman logging and this seems to be what shows when attempting to configure reporting:

https://pastebin.com/1CY2KL4w

Koenkk, 2021-12-27:
@chrishae could you check this?

Koenkk, 2022-01-10:
Should be fixed now, can you confirm?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

ChrisHae, 2022-01-11:
Now the correct error UNSUP_GENERAL_COMMAND is shown.

6cUbi57z, 2022-01-11:
As above, seems to be showing the correct error now. I also found the part in the documentation that says Hue bulbs don't support reporting except on/off (although that doesn't seem to work either).

closed 2022-01-11 by 6cUbi57z