Koenkk/zigbee2mqtt #10303: Schneider Wiser roller shutter  S520567

toto9262, 2021-12-21:
Would like to add support for the wiser roller shutter. 
I added an external converter that works. My js : 
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const definition = {
    zigbeeModel: ['NHPB/SHUTTER/1'],
    model: 'S520567',
    vendor: 'Schneider Electric',
    description: 'Volet_SDB',
        fromZigbee: [fz.cover_position_tilt, fz.command_cover_close, fz.command_cover_open, fz.command_cover_stop],
        toZigbee: [tz.cover_position_tilt, tz.cover_state, tz.command_cover_stop],
        exposes: [e.cover_position()],
        meta: {coverInverted: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1) || device.getEndpoint(5);
            await reporting.bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
            await reporting.currentPositionLiftPercentage(endpoint);
module.exports = definition;

Information about the device + link
Have both zigbee and bluetooth functionality.
[S520567 - Wiser Odace - interrupteur volet-roulant - 4A - Bluetooth - blanc - Professionnels | Schneider Electric France](https://www.se.com/fr/fr/product/S520567/wiser-odace-interrupteur-voletroulant-4a-bluetooth-blanc/)
data/database.db entry of the device

{"id":23,"type":"Router","ieeeAddr":"0x000d6ffffe086a67","nwkAddr":45670,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Mains (single phase)","modelId":"NHPB/SHUTTER/1","epList":[5,21],"endpoints":{"5":{"profId":260,"epId":5,"devId":514,"inClusterList":[0,3,4,5,258,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0,"dateCode":""}},"closuresWindowCovering":{"attributes":{"57360":0,"57363":0,"currentPositionLiftPercentage":100}}},"binds":[{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x00212effff0666f3","endpointID":1}],"configuredReportings":[{"cluster":258,"attrId":8,"minRepIntval":1,"maxRepIntval":62000,"repChange":1}],"meta":{}},"21":{"profId":260,"epId":21,"devId":260,"inClusterList":[0,3,2821,65303],"outClusterList":[3,5,6,8,25,258],"clusters":{"genBasic":{"attributes":{"modelId":"NHPB/SHUTTER/1","manufacturerName":"Schneider Electric","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":5,"hwVersion":0,"dateCode":"","swBuildId":"001.005.000R"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":5,"hwVersion":0,"dateCode":"","swBuildId":"001.005.000 R","zclVersion":2,"interviewCompleted":true,"meta":{"configured":-2032335871},"lastSeen":1640119479840}

Koenkk, 2021-12-22:
Added. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-22 by Koenkk

toto9262, 2021-12-22:
Thanks a lot 👍

toto9262, 2022-06-20:
Hi Koenkk

Can you add lift_duration to the roller shutter S520567.
This external converter works fine.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    lift_duration: {
        key: ['lift_duration'],
        convertSet: async (entity, key, value, meta) => {
            await entity.write(0x0102, {0xe000: {value, type: 0x21}}, {manufacturerCode: 0x105e});
            return {state: {lift_duration: value}};
        },
    }
}

const definition = {
    zigbeeModel: ['NHPB/SHUTTER/1'],
    model: 'S520567',
    vendor: 'Schneider Electric',
    description: 'Roller shutter',
    fromZigbee: [fz.cover_position_tilt],
    toZigbee: [tz.cover_position_tilt, tz.cover_state, tzLocal.lift_duration],
    exposes: [e.cover_position(), exposes.numeric('lift_duration', ea.STATE_SET).withUnit('seconds').withValueMin(0).withValueMax(300).withDescription('Duration of lift')],
    meta: {coverInverted: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(5);
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
        await reporting.currentPositionLiftPercentage(endpoint);
    },
};

Thanks.
Have a nice day.

Koenkk, 2022-06-21:
@toto9262 done

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

gonzo09, 2023-05-07:
Thanks @Koenkk  for your amazing job !

I don't understand how to invert cover status. 
For some devices, the state is always reversed from reality. I tried to check/uncheck the invert_cover parameter and refresh the cover position but the state is not updated. So if I click on close it opens and vice versa.
Is there something I didn't figure out ? Or do have to invert wires on the switch ?