Koenkk/zigbee2mqtt #10307: Hue Resonate 1746447P7

daballiemo, 2021-12-22:
Hello.

I migrated from ZHA to zigbee2mqtt recently and after a lot of migration activities I was not able to add my porchlight. 
The porchlight, a [Hue Resonate 1746447P7](https://www.philips-hue.com/en-gb/p/hue-white-and-colour-ambiance-resonate-outdoor-wall-light/1746447P7), is what it looks like a copy of the [Hue Resonate 1746430P7](https://www.philips-hue.com/en-gb/p/hue-white-and-colour-ambiance-resonate-outdoor-wall-light/1746430P7), only with a steel exterior and other light form. I was hoping that using the dev version of zigbee2MQTT was able to support this light, as described in this other [new device request](https://github.com/koenkk/zigbee2mqtt/issues/8860), but it did not.

I am using the Sonoff ZBBridge with Tasmota 10.1 and the 1.22.1 version of zigbee2MQTT.

It appears in device list of Zigbee2MQTT as follow:

Friendly name
buitenlamp voordeur
Last seen
N/A
avaliability.avaliability
Disabled
Device type
Router
Zigbee Model
1746447P7
Zigbee Manufacturer
Philips
Support status
Not supported
IEEE Address
0x00178801067cf514
Network address
0x6B70
Firmware build date
20201012
Firmware version
1.76.10
Power
Interview completed
True

Have I to go through the process of adding a new device or can I copy / paste something ?

TIA

Han

SHxKM, 2021-12-22:
> and other light form

What does this mean?

The fact that the `1746430P7` was in the `dev` branch at the time that issue was open does not mean that `1746447P7` is in `dev` now. This is probably why it's not working for you. However, [writing your own converter is pretty easy](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html).

In your case it would look something like [this](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/philips.js#L1456-L1464). You'd also do everyone using Z2m a service if you renamed the description of `1746430P7` to indicate it's black, while adding yours as a silver variant.

daballiemo, 2021-12-22:
Another light form -> the black edition has squar angles, the silver one (mine) has rounded angles ;-)

I made a file 1746447P7.js that I placed alongside configuration.yaml and included a referral to this file inside configuration.yaml

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const hueExtend = {
    light_onoff_brightness: (options={}) => ({
        ...extend.light_onoff_brightness(options),
        toZigbee: extend.light_onoff_brightness(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp(options),
        toZigbee: extend.light_onoff_brightness_colortemp(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_color: (options={}) => ({
        ...extend.light_onoff_brightness_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_color({supportsHS: true, ...options}).toZigbee
            .concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp_color: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options})
            .toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
};
module.exports = [
    {
        zigbeeModel: ['1746447P7'],
        model: '1746447P7',
        vendor: 'Philips',
        description: 'Hue outdoor Resonate wall lamp',
        meta: {turnsOffAtBrightness1: true},
        extend: hueExtend.light_onoff_brightness_colortemp_color(),
    },
];

han@homeautomation:/docker/zigbee2mqtt_sonoff_usb3_dongle/data$ nano 1746447P7.js
han@homeautomation:/docker/zigbee2mqtt_sonoff_usb3_dongle/data$ cat 1746447P7.js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const hueExtend = {
    light_onoff_brightness: (options={}) => ({
        ...extend.light_onoff_brightness(options),
        toZigbee: extend.light_onoff_brightness(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp(options),
        toZigbee: extend.light_onoff_brightness_colortemp(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_color: (options={}) => ({
        ...extend.light_onoff_brightness_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_color({supportsHS: true, ...options}).toZigbee
            .concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp_color: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options})
            .toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
};
module.exports = [
    {
        zigbeeModel: ['1746447P7'],
        model: '1746447P7',
        vendor: 'Philips',
        description: 'Hue outdoor Resonate wall lamp - Silver',
        meta: {turnsOffAtBrightness1: true},
        extend: hueExtend.light_onoff_brightness_colortemp_color(),
    },
];`

Koenkk, 2021-12-22:
Added it, could you also provide the color temp range? (https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

daballiemo, 2021-12-22:
color temp range reports the following:

```
Zigbee2MQTT:info  2021-12-22 18:39:06: MQTT publish: topic 'zigbee2mqtt_sonoff_usb3_dongle/buitenlamp voordeur', payload '{"brightness":84,"color":{"h":25,"hue":25,"s":95,"saturation":95,"x":0.5267,"y":0.4133},"color_mode":"color_temp","color_temp":500,"linkquality":12,"state":"ON"}'
```

closed 2021-12-22 by Koenkk