Koenkk/zigbee2mqtt #10308: communication with CC2530 - only reflash helps

englishman-hu, 2021-12-22:
## What happened

Communication with adapter fails after several hours of run time.
Restart/reboot does not help. 
It appears, that the adapter firmware gets somehow corrupted, because the only way to get it working again is reflashing the adapter. It solves the problem temporarily, but comes back after few hours.

The system has worked for years, and the issue started with upgrading to 1.22.1.

## What did you expect to happen

## How to reproduce it (minimal and precise)

Upgrade Zigbee2MQTT to 1.22.1
Flash adapter firmware Z-Stack_Home_1.2 20211115/20211116
Run

## Debug info
Zigbee2MQTT version: 1.22.1 (rpi zero, docker)
Adapter hardware:  CC2530
Adapter firmware version: Z-Stack_Home_1.2 20211115/20211116

log entries around failure: [https://pastebin.com/gFKNkE0Z](url)

log entries after restart: [https://pastebin.com/PMRhV4gt](url)

log entries after reflash: [https://pastebin.com/zLwDFr20](url)

Koenkk, 2021-12-24:
> The system has worked for years, and the issue started with upgrading to 1.22.1.

What is the last known working version? Can you start at z2m 1.22.1 and go one version back until the problem appears?

englishman-hu, 2021-12-25:
I did the exercise and the failure kept on popping up with earlier releases and it turned out the the upgrade was just a coincidence and not the root cause.
The issue is most likely caused by a device recently joined to the network (Moes BHT-002).
Having removed it eliminated the failure and the system has been running OK for 48+ hours.

However that device is handled inconsistently by different Z2M instances, for which I may open another issue.

closed 2021-12-25 by englishman-hu