Koenkk/zigbee2mqtt #10312: Philips Hue Dimmer Switch - support for double, triple clicks

Shoomaan, 2021-12-22:
## Is your feature request related to a problem? Please describe
Until Not I've been using ZHA and had the possibility to use double and triple clicks with the dimmer.
Right now I am facing the Problem, that in zigbee2mqtt this is not supported with this switch.

## Describe the solution you'd like

Add support for double and triple clicks like with the Aqara switch. The switch supports it (worked in ZHA) 
zigbee2mqtt supports it (it works with Aqara)

## Describe alternatives you've considered
-

## Additional context
If any Information is needed, I will do my best do help as good as possible.

ant-ds, 2021-12-24:
When looking at the state of a dimmer, there is actually a counter that is being reported which gives you the amount of clicks. Right now it isn't exposed.
`{
    "battery": 25,
    "brightness": 255,
    "counter": 2,
    "linkquality": 61,
    "update": {
        "state": "idle"
    },
    "update_available": false,
    "action": "",
    "duration": 0
}`

closed 2022-02-01 by github-actions[bot]

nioakeim, 2022-07-28:
I would really love to see this coming Z2M. I was using ZHA for a year and this was supported. i had available up to 4x clicks for each button. I switched to Z2M for better device support, and the only device with "limited" support is now my hue dimmer switch.

I am willing to test and troubleshoot to help this development out. If anyone is willing to assist in the coding side let me know

kibuan, 2022-10-09:
I would also like this feature.

p13rr37, 2023-01-04:
Feature is still missing and dearly missed.

domosgabor, 2023-01-07:
Feature is there, just set legacy mode to TRUE in specific settings. It is set to false by default, so you have to make the adjustment. Please note that the structure of the payload will change, but you will be able to capture the COUNTER that can be used for automation.

ecto1a2003, 2023-04-24:
> Feature is there, just set legacy mode to TRUE in specific settings. It is set to false by default, so you have to make the adjustment. Please note that the structure of the payload will change, but you will be able to capture the COUNTER that can be used for automation.

Hi, I don't have that option under specific settings, am I missing something?

domosgabor, 2023-04-25:
> Hi, I don't have that option under specific settings, am I missing something?

Hi, please check the attached screenshot.

[image]

ecto1a2003, 2023-04-25:
This is what I have,
[image]

domosgabor, 2023-04-25:
Strange... 
- what Zigbee2MQTT version do you use?
- what type of coordinator do you have?
- what's your coordinator revision?

You can find all info in the settings --> about section.

Here is mine:
[image]

ecto1a2003, 2023-04-25:
[image]

domosgabor, 2023-04-25:
It seems that something was not ok with your latest screenshot, I cannot download it.

ecto1a2003, 2023-04-25:
[image]

domosgabor, 2023-04-25:
Unfortunately I do not have any experience with the Conbee coordinator.
My guess is that is does not support device specific configuration from the frontend.  I would give give it a try to add  this line to the z2m devices.yaml file below the device:
legacy: true
I you add it don't forget to restart zigbee2mqtt and check the payload.

ecto1a2003, 2023-04-25:
If that file doesn't exist should I create it? I only have the configuration.yaml file

ecto1a2003, 2023-04-26:
All I see is configuration.yanl, database.db and state.json

kibuan, 2023-04-26:
I also do not have the legacy option in the "Settings (specific)" section for my Dimmer switch. I have the Sonoff ZBDongle-P Coordinator.

domosgabor, 2023-04-26:
Do you have this model? If I remember correctly this is version 1 and as I know there is a newer version.
https://www.zigbee2mqtt.io/devices/324131092621.html#philips-324131092621

kibuan, 2023-04-26:
I have the newer version now. But I did have that old version when I first commented on this issue. The old version I had stopped responding after installing a OTA firmware update.

kibuan, 2023-04-26:
It's this one: https://www.zigbee2mqtt.io/devices/929002398602.html#philips-929002398602

domosgabor, 2023-04-26:
Ok, I've just found that the new version does not support double/tripple clicks and there is no legacy mode. :(

kibuan, 2023-04-26:
Okay, sorry I didn't check that, but as I wrote I had the old version when I first posted in this thread. Thank you for your time looking into this.

ecto1a2003, 2023-04-26:
I have the 9290023986,  I had double and triple click functionality via zha

leetNightshade, 2024-01-14:
> I have the 9290023986, I had double and triple click functionality via zha

Same here, I have the latest Philips dimmer and in ZHA it supports double and triple clicks, but for some reason Z2M does not. This **is** a Z2M problem, not a device problem afaik.

SashaBusinaro, 2024-01-25:
Same here, the problem has to be z2m. With ZHA everything was flawless

wernerb, 2024-02-08:
Double click/counter etc does not seem to be a native zigbee feature of the older remote. You can see here the code that provides the multi tap feature just for this dimmer switch.

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/legacy.ts#L3655

The non-legacy new remote does not use this code and so does not have these attributes.

The timeout configured is 250 milliseconds. I think what can be done best is to create an home assistant automation that does the same.

Mar1usW3, 2024-04-03:
i have the "old" 324131092621 remote but if i set the legacy mode to true the counter will not increase. Any idea?

techhitnz, 2024-04-10:
Same here (ex ZHA user)

morettigiorgio, 2024-04-16:
I have just migrated from ZHA to Z2M, and now I have the same problem. 
[image]
[image]
Has anyone managed to solve this problem?
Thanks

Mar1usW3, 2024-04-16:
my solution was to create a automation that store the amount of clicks and do stuff.

HelixX23, 2024-12-05:
I have three 929002398602 Hue Dimmer Switch...
https://www.zigbee2mqtt.io/devices/929002398602.html

there is also no double nor tripple click event :(
How do i activate the "counter" value?

glassbase, 2024-12-10:
Found this thread after migrating. Same issue as everyone else. Had the multiple clicks action in ZHA, not in Z2M.

myevit, 2024-12-14:
Same! Need double click support!

Bastiencc, 2025-01-10:
I used this on version 1.4 and it worked fine. I don't know how to convert this automation to 2.0, any idea?

```
- id: chambre_amis_switch_mur_on_press_on
  triggers:
    - trigger: mqtt
      topic: 'zigbee2mqtt/chambre_amis_switch_chevet'
  conditions:
    - condition: template
      value_template: '{{ "on-press" == trigger.payload_json.action }}'
    - condition: template
      value_template: '{{ 2 == trigger.payload_json.counter|int }}'
  actions:
    (...)
```

Bastiencc, 2025-01-10:
create a mqtt sensor in HA :

```
mqtt:
  sensor:
    - name: "switch_hue_counter"
      unique_id: bb0c5bd6110d41968f986ad6b1c57c87
      state_topic: "zigbee2mqtt/switch_hue"
      value_template: "{{ value_json.counter | lower }}"
```

Bastiencc, 2025-01-10:
> my solution was to create a automation that store the amount of clicks and do stuff.

can you share you code please? thanks !

Mar1usW3, 2025-01-19:
> > my solution was to create a automation that store the amount of clicks and do stuff.
> 
> can you share you code please? thanks !

yes sure...
You need also a counter helper and configured it this way:
Minimum Value: 0
Maximum Value: 2
Initial Value: 0
Step Size: 1

```
alias: Esszimmer - Schalter
description: ""
mode: single
triggers:
  - domain: mqtt
    device_id: xxxx
    type: action
    subtype: on_press
    id: on_press
    trigger: device
  - domain: mqtt
    device_id: xxxx
    type: action
    subtype: off_press
    id: off_press
    trigger: device
  - domain: mqtt
    device_id: xxxx
    type: action
    subtype: up_press
    id: up_press
    trigger: device
  - domain: mqtt
    device_id: xxxx
    type: action
    subtype: down_press
    id: down_press
    trigger: device
  - entity_id:
      - counter.esszimmer_schalter_counter
    to: "1"
    for:
      hours: 0
      minutes: 0
      seconds: 0
      milliseconds: 350
    id: counter_1
    trigger: state
  - entity_id:
      - counter.esszimmer_schalter_counter
    to: "2"
    for:
      hours: 0
      minutes: 0
      seconds: 0
      milliseconds: 350
    id: counter_2
    trigger: state
conditions: []
actions:
  - choose:
      - conditions:
          - condition: trigger
            id:
              - on_press
        sequence:
          - metadata: {}
            data: {}
            target:
              entity_id: counter.esszimmer_schalter_counter
            action: counter.increment
      - conditions:
          - condition: trigger
            id:
              - off_press
        sequence:
          - metadata: {}
            data:
              transition: 1
            target:
              entity_id: light.esszimmertisch
            action: light.turn_off
      - conditions:
          - condition: trigger
            id:
              - counter_1
        sequence:
          - metadata: {}
            data: {}
            target:
              entity_id: counter.esszimmer_schalter_counter
            action: counter.reset
          - metadata: {}
            target:
              entity_id: scene.esstisch_an
            data:
              transition: 1
            action: scene.turn_on
      - conditions:
          - condition: trigger
            id:
              - counter_2
        sequence:
          - metadata: {}
            data: {}
            target:
              entity_id: counter.esszimmer_schalter_counter
            action: counter.reset
          - metadata: {}
            data:
              transition: 1
            target:
              entity_id: scene.esstisch_gemutlich
            action: scene.turn_on
      - conditions:
          - condition: trigger
            id:
              - up_press
        sequence:
          - metadata: {}
            data:
              brightness_step: 25
            target:
              entity_id: light.esszimmertisch
            alias: Increase Brightness
            action: light.turn_on
      - conditions:
          - condition: trigger
            id:
              - down_press
        sequence:
          - metadata: {}
            data:
              brightness_step_pct: -25
            target:
              entity_id: light.esszimmertisch
            alias: Decrease Brightness
            action: light.turn_on
```
This is working fine for me since years. I hope this helps