Koenkk/zigbee2mqtt #10313: Replacing coordinator ( CC2652R to CC2652R)

miljbee, 2021-12-22:
Hello,

I am trying to switch my adapter without having to repair. The old one and the new one are the same, CC2652R from slaesh.
My first try was just to unplug the old one and replug the new one. At first it seemed to work, I was receiving all the updates from my temp sensors. But I quickly realized that commands sent to my switches weren't working - timeout. Occasionally some command would pass, but most of them timed out.

So I read some docs and realize that the ieeAddress from the new adapter should match the old one. After digging around, I found zigpy-znp, and used it to backup/restore the conf from the nvram.

The result was unexpected, the logs says "adapter state does not match configuration", **and report the ieeeAddress in the extpanid**

After that, using zigpy-znp, I reset the adpater conf, and restart Zigbee2MQTT. No errors in logs, receiving sensors data, but unable to send commands to switches or plugs.

If you are wondering, why I am trying this, It's because, my ZigBee2Mqtt is inside a virtual machine, that I would like to be able to move between hosts.

# Bug Report
## What happened
I have a perfectly working ZigBee Network. The coordinator is a CC2652R from slaesh
Using zigpy_znp, I backed up the nvram from the coordinator and restored it to a new CC2652R from slaesh
ZigBee2MQTT won't start after pluging the new CC2652R, and the logs are not clear:
(stage-1) adapter is configured
(stage-2) adapter state does not match configuration
(stage-3) got adapter backup
(stage-4) adapter state matches backup

Configuration is not consistent with adapter state/backup!
PAN ID: configured=xxxx, adapter=xxxx (ok)
**Extended PAN ID: configured=00124bxxxxxxxxxx, adapter=dddddddddddddddd (configured = ieeeAddress :o)**
Network Key: configured=mykey, adapter=mykey (ok)
Channel List: configured=15, adapter=15 (ok)
**Please update configuration to prevent further issues. (<= How can I do that ? )**
If you wish to re-commission your network, please remove coordinator backup at /config/zigbee2mqtt/coordinator_backup.json.
Re-commissioning your network will require re-pairing of all devices!
Error while starting zigbee-herdsman
Failed to start zigbee
...
## What did you expect to happen
Zigbee2MQTT working fine

## How to reproduce it (minimal and precise)
Backup restore the NVRAM config of a CC2652R to a new one with this command
python -m zigpy_znp.tools.network_backup /dev/serial/by-id/old_radio -o network_backup.json
python -m zigpy_znp.tools.network_restore /dev/serial/by-id/new_radio -i network_backup.json
plug the new dongle to the Zigbe2MQTT Host, and start it

## Debug Info
Zigbee2MQTT version: 1.22.1-1
Adapter hardware: CC2652R
Adapter firmware version: Sorry, don't know how to read it

Koenkk, 2021-12-23:
I don't know the zigpy tool, but the backup/restore might not be compatible with z2m.

I recommend to:
- Reflash your adapter, but before this write the old adapters ieeeAddr to the new one (you can e.g. use https://github.com/xyzroe/ZigStarGW-MT)
- Put back your old `coordinator_backup.json` in the data folder
- Start z2m
- If you don't have your old `coordinator_backup.json` you have to repair all devices.

miljbee, 2022-01-03:
Ok, thanks for the guidance, I think I managed to set the IEEEE Address on the new adapter, but it was a pain for several reasons.
- ZigStarGW-MT doesn't work on my windows computer. Verifying after flashing always reports a CRC mismatch
- From what I have understood, there is no way to change the "Primary" IEEE Address, what is possible is adding the old IEEE as a secondary address. Unfortunately, it seems cc2538_bsl.py reads only the primary. So after setting the secondary address, reading still reports the primary, which is misleading.
- After being able to set the old IEEE and flash using cc2538_bsl.py, I found that the behaviour of my zigbee network with the new adapter was the same than before flashing: I get reports from (nearly) all my devices, but can't set states. I read somewhere that repowering devices might be required, which I did, which fixed the set state problem.
- I have 2 aqara buttons and 1 vibrating sensor, which needed to be re-paired

Finally, I don't really know what to conclude of all of this. The doc states that the IEEE Address should be set, but it seems that it makes no difference, Setting IEEE Address in windows is a pain. Having to re-pair some devices is not great.

I would have liked to be able to easily switch the coordinator, I will do without this.

Anyway, thanks for your support

closed 2022-02-10 by github-actions[bot]