Koenkk/zigbee2mqtt #10318: Schneider Electric Roller Shutter CCT5015-0001

toto9262, 2021-12-22:
Would like to add support for the wiser roller shutter (the last ;) ).
I added an external converter that works. My js :
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const definition = {
    zigbeeModel: ['PUCK/SHUTTER/1'],
    model: 'CCT5015-0001',
    vendor: 'Schneider Electric',
    description: 'Volet_SB',

        fromZigbee: [fz.cover_position_tilt, fz.command_cover_close, fz.command_cover_open, fz.command_cover_stop],
        toZigbee: [tz.cover_position_tilt, tz.cover_state, tz.command_cover_stop],
        exposes: [e.cover_position()],
        meta: {coverInverted: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1) || device.getEndpoint(5);
            await reporting.bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
            await reporting.currentPositionLiftPercentage(endpoint);
   },
};
module.exports = definition;

Information about the device + link
Have both zigbee and bluetooth functionality.
[https://www.se.com/fr/fr/product/CCT5015-0001/wiser-micromodule-encastr%C3%A9-voletroulant/](url)

database.db : 
{"id":11,"type":"Router","ieeeAddr":"0x000d6ffffec78c96","nwkAddr":9000,"manufId":4190,"manufName":"Schneider Electric",
"powerSource":"Mains (single phase)","modelId":"PUCK/SHUTTER/1","epList":[5],"endpoints":{"5":{"profId":260,"epId":5,
"devId":514,"inClusterList":[0,3,4,5,258,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"PUCK/SHUTTER/1",
"appVersion":1,"stackVersion":5,"hwVersion":1,"dateCode":"","swBuildId":"001.005.003 R"}}},"binds":[{"cluster":258,"type":"endpoint",
"deviceIeeeAddress":"0x00212effff0666f3","endpointID":1}],"configuredReportings":[{"cluster":258,"attrId":8,"minRepIntval":1,
"maxRepIntval":62000,"repChange":1}],"meta":{}}},"appVersion":1,"stackVersion":5,"hwVersion":1,"dateCode":"","swBuildId":"001.005.003 R","zclVersion":2,"interviewCompleted":true,"meta":{"configured":-321561689},"lastSeen":1640208126132}

Thank you

Koenkk, 2021-12-23:
added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-23 by Koenkk

toto9262, 2021-12-26:
Thx, works very well