Koenkk/zigbee2mqtt #10321: Livarno Lux E27 bulb RGB HG06106C does not works correctly

canassa, 2021-12-22:
## What happened

These settings are not working:
- Color changing does not work (always white)
- Temperature color strange behavior (not correspond what expected: 1000k is cool, 3000k is warm ..)

Turning on/off and brightness works fine.

## Debug info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: Sonoff zigbee adapter (CC2652P + CP2102N)
Adapter firmware version: Pre-flashed with Z-Stack 3.x.0 coordinator firmware

Koenkk, 2021-12-23:
Mede Helmonder 🎉 

First make sure you are on the latest z2m. 
If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

canassa, 2021-12-23:
@Koenkk A quick question,

While the HG06106C RGB bulbs are working now, I noticed the same problem when I tried to pair a Livarno Lux smart LED light strip 2.5m HG06104A. I am on version  1.22.1 now, shall I update to the edge?

Koenkk, 2021-12-23:
`HG06104A` uses the same definition `HG06106C` (which matches the Zigbee spec) so I'm not sure why it doesn't work. @szporwolik  does this device work correctly for you on z2m 1.22.1?