Koenkk/zigbee2mqtt #10324: Philips 9290030171 is unsupported

mithek, 2021-12-23:
I'm wondering if it's my fault or if the model is maybe newer from Philips and has to be added to the devices.js

## What happened
My Philips hue wall switch with model: 9290030171 paired but got the message unsupported.
It did recognize the zigbeeModel: RDM001

## What did you expect to happen
That the device paired successfully and got recognized.

## How to reproduce it (minimal and precise)
Pair a Philips hue wall switch with model: 9290030171

## supported Devices.js 
The device with a slight different model number is supported in the devices.js
```
 {
        zigbeeModel: ['RDM001'],
        model: '929003017102',
        vendor: 'Philips',
        description: 'Hue wall switch module',
        fromZigbee: [fz.battery, fz.hue_wall_switch_device_mode, fz.hue_wall_switch],
        exposes: [e.battery(), e.action(['left_press', 'left_press_release', 'right_press', 'right_press_release']),
            exposes.enum('device_mode', ea.ALL, ['single_rocker', 'single_push_button', 'dual_rocker', 'dual_push_button'])],
        toZigbee: [tz.hue_wall_switch_device_mode],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff', 'manuSpecificPhilips']);
            await reporting.batteryPercentageRemaining(endpoint);
            const options = {manufacturerCode: 0x100B, disableDefaultResponse: true};
            await endpoint.write('genBasic', {0x0034: {value: 0, type: 48}}, options);
        },
    },
```

## Debug info
Home Assistant: 2021.12.4
Zigbee2MQTT is installed via Home Assistant 
Zigbee2MQTT version: 1.18.1.1-1
Adapter hardware: CC2652RB
Adapter firmware version:

Koenkk, 2021-12-23:
Update to the latest z2m version first.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

mithek, 2021-12-29:
Thank you, that solved it!

closed 2021-12-29 by mithek