Koenkk/zigbee2mqtt #10340: 'TS011F' and manufacturer name '_TZ3000_typdpbpg'

calouis, 2021-12-24:
### Link

https://fr.aliexpress.com/item/1005002112376679.html?spm=a2g0o.9042311.0.0.78786c37j9c7wH

### Database entry

MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c138fbb43c04bf","ieee_address":"0xa4c138fbb43c04bf","status":"successful","supported":false},"type":"device_interview"}' MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0xa4c138fbb43c04bf","supported":false},"type":"pairing"}' Zigbee: disabling joining new devices. MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"slrm9-2"}'

### Comments

I purchased following plug : https://fr.aliexpress.com/item/1005002112376679.html?spm=a2g0o.9042311.0.0.78786c37j9c7wH
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

1.22.1-dev commit: 61bd95f8

`Debug Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"slrm9-1","value":true}'
Info Zigbee: allowing new devices to join.
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"slrm9-1"}'
Info Device '0xa4c138fbb43c04bf' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138fbb43c04bf","ieee_address":"0xa4c138fbb43c04bf"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c138fbb43c04bf"},"type":"device_connected"}'
Info Starting interview of '0xa4c138fbb43c04bf'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138fbb43c04bf","ieee_address":"0xa4c138fbb43c04bf","status":"started"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c138fbb43c04bf"},"type":"pairing"}'
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS011F"}' from endpoint 1 with groupID null
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ3000_typdpbpg"}' from endpoint 1 with groupID null
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID null
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'readResponse', cluster 'genBasic', data '{"appVersion":68}' from endpoint 1 with groupID null
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID null
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID null
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID null
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID null
Debug Skipping message, definition is undefined and still interviewing
Info Successfully interviewed '0xa4c138fbb43c04bf', device has successfully been paired
Warning Device '0xa4c138fbb43c04bf' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c138fbb43c04bf","ieee_address":"0xa4c138fbb43c04bf","status":"successful","supported":false},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0xa4c138fbb43c04bf","supported":false},"type":"pairing"}'
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,0]}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'raw', cluster '57344', data '{"data":[8,42,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'raw', cluster '57344', data '{"data":[8,43,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'attributeReport', cluster 'genOnOff', data '{"20480":1,"32768":0,"moesStartUpOnOff":0,"offWaitTime":0,"onOff":1,"onTime":0,"tuyaBacklightMode":1}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53248":1,"53249":0,"53250":2297791,"53251":138,"53252":12276,"53253":1566,"53264":0,"switchType":0}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"slrm9-2","value":false}'
Info Zigbee: disabling joining new devices.
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"slrm9-2"}'`

`        "0xa4c138fbb43c04bf": {
[... 25 lines cut ...]
                "242": {
                    "bindings": [],
                    "clusters": {
                        "input": [],
                        "output": [
                            "greenPower"
                        ]
                    },
                    "configured_reportings": [],
                    "scenes": []
                }
            },
            "friendly_name": "0xa4c138fbb43c04bf",
            "ieee_address": "0xa4c138fbb43c04bf",
            "interview_completed": true,
            "interviewing": false,
            "manufacturer": "_TZ3000_typdpbpg",
            "model_id": "TS011F",
            "network_address": 36680,
            "power_source": "Mains (single phase)",
            "supported": false,
            "type": "Router"
        }`

Regards

Koenkk, 2021-12-24:
added 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-24 by Koenkk

calouis, 2021-12-28:
Sorry 
Version Zigbee2MQTT
1.22.1-dev commit: 1e75dd56

`Debug Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"d20o3-5","value":true}'
Info Zigbee: allowing new devices to join.
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"d20o3-5"}'
Info Device '0xa4c138fbb43c04bf' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138fbb43c04bf","ieee_address":"0xa4c138fbb43c04bf"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c138fbb43c04bf"},"type":"device_connected"}'
Info Starting interview of '0xa4c138fbb43c04bf'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138fbb43c04bf","ieee_address":"0xa4c138fbb43c04bf","status":"started"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c138fbb43c04bf"},"type":"pairing"}'
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID null
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID null
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID null
Debug Skipping message, definition is undefined and still interviewing
Info Successfully interviewed '0xa4c138fbb43c04bf', device has successfully been paired
Warning Device '0xa4c138fbb43c04bf' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c138fbb43c04bf","ieee_address":"0xa4c138fbb43c04bf","status":"successful","supported":false},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0xa4c138fbb43c04bf","supported":false},"type":"pairing"}'
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"d20o3-6","value":false}'
Info Zigbee: disabling joining new devices.
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"d20o3-6"}'
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,0]}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'raw', cluster '57344', data '{"data":[8,96,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'raw', cluster '57344', data '{"data":[8,97,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'attributeReport', cluster 'genOnOff', data '{"20480":1,"32768":0,"moesStartUpOnOff":0,"offWaitTime":0,"onOff":1,"onTime":0,"tuyaBacklightMode":1}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53248":1,"53249":0,"53250":2297791,"53251":138,"53252":12276,"53253":1566,"53264":0,"switchType":0}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0xa4c138fbb43c04bf', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_typdpbpg'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Saving state to file /opt/zigbee2mqtt/data/state.json
Debug Saving state to file /opt/zigbee2mqtt/data/state.json`

i don't know what to do now

Regards

Koenkk, 2021-12-29:
If you are not running z2m through HA addon or docker container, did you do a `npm ci` in the z2m dir after updating?

Snoozy01, 2022-01-16:
TS011F **_TZ3000_typdpbpg** works for me (ON/OFF + linkquality)
but I can't get any current metrics and even if it's advertised for it, I don't know if the current metering is working. Do you know if  can get access to these information ?

Here is the device purchased (Zigbee version):
[https://www.aliexpress.com/item/1005003646930652.html?spm=a2g0o.9042311.0.0.1e284c4dxjfipR](url)

Used solution:
I've added the following to my Z2M configuration file:

```
external_converters:
 - TS011F.js
```

and have added TS011F.js in the same folder:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    fingerprint: [
        {
            modelID: 'TS011F',
            manufacturerName: '_TZ3000_typdpbpg'
        },
    ],
    model: 'TS011F',
    vendor: '_TZ3000_typdpbpg', 
    description: 'Zigbee plug',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [e.switch()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;

````

Koenkk, 2022-01-17:
Without using any external converter, how is the plug identified in the frontend? As TS011F_plug_3 or TS011F_plug_1?

Snoozy01, 2022-01-17:
> Without using any external converter, how is the plug identified in the frontend? As TS011F_plug_3 or TS011F_plug_1?

I tried the 2 external converters (plug_1 and _3) but it wasn't working. Z2M was even not starting. maybe due to an error from my side. I'm quite a new bee.

I've never seen any of these 2 models in logs. I just removed external converter and see the following (log level debug enabled):

`
info MQTT publish: topic 'zigbee2mqtt/Light - Aquarium', payload '{"state":"OFF"}' 

debug Received Zigbee message from 'Light - Aquarium', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,8]}' from endpoint 1 with groupID null

warning Received message from unsupported device with Zigbee **model 'TS011F'** and **manufacturer name '_TZ3000_typdpbpg'**
warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
`

Snoozy01, 2022-01-18:
Update: with the following configuration of external converter, it runs but do report only few metrics : 

> "energy":0.09
> "linkquality":183
> "power_outage_memory":"restore"
> ,"state":"ON"

but nothing related to:

> "current":null
> "power":null
> "voltage":null

and I don't know what to do with :

> cluster 'genTime', data '["localTime"]'

log extract:

> debug Received Zigbee message from 'Light - Aquarium', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,9]}' from endpoint 1 with groupID null
> info MQTT publish: topic 'zigbee2mqtt/Light - Aquarium', payload '{"current":null,"energy":0.09,"linkquality":183,"power":null,"power_outage_memory":"restore","state":"ON","voltage":null}'
> debug Received Zigbee message from 'Light - Aquarium', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID null

converter:
[TS011F.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/7886926/TS011F.js.txt)

[image]

Snoozy01, 2022-01-18:
when I change the power_outage_memory it gives the following answer in log:

> debug Received Zigbee message from 'Light - Aquarium', type 'raw', cluster '57345', data '{"data":[8,235,10,16,208,48,0],"type":"Buffer"}' from endpoint 1 with groupID null

Snoozy01, 2022-01-20:
ok thanks. I tested with the latest-dev, I see in the log it works from Z2M but it do not bring the metering attributes into home-assistant and I can't find why. It seems like these attributes are not exposed.

I'm with a docker setup. I used the following image: 
`    image: koenkk/zigbee2mqtt:latest-dev
`
I read in doc the notion of Home Assistant addon with the use of the `edge `version. but don't understand. What does it mean?

I also tried to replicate the tuya.js as an external converter but can't make it working (error msg at start regarding `measurement_poll_interval.`

I'll try again later. any guidance would be appreciated.

Snoozy01, 2022-01-20:
I cleaned my setup (force remove device in home assistant, regenerate the docker image...)
with 
`image: koenkk/zigbee2mqtt:latest-dev`
I got the following result (every 60 seconds):
`info MQTT publish: topic 'zigbee2mqtt/Light - Aquarium', payload '{"current":0.1,"energy":0.23,"linkquality":199,"power":15,"power_outage_memory":null,"state":"ON","voltage":231}'`

but I confirm in home assistant I have only 1 entity corresponding to the switch (nothing about current, voltage...):
[image]

Koenkk, 2022-01-20:
What ha version are you runnign?

Snoozy01, 2022-01-20:
docker :
image: homeassistant/home-assistant:latest

**Home Assistant 2021.8.8**
<html><body>

Version | core-2021.8.8
-- | --
Installation Type | Home Assistant Container
Development | false
Supervisor | false
Docker | true

</body>
</html>

Snoozy01, 2022-01-20:
in the payload, I can see last values updated. Could it be related to the `value_template: '{{ value_json.state }}'` ? Here I don't know what to do.

Payload:

current: 0.13
energy: 0.23
linkquality: 175
power: 18
power_outage_memory: null
state: 'ON'
voltage: 233

Snoozy01, 2022-01-20:
HA log is showing an error:

Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'device': {'identifiers': ['zigbee2mqtt_0xa4c13855699c5ca4'], 'manufacturer': 'TuYa', 'model': 'Smart plug (with power monitoring by polling) (TS011F_plug_3)', 'name': 'Light - Aquarium'}, 'device_class': 'voltage', 'enabled_by_default': False, 'entity_category': 'diagnostic', 'json_attributes_topic': 'zigbee2mqtt/Light - Aquarium', 'name': 'Light - Aquarium voltage', 'state_class': 'measurement', 'state_topic': 'zigbee2mqtt/Light - Aquarium', 'unique_id': '0xa4c13855699c5ca4_voltage_zigbee2mqtt', 'unit_of_measurement': 'V', 'value_template': '{{ value_json.voltage }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 180, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: extra keys not allowed @ data['entity_category']

Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'device': {'identifiers': ['zigbee2mqtt_0xa4c13855699c5ca4'], 'manufacturer': 'TuYa', 'model': 'Smart plug (with power monitoring by polling) (TS011F_plug_3)', 'name': 'Light - Aquarium'}, 'device_class': 'energy', 'enabled_by_default': True, 'json_attributes_topic': 'zigbee2mqtt/Light - Aquarium', 'name': 'Light - Aquarium energy', 'state_class': 'total_increasing', 'state_topic': 'zigbee2mqtt/Light - Aquarium', 'unique_id': '0xa4c13855699c5ca4_energy_zigbee2mqtt', 'unit_of_measurement': 'kWh', 'value_template': '{{ value_json.energy }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 180, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: value must be one of ['measurement'] for dictionary value @ data['state_class']

Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'device': {'identifiers': ['zigbee2mqtt_0xa4c13855699c5ca4'], 'manufacturer': 'TuYa', 'model': 'Smart plug (with power monitoring by polling) (TS011F_plug_3)', 'name': 'Light - Aquarium'}, 'enabled_by_default': False, 'entity_category': 'config', 'icon': 'mdi:power-settings', 'json_attributes_topic': 'zigbee2mqtt/Light - Aquarium', 'name': 'Light - Aquarium power outage memory', 'state_topic': 'zigbee2mqtt/Light - Aquarium', 'unique_id': '0xa4c13855699c5ca4_power_outage_memory_zigbee2mqtt', 'value_template': '{{ value_json.power_outage_memory }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 180, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: extra keys not allowed @ data['entity_category']

Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'device': {'identifiers': ['zigbee2mqtt_0xa4c13855699c5ca4'], 'manufacturer': 'TuYa', 'model': 'Smart plug (with power monitoring by polling) (TS011F_plug_3)', 'name': 'Light - Aquarium'}, 'enabled_by_default': False, 'entity_category': 'diagnostic', 'icon': 'mdi:signal', 'json_attributes_topic': 'zigbee2mqtt/Light - Aquarium', 'name': 'Light - Aquarium linkquality', 'state_class': 'measurement', 'state_topic': 'zigbee2mqtt/Light - Aquarium', 'unique_id': '0xa4c13855699c5ca4_linkquality_zigbee2mqtt', 'unit_of_measurement': 'lqi', 'value_template': '{{ value_json.linkquality }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 180, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_b [... cut ...] e "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: extra keys not allowed @ data['entity_category']

Snoozy01, 2022-01-21:
I upgraded to the last version. and it seems to work fine. a big THANK YOU for your support.

Snoozy01, 2022-01-21:
I was thinking you're doing an amazing work ! I'm impressed.

tenkaibr, 2024-11-15:
Does indicator_mode work for you? It doesn't work here. How should I proceed?