Koenkk/zigbee2mqtt #10341: [Problem]: Stuck at loading - Not loading MQTT Server - Not loading configmap (kubernetes)

OlafKocanda, 2021-12-24:
### What happened?

I am trying to run zigbee2mqtt on a kubernetes deployment on my rpi4 (k3s) with mosquitto as a MQTT broker.
Mosquitto is deployed with a service with port 1883 and named mosquitto in my <namespace>. To adapt the MQTT Server variable, I tried adding a configmap at the location /config. 

However, it seems like the configmap is not loaded as:
1. zigbee2mqtt is stuck at the stage "Connecting to MQTT server at mqtt://mosquitto" with a wrong MQTT Server
2. when looking inside the pod, the config at the location /config is not the expected config but rather seems to be a standard file.

Another short term solution would be to change the config at /config but I cannot edit it as it seems to be Read-Only.
Following this, is there a way to add a variable for startup, similar to the "ZIGBEE2MQTT_CONFIG" variable?

The config at /config looks like this:
` 
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# Enable frontend
frontend:
  port: 8080

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: "mqtt://mosquitto"
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0
  # Also possible to use the tcp://
  # port: tcp://192.168.1.10:23

# Map devices to persistent /data directory
devices: /data/devices.yaml
groups: /data/groups.yaml

advanced:
  log_output: ["console"]`

### What did you expect to happen?

My configmap is loaded into zigbee2mqtt at the place /config and the new MQTT server is part of the configuration.

### How to reproduce it (minimal and precise)

My deployment looks like the following:
`---
apiVersion: apps/v1
kind: Deployment
metadata:
  creationTimestamp: null
  labels:
    app: zigbee2mqtt
  name: zigbee2mqtt
spec:
  replicas: 1
  selector:
    matchLabels:
      app: zigbee2mqtt
  strategy:
    type: Recreate
  template:
    metadata:
      creationTimestamp: null
      labels:
        app: zigbee2mqtt
    spec:
      containers:
        - env:
            - name: TZ
              value: Europe/Berlin
            - name: ZIGBEE2MQTT_CONFIG
              value: "/config"
          image: koenkk/zigbee2mqtt
          name: zigbee2mqtt
          securityContext:
            privileged: true
          ports:
            - containerPort: 8080
          resources: {}
          volumeMounts:
            - name: zigbee2mqtt-data
              mountPath: /app/data
            - name: zigbee2mqtt-udev
              mountPath: /run/udev
            - name: zigbee2mqtt-config
              mountPath: /config
              subPath: configuration.yaml
            - mountPath: /dev/ttyACM0
              name: ttyacm
      volumes:
        - name: zigbee2mqtt-data
          hostPath:
            path: /data/zigbee2mqtt
        - name: zigbee2mqtt-udev
          hostPath:
            path: /run/udev
        - name: ttyacm
          hostPath:
            path: /dev/ttyACM0
        - name: zigbee2mqtt-config
          configMap:
            name: zigbee2mqtt-config`

and the configmap is: 
`---
apiVersion: v1
kind: ConfigMap
metadata:
  name: zigbee2mqtt-config
  namespace: homeassistant
  labels:
    app: zigbee2mqtt
data:
  configuration.yaml: |-

    # Let new devices join our zigbee network
    permit_join: true
    # Docker-Compose makes the MQTT-Server available using "mqtt" hostname
    mqtt:
      base_topic: zigbee2mqtt
      server: mqtt://mosquitto.<namespace>:1883
    # Zigbee Adapter path
    serial:
      port: /dev/ttyUSB0
    # Enable the Zigbee2MQTT frontend
    frontend:
      port: 8080
    # Let Zigbee2MQTT generate a new network key on first start
    advanced:
      network_key: GENERATE`

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20210000

### Adapter

Conbee II

### Debug log

`kubectl -n <namespace> logs deployment/zigbee2mqtt -f
Using '/app/data' as data directory
Zigbee2MQTT:info  2021-12-24 11:21:27: Logging to console only'
Zigbee2MQTT:info  2021-12-24 11:21:28: Starting Zigbee2MQTT version 1.22.1 (commit #c2b5229)
Zigbee2MQTT:info  2021-12-24 11:21:28: Starting zigbee-herdsman (0.13.176)
CREATED DECONZ ADAPTER
Zigbee2MQTT:info  2021-12-24 11:21:28: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2021-12-24 11:21:28: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:info  2021-12-24 11:21:28: Currently 0 devices are joined:
Zigbee2MQTT:warn  2021-12-24 11:21:28: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2021-12-24 11:21:28: Allowing new devices to join.
Zigbee2MQTT:warn  2021-12-24 11:21:28: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2021-12-24 11:21:28: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2021-12-24 11:21:29: Connecting to MQTT server at mqtt://mosquitto`

lux73, 2021-12-24:
i don't think your double quotes are right

try `server: 'mqtt://mosquitto'`

btw if you ping mosquitto it resolves the right IP to your mosquitto Server?

OlafKocanda, 2021-12-24:
Thanks for your answer! My configuration is not even loaded / mounted
The config with the double quotes is an auto-generated one which is not the one I intend to load.

So even one step before your solution of changing the server as I cannot change the configuration 
1. config is not loaded
2. /config file is Read-Only
3. I have not found a variable to change the MQTT Server at the start (similar to the "ZIGBEE2MQTT_CONFIG" variable)

Koenkk, 2021-12-24:
> I have not found a variable to change the MQTT Server at the start (similar to the "ZIGBEE2MQTT_CONFIG" variable)

You can set any config value with environment variables, see https://www.zigbee2mqtt.io/guide/configuration/#environment-variables

OlafKocanda, 2022-01-21:
I figured it out after a lot of sweat and tears... Check out my deployment files here 
https://github.com/Koenkk/zigbee2mqtt/discussions/10899

closed 2022-01-21 by OlafKocanda