Koenkk/zigbee2mqtt #10348: [New device support]: Neo Alarm/Temperature/Humidity NAS-AB02B2

elperepat, 2021-12-25:
### Link

http://www.szneo.com/en/products/show.php?id=241

### Database entry

{"id":18,"type":"Router","ieeeAddr":"0xa4c138e870adce69","nwkAddr":30599,"manufId":4417,"manufName":"_TZE200_t1blo2bj","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"߾X)f߾X)\u0012","65506":48,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_t1blo2bj","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc3b7d","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1640396986513,"useImplicitCheckin":true} {"id":19,"type":"Router","ieeeAddr":"0xa4c1384ee1333979","nwkAddr":53407,"manufId":4417,"manufName":"_TZE200_t1blo2bj","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u001f�X)f\u001f�X)\u0012","65506":48,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_t1blo2bj","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc3b7d","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1640399001813,"useImplicitCheckin":true}

### Comments

I just bought 3 alarms on [Aliexpress](https://www.aliexpress.com/item/1005003626375818.html). I thought they would be AB02B0 but I can't get them to work. I find out they are AB02B2.

Here an extract from my external converter:

`
fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_t1blo2bj'
        },
`

[image]

Here is the output from the debug log:

`
Device '0xa4c1384ee1333979' announced itself

MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1384ee1333979","ieee_address":"0xa4c1384ee1333979"},"type":"device_announce"}'

MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xa4c1384ee1333979"},"type":"device_announced"}'

Retrieving state of '0xa4c1384ee1333979' after reconnect

Received Zigbee message from '0xa4c1384ee1333979', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,87,17,0,26,66],"type":"Buffer"}' from endpoint 1 with groupID 0

Received Zigbee message from '0xa4c1384ee1333979', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":5,"fn":0,"status":0,"transid":26}' from endpoint 1 with groupID 0

Received Zigbee message from '0xa4c1384ee1333979', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,10],"type":"Buffer"},"datatype":2,"dp":7,"fn":0,"status":0,"transid":27}' from endpoint 1 with groupID 0

Received Zigbee message from '0xa4c1384ee1333979', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15,"fn":0,"status":0,"transid":28}' from endpoint 1 with groupID 0

Received Zigbee message from '0xa4c1384ee1333979', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[5],"type":"Buffer"},"datatype":4,"dp":21,"fn":0,"status":0,"transid":29}' from endpoint 1 with groupID 0

Received Zigbee message from '0xa4c1384ee1333979', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,87,17,0,26,66],"type":"Buffer"}' from endpoint 1 with groupID 0

Received Zigbee message from '0xa4c1384ee1333979', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,87,17,0,26,66],"type":"Buffer"}' from endpoint 1 with groupID 0

Received Zigbee message from '0xa4c1384ee1333979', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0

Failed to configure '0xa4c1384ee1333979', attempt 2 (Error: Bind 0xa4c1384ee1333979/1 genBasic from '0x00124b0021cc3b7d/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7))

Received Zigbee message from '0xa4c1384ee1333979', type 'attributeReport', cluster 'genBasic', data '{"65506":48,"65508":0,"appVersion":68}' from endpoint 1 with groupID 0

Configuring '0xa4c1384ee1333979'

Successfully configured '0xa4c1384ee1333979'

Received Zigbee message from '0xa4c1384ee1333979', type 'attributeReport', cluster 'genBasic', data '{"65503":"��X)f��X)\u0012"}' from endpoint 1 with groupID 0`

I just don't know what to do from here...

Koenkk, 2021-12-25:
Not sure what the difference with `NAS-AB02B0` is, but I've added it to the `NAS-AB02B0`, please check if it works in the latest dev branch.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

elperepat, 2021-12-25:
Device pairs successfully, but not exposed data is shown.

[image]

Here is the log when pairing:

`info  2021-12-25 11:41:16: Device '0xa4c1380d515bf03c' joined
info  2021-12-25 11:41:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1380d515bf03c","ieee_address":"0xa4c1380d515bf03c"},"type":"device_joined"}'
info  2021-12-25 11:41:16: Configuring '0xa4c1380d515bf03c'
info  2021-12-25 11:41:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c1380d515bf03c"},"type":"device_connected"}'
info  2021-12-25 11:41:16: Starting interview of '0xa4c1380d515bf03c'
info  2021-12-25 11:41:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1380d515bf03c","ieee_address":"0xa4c1380d515bf03c","status":"started"},"type":"device_interview"}'
info  2021-12-25 11:41:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c1380d515bf03c"},"type":"pairing"}'
debug 2021-12-25 11:41:16: Device '0xa4c1380d515bf03c' announced itself
info  2021-12-25 11:41:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1380d515bf03c","ieee_address":"0xa4c1380d515bf03c"},"type":"device_announce"}'
info  2021-12-25 11:41:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xa4c1380d515bf03c"},"type":"device_announced"}'
debug 2021-12-25 11:41:17: Received Zigbee message from '0xa4c1380d515bf03c', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
debug 2021-12-25 11:41:17: Received Zigbee message from '0xa4c1380d515bf03c', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
debug 2021-12-25 11:41:17: Received Zigbee message from '0xa4c1380d515bf03c', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
info  2021-12-25 11:41:17: Successfully interviewed '0xa4c1380d515bf03c', device has successfully been paired
info  2021-12-25 11:41:17: Device '0xa4c1380d515bf03c' is supported, identified as: Neo Temperature & humidity sensor and alarm (NAS-AB02B0)
info  2021-12-25 11:41:17: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature & humidity sensor and alarm","exposes":[{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":3,"name":"humidity_alarm","property":"humidity_alarm","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Indicates if the battery of this device is almost empty","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"access":3,"name":"temperature_alarm","property":"temperature_alarm","type":"binary","value_off":false,"value_on":true},{"access":3,"name":"alarm","property":"alarm","type":"binary","value_off":false,"value_on":true},{"access":3,"name":"melody","property":"melody","type":"enum","values":["1","2","3","4","5","6","7","8","9","10","11","12","13","14","15","16","17","18"]},{"access":3,"name":"duration","property":"duration","type":"numeric","unit":"second","value_max":1000,"value_min":0},{"access":3,"name":"temperature_min","property":"temperature_min","type":"numeric","unit":"°C","value_max":35,"value_min":-10},{"access":3,"name":"temperature_max","property":"temperature_max","type":"numeric","unit":"°C","value_max":35,"value_min":-10},{"access":3,"name":"humidity_min","property":"humidity_min","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":3,"name":"humidity_max","property":"humidity_max","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":3,"name":"volume","property":"volume","type":"enum","values":["low","medium","high"]},{"access":1,"name":"power_type","property":"power_type","type":"enum","values":["battery_full","battery_high","battery_medium","battery_low","usb"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"NAS-AB02B0","options":[],"supports_ota":false,"vendor":"Neo"},"friendly_name":"0xa4c1380d515bf03c","ieee_address":"0xa4c1380d515bf03c","status":"successful","supported":true},"type":"device_interview"}'
info  2021-12-25 11:41:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Temperature & humidity sensor and alarm","friendly_name":"0xa4c1380d515bf03c","model":"NAS-AB02B0","supported":true,"vendor":"Neo"},"type":"pairing"}'
debug 2021-12-25 11:41:22: Received Zigbee message from '0xa4c1380d515bf03c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":5,"fn":0,"seq":4608}' from endpoint 1 with groupID 0
warn  2021-12-25 11:41:22: Unhandled DP #5: {"seq":4608,"dp":5,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[2]}}
debug 2021-12-25 11:41:23: Received Zigbee message from '0xa4c1380d515bf03c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,10],"type":"Buffer"},"datatype":2,"dp":7,"fn":0,"seq":4864}' from endpoint 1 with groupID 0
[... cut ...]
debug 2021-12-25 11:41:23: Received Zigbee message from '0xa4c1380d515bf03c', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
info  2021-12-25 11:41:24: Successfully configured '0xa4c1380d515bf03c'
debug 2021-12-25 11:41:29: Received Zigbee message from '0xa4c1380d515bf03c', type 'attributeReport', cluster 'genBasic', data '{"65506":48,"65508":0,"appVersion":68}' from endpoint 1 with groupID 0
debug 2021-12-25 11:41:29: Received Zigbee message from '0xa4c1380d515bf03c', type 'attributeReport', cluster 'genBasic', data '{"65503":"ֿY)fֿY)\u0012"}' from endpoint 1 with groupID 0
`

Koenkk, 2021-12-26:
1. Can you try with the converter below (see https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#instructions how to add this converter)
2. In the frontend make sure the converter is loaded correctly (description should contain `CUSTOM`)
3. Are any temperature and humidity values reported now?
4. If not, it seems this device works differently then the NAS-AB02B0, you will need to have a TuYa gateway and sniff to figure out how to correctly configure this device (https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531)

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_t1blo2bj'}],
    model: 'NAS-AB02B2',
    vendor: 'Neo',
    description: 'Temperature & humidity sensor and alarm CUSTOM',
    fromZigbee: [fz.temperature, fz.humidity],
    toZigbee: [],
    exposes: [e.temperature(), e.humidity()],
    onEvent: tuya.onEventSetLocalTime,
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['msTemperatureMeasurement', 'msRelativeHumidity'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.temperature(endpoint);
        await reporting.humidity(endpoint);
    },
};

module.exports = definition;
```

elperepat, 2021-12-27:
Just for the benefit of others, you forgot the const declaration for tuya in the above js file:

`const tuya = require("zigbee-herdsman-converters/lib/tuya");`

It doesn't work:

[image]

I ordered a CC2531 and a tuya gateway. When I get a chance to sniff the network I'll come back with the results.

Just in case you need it, here is the log output:

`Dec 26 19:50:29  npm[2230367]: Zigbee2MQTT:info  2021-12-26 19:50:29: Device '0xa4c1380d515bf03c' joined`

`Dec 26 19:50:29  npm[2230367]: Zigbee2MQTT:info  2021-12-26 19:50:29: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1380d515bf03c","ieee_address":"0xa4c1380d515bf03c"},"type":"device_joined"}'`

`Dec 26 19:50:29  npm[2230367]: Zigbee2MQTT:info  2021-12-26 19:50:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c1380d515bf03c"},"type":"device_connected"}'`

`Dec 26 19:50:29  npm[2230367]: Zigbee2MQTT:debug 2021-12-26 19:50:29: Device '0xa4c1380d515bf03c' announced itself`

`Dec 26 19:50:29  npm[2230367]: Zigbee2MQTT:info  2021-12-26 19:50:29: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1380d515bf03c","ieee_address":"0xa4c1380d515bf03c"},"type":"device_announce"}'`
`
Dec 26 19:50:29  npm[2230367]: Zigbee2MQTT:info  2021-12-26 19:50:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xa4c1380d515bf03c"},"type":"device_announced"}'`

`Dec 26 19:50:31  npm[2230367]: Zigbee2MQTT:debug 2021-12-26 19:50:31: Retrieving state of '0xa4c1380d515bf03c' after reconnect`

`Dec 26 19:50:35  npm[2230367]: Zigbee2MQTT:debug 2021-12-26 19:50:35: Received Zigbee message from '0xa4c1380d515bf03c', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,102,17,0,14,66],"type":"Buffer"}' from endpoint 1 with groupID 0`

`Dec 26 19:50:35  npm[2230367]: Zigbee2MQTT:debug 2021-12-26 19:50:35: Received Zigbee message from '0xa4c1380d515bf03c', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":5,"fn":0,"status":0,"transid":14}' from endpoint 1 with groupID 0`

`Dec 26 19:50:36  npm[2230367]: Zigbee2MQTT:debug 2021-12-26 19:50:36: Received Zigbee message from '0xa4c1380d515bf03c', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,10],"type":"Buffer"},"datatype":2,"dp":7,"fn":0,"status":0,"transid":15}' from endpoint 1 with groupID 0`

`Dec 26 19:50:36  npm[2230367]: Zigbee2MQTT:debug 2021-12-26 19:50:36: Received Zigbee message from '0xa4c1380d515bf03c', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15,"fn":0,"status":0,"transid":16}' from endpoint 1 with groupID 0`

`Dec 26 19:50:36  npm[2230367]: Zigbee2MQTT:debug 2021-12-26 19:50:36: Received Zigbee message from '0xa4c1380d515bf03c', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[5],"type":"Buffer"},"datatype":4,"dp":21,"fn":0,"status":0,"transid":17}' from endpoint 1 with groupID 0`

`Dec 26 19:50:36  npm[2230367]: Zigbee2MQTT:debug 2021-12-26 19:50:36: Received Zigbee message from '0xa4c1380d515bf03c', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,102,17,0,14,66],"type":"Buffer"}' from endpoint 1 with groupID 0`

`Dec 26 19:50:36  npm[2230367]: Zigbee2MQTT:debug 2021-12-26 19:50:36: Received Zigbee message from '0xa4c1380d515bf03c', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,102,17,0,14,66],"type":"Buffer"}' from endpoint 1 with groupID 0`

`Dec 26 19:50:37  npm[2230367]: Zigbee2MQTT:debug 2021-12-26 19:50:37: Received Zigbee message from '0xa4c1380d515bf03c', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0`

`Dec 26 19:50:37  npm[2230367]: Zigbee2MQTT:debug 2021-12-26 19:50:37: Received Zigbee message from '0xa4c1380d515bf03c', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0`

`Dec 26 19:50:44  npm[2230367]: Zigbee2MQTT:debug 2021-12-26 19:50:44: Received Zigbee message from '0xa4c1380d515bf03c', type 'attributeReport', cluster 'genBasic', data '{"65506":48,"65508":0,"appVersion":68}' from endpoint 1 with groupID 0`

`Dec 26 19:50:44  npm[2230367]: Zigbee2MQTT:debug 2021-12-26 19:50:44: Received Zigbee message from '0xa4c1380d515bf03c', type 'attributeReport', cluster 'genBasic', data '{"65503":"ǃ[)\u0013�[)e�[)\u0012"}' from endpoint 1 with groupID 0`

nighthawkii, 2022-01-03:
Hello,

I was in the same situation as I received this device also 10d ago. This device has no Temperature nor Humidity sensors.

I followed the guide to discover the DataPoints, by using a Tuya gateway, and I updated the converters  files.
Now it works, I am able to control fully the Alarm :)
n.b: only thing I am not sure is the exposition of the " e.battery_low(),"

Here are the modifications performed:

**file "node_modules/zigbee-herdsman-converters/converters/lib/tuya.js"**
location: after "    // Neo T&H"
```
    // Neo AlarmOnly
    neoAOBattPerc: 15,
    neoAOMelody: 21,
    neoAODuration: 7,
    neoAOAlarm: 13,
    neoAOVolume: 5,
```

**node_modules/zigbee-herdsman-converters/devices/neo.js** 
```
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_t1blo2bj'}],
        zigbeeModel: ['1blo2bj'],
        model: 'NAS-AB02B2',
        vendor: 'Neo',
        description: 'Alarm',
        fromZigbee: [fz.neo_alarm, fz.ignore_basic_report],
        toZigbee: [tz.neo_alarm],
        exposes: [
            e.battery_low(),
            exposes.binary('alarm', ea.STATE_SET, true, false),
            exposes.enum('melody', ea.STATE_SET, Array.from(Array(18).keys()).map((x)=>(x+1).toString())),
            exposes.numeric('duration', ea.STATE_SET).withUnit('second').withValueMin(0).withValueMax(1800),
            exposes.enum('volume', ea.STATE_SET, ['low', 'medium', 'high']),
            exposes.numeric('battpercentage', ea.STATE_SET).withUnit('%').withValueMin(0).withValueMax(100),
        ],
        onEvent: tuya.onEventSetLocalTime,
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.command('manuSpecificTuya', 'dataQuery', {});
            await endpoint.command('manuSpecificTuya', 'mcuVersionRequest', {'seq': 0x0002});
        },
    },
```

**node_modules/zigbee-herdsman-converters/converters/fromZigbee.js**
```
    neo_alarm: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataReport', 'commandDataResponse'],
        convert: (model, msg, publish, options, meta) => {
            const dp = msg.data.dp;
            const value = tuya.getDataValue(msg.data.datatype, msg.data.data);

            switch (dp) {
            case tuya.dataPoints.neoAOAlarm: // 0x13 [TRUE,FALSE]
                return {alarm: value};
            case tuya.dataPoints.neoAODuration: // 0x7 [0,0,0,10] duration alarm in second
                return {duration: value};
            case tuya.dataPoints.neoAOBattPerc: // 0x15 [0,0,0,100] battery percentage
                return {battpercentage: value};
            case tuya.dataPoints.neoAOMelody: // 0x21 [5] Melody
                return {melody: value};
            case tuya.dataPoints.neoAOVolume: // 0x5 [0]/[1]/[2] Volume 0-max, 2-low
                return {volume: {2: 'low', 1: 'medium', 0: 'high'}[value]};
            default: // Unknown code
                meta.logger.warn(`Unhandled DP #${dp}: ${JSON.stringify(msg.data)}`);
            }
        },
    },

```

**node_modules/zigbee-herdsman-converters/converters/toZigbee.js**
```
    neo_alarm: {
        key: [
            'alarm', 'melody', 'volume', 'duration',
        ],
        convertSet: async (entity, key, value, meta) => {
            switch (key) {
            case 'alarm':
                await tuya.sendDataPointBool(entity, tuya.dataPoints.neoAOAlarm, value);
                break;
            case 'melody':
                await tuya.sendDataPointEnum(entity, tuya.dataPoints.neoAOMelody, parseInt(value, 10));
                break;
            case 'volume':
                await tuya.sendDataPointEnum(
                    entity,
                    tuya.dataPoints.neoAOVolume,
                    {'low': 2, 'medium': 1, 'high': 0}[value]);
                break;
            case 'duration':
                await tuya.sendDataPointValue(entity, tuya.dataPoints.neoAODuration, value);
                break;
            default: // Unknown key
                throw new Error(`Unhandled key ${key}`);
            }
        },
    },

```

elperepat, 2022-01-04:
> This device has no Temperature nor Humidity sensors.

I'm not home right now, but I remember seeing them sold as temperature and humidity sensor.

Indeed: https://www.globalsources.com/Wireless-intruder/wireless-intruder-alarms-1187144522p.htm

When I tried to make it work, I saw the log populated with regular updates with a value changing. I thought it was the temperature/humidity report.

I'll crack it open tomorrow to see if I can spot a sensor under a microscope.

nighthawkii, 2022-01-04:
> I'm not home right now, but I remember seeing them sold as temperature and humidity sensor.
>
> Indeed: https://www.globalsources.com/Wireless-intruder/wireless-intruder-alarms-1187144522p.htm
>
> When I [...]

There are 2 versions : 1 cheap without sensors which is this one (around 14€), and 1 with sensors (around 20€).
I bought it on aliexpress : https://a.aliexpress.com/_mNAEB2s
In the description is written "alarm on humidity and temp" but not on the pictures.
Sellers are mixing the 2 variants.
I opened a dispute under aliexpress due to that.
The only values changing regularly are the signal strength, and if you are on battery (not usb), the battery percentage.

ririck0, 2022-01-11:
Hi

@elperepat, have you ever sniff the communication between your device and the Tuya gateway ? I tried  @nighthawkii 's modification but it's not working for me.
[image]

I'm totally lost between multiple neo references (NAS-AN02B2 with or without sensors). Is there finally a way to know which AB02B2 we got ?

Thanks,

nighthawkii, 2022-01-11:
If you open it and see the same version than elperepat and myself (NAS-AB02B2 v2) then the modifications I made shall work.
You can confirm also the version you have by observing the debug data when you start or interview the device : you should have "TZE200_t1blo2bj".
Can you show the debug data relevant to the device ? it will allow to see if the DataPoints are the same or not

ririck0, 2022-01-11:
I have open it and saw on the PCB "NAS-AB02B2 v2".

Please find below the logs during pairing:
`zigbee2mqtt      | Zigbee2MQTT:info  2022-01-11 15:18:59: Device '0xa4c1380b2a3f5489' joined
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-11 15:18:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1380b2a3f5489","ieee_address":"0xa4c1380b2a3f5489"},"type":"device_joined"}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-11 15:18:59: Configuring '0xa4c1380b2a3f5489'
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-11 15:18:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c1380b2a3f5489"},"type":"device_connected"}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-11 15:18:59: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c1380b2a3f5489/battery_low/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c1380b2a3f5489/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c1380b2a3f5489"],"manufacturer":"Neo","model":"Alarm (NAS-AB02B2)","name":"0xa4c1380b2a3f5489"},"device_class":"battery","entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0xa4c1380b2a3f5489","name":"0xa4c1380b2a3f5489 battery low","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xa4c1380b2a3f5489","unique_id":"0xa4c1380b2a3f5489_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-11 15:18:59: MQTT publish: topic 'homeassistant/switch/0xa4c1380b2a3f5489/switch_alarm/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c1380b2a3f5489/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xa4c1380b2a3f5489/set/alarm","device":{"identifiers":["zigbee2mqtt_0xa4c1380b2a3f5489"],"manufacturer":"Neo","model":"Alarm (NAS-AB02B2)","name":"0xa4c1380b2a3f5489"},"json_attributes_topic":"zigbee2mqtt/0xa4c1380b2a3f5489","name":"0xa4c1380b2a3f5489 alarm","payload_off":"false","payload_on":"true","state_topic":"zigbee2mqtt/0xa4c1380b2a3f5489","unique_id":"0xa4c1380b2a3f5489_switch_alarm_zigbee2mqtt","value_template":"{% if value_json.alarm %} true {% else %} false {% endif %}"}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-11 15:18:59: MQTT publish: topic 'homeassistant/sensor/0xa4c1380b2a3f5489/melody/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c1380b2a3f5489/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c1380b2a3f5489"],"manufacturer":"Neo","model":"Alarm (NAS-AB02B2)","name":"0xa4c1380b2a3f5489"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:music-note","json_attributes_topic":"zigbee2mqtt/0xa4c1380b2a3f5489","name":"0xa4c1380b2a3f5489 melody","state_topic":"zigbee2mqtt/0xa4c1380b2a3f5489","unique_id":"0xa4c1380b2a3f5489_melody_zigbee2mqtt","value_template":"{{ value_json.melody }}"}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-11 15:18:59: MQTT publish: topic 'homeassistant/select/0xa4c1380b2a3f5489/melody/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c1380b2a3f5489/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xa4c1380b2a3f5489/set/melody","device":{"identifiers":["zigbee2mqtt_0xa4c1380b2a3f5489"],"manufacturer":"Neo","model":"Alarm (NAS-AB02B2)","name":"0xa4c1380b2a3f5489"},"entity_category":"config","icon":"mdi:music-note","json_attributes_topic":"zigbee2mqtt/0xa4c1380b2a3f5489","name":"0xa4c1380b2a3f5489 melody","options":["1","2","3","4","5","6","7","8","9","10","11","12","13","14","15","16","17","18"],"state_topic":"zigbee2mqtt/0xa4c1380b2a3f5489","unique_id":"0xa4c1380b2a3f5489_melody_zigbee2mqtt","value_template":"{{ value_json.melody }}"}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-11 15:18:59: MQTT publish: topic 'homeassistant/sensor/0xa4c1380b2a3f5489/duration/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c1380b2a3f5489/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c1380b2a3f5489"],"manufacturer":"Neo","model":"Alarm (NAS-AB02B2)","name":"0xa4c1380b2a3f5489"},"enabled_by_default":false,"json_attributes_topic":"zigbee2mqtt/0xa4c1380b2a3f5489","name":"0xa4c1380b2a3f5489 duration","state_topic":"zigbee2mqtt/0xa4c1380b2a3f5489","unique_id":"0xa4c1380b2a3f5489_duration_zigbee2mqtt","unit_of_measurement":"second","value_template":"{{ value_json.duration }}"}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-11 15:18:59: MQTT publish: topic 'homeassistant/number/0xa4c1380b2a3f5489/duration/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c1380b2a3f5489/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xa4c1380b2a3f5489/set/duration","device":{"identifiers":["zigbee2mqtt_0xa4c1380b2a3f5489"],"manufacturer":"Neo","model":"Alarm (NAS-AB02B2)","name":"0xa4c1380b2a3f5489"},"json_attributes_topic":"zigbee2mqtt/0xa4c1380b2a3f5489","max":1800,"min":0,"name":"0xa4c1380b2a3f5489 duration","state_topic":"zigbee2mqtt/0xa4c1380b2a3f5489","unique_id":"0xa4c1380b2a3f5489_duration_zigbee2mqtt","unit_of_measurement":"second","value_template":"{{ value_json.duration }}"}'
[... cut ...]
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-11 15:19:21: MQTT publish: topic 'zigbee2mqtt/0xa4c1380b2a3f5489', payload '{"alarm":null,"battery_low":null,"battpercentage":null,"duration":null,"last_seen":"2022-01-11T14:19:21.318Z","linkquality":60,"melody":null,"volume":null}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-11 15:19:21: MQTT publish: topic 'zigbee2mqtt/0xa4c1380b2a3f5489', payload '{"alarm":null,"battery_low":null,"battpercentage":null,"duration":null,"last_seen":"2022-01-11T14:19:21.370Z","linkquality":60,"melody":null,"volume":null}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-11 15:19:21: Successfully configured '0xa4c1380b2a3f5489'
`

nighthawkii, 2022-01-11:
So you have the same device.
Are you sure you performed the modifications in all the files and you restarted Z2M ? 
Only difference on my side (I don't know if it may have an impact), is that I am using Domoticz, not Home Assistant. But it should not impact the Z2M, and not the frontend.

Make a double check on the 4 files, take care also to the paths (tuya.js exists in several locations)

ririck0, 2022-01-12:
I've patched these files:

`└── node_modules
    └── zigbee-herdsman-converters
        ├── converters
        │   ├── fromZigbee.js
        │   ├── lib
        │   │   └── tuya.js
        │   └── toZigbee.js
        └── devices
            └── neo.js`

ririck0, 2022-01-14:
@nighthawkii Have I done something wrong ?

nighthawkii, 2022-01-14:
In the debug, you have the error : 
zigbee2mqtt | Zigbee2MQTT:error 2022-01-11 15:19:06: Exception while calling fromZigbee converter: Cannot read property 'datatype' of undefined}

It is a parser issue, the code is not interpreted. 
I am wondering if you did not place the additional code at the end of the file "from Zigbee" and missed a bracket or put an unexpected coma (,)

On my side I inserted the code in FromZigbee and ToZigbee  just after the neo_t_h_alarm, so I respected the declaration like the previous block. 

Just check the syntax, and keep us informed. As it works on my side, no reason it would not work on yours. 

Currently the interview is ok, it is just when the device is sending the data, it crashes due to the issue in the FromZigbee file.

elperepat, 2022-01-15:
I was able to test @nighthawkii patches and they work for my siren.

However, I would change the line in neo.js for battery percentage from 

`exposes.numeric('battpercentage', ea.STATE_SET).withUnit('%').withValueMin(0).withValueMax(100),`
        
to

`exposes.numeric('battpercentage', ea.STATE).withUnit('%').withValueMin(0).withValueMax(100),`

To hide the slider.

nighthawkii, 2022-01-15:
Great it worked for you!
Yes I will try to do asap a pull request to get the support in the next releases!

TheNutHouse84, 2022-01-16:
I'm getting the same error as @ririck0 
Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'datatype')}

I copied and pasted what @nighthawkii posted above and in the same location in the file.

Any suggestions?

ririck0, 2022-01-16:
> In the debug, you have the error : zigbee2mqtt | Zigbee2MQTT:error 2022-01-11 15:19:06: Exception while calling fromZigbee converter: Cannot read property 'datatype' of undefined}
> 
> It is a parser issue, the code is not interpreted. I am wondering if you did not place the additional code at the end of the file "from Zigbee" and missed a bracket or put an unexpected coma (,)
> 
> On my side I inserted the code in FromZigbee and ToZigbee just after the neo_t_h_alarm, so I respected the declaration like the previous block.
> 
> Just check the syntax, and keep us informed. As it works on my side, no reason it would not work on yours.
> 
> Currently the interview is ok, it is just when the device is sending the data, it crashes due to the issue in the FromZigbee file.
> 
> N. B : to test the ToZigbee validity, you can activate the alarm on the front end and check the debug if you have the same message for "To Zigbee"

I have tried to move the code after neo_t_h_alarm part, but get the same error...
Could you please upload your files so I can test with your changes?

Thanks

SimonSezKossel, 2022-01-18:
Great work all!

Can I assist in testing somehow?  Trying to link the same Neo Siren Alarm (without temperature sensor): Zigbee model 'TS0601' and manufacturer name '_TZE200_t1blo2bj'
I have limited programming experience and running z2m edge on homeassistant, tried the latest edge but not recognised (yet).

Regards,

nighthawkii, 2022-01-18:
[Modifs Neo Alarm only.zip](https://github.com/Koenkk/zigbee2mqtt/files/7892352/Modifs.Neo.Alarm.only.zip)

Hi all,

Here are the files attached. I am based on the release "1.22.1-dev commit: 8b4be3d6"
@TheNutHouse84 > I don't think the zigbee model name has an impact, I made the same than the model with T°/Hum, the 1st letter was removed (maybe a mistake in this one).

The battery_low is useless on this device I think

I try to make the pull request, I keep you posted.

nighthawkii, 2022-01-18:
Ok, Pull request is done, and I modified the battpercentage display to get only the number (no more slider).
Hope it will be taken in next release

Koenkk, 2022-01-19:
Added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/3699 by @nighthawkii , will be in the latest-dev in a few hours and in the next release on 1 February

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-19 by Koenkk

SimonSezKossel, 2022-01-20:
Getting the following errors after edge update:

Debug Received MQTT message on 'zigbee2mqtt/0xa4c138e08c5e32f4/set' with data '{"alarm":false}'
Debug Publishing 'set' 'alarm' to '0xa4c138e08c5e32f4'
Debug Received Zigbee message from '0xa4c138e08c5e32f4', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":13}],"seq":43776}' from endpoint 1 with groupID 0
Error Exception while calling fromZigbee converter: Cannot read property 'datatype' of undefined}

Could I even have a different version? I`m going to open my alarm

Koenkk, 2022-01-20:
@SimonSezKossel should be fixed now, can you confirm?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

SimonSezKossel, 2022-01-22:
Yes, great work!

No errors or warnings.
One small thing we could improve but not essential:
The high and low volume setting are inverted. Low setting is the highest volume.

elperepat, 2022-01-22:
Yes, volume is inverted in the dev branch:

File **node_modules/zigbee-herdsman-converters/converters/toZigbee.js**

In section _neo_alarm: {_

should read
                   ` {'low': 0, 'medium': 1, 'high': 2}[value]);`

Koenkk, 2022-01-23:
@elperepat thanks added the fix

KaHeMu, 2022-01-26:
I accidently purchased this 'low cost' version of the NEO NAS-AB020B**2** (reporting manufacturerName: '_TZE200_t1blo2bj') recently and wondered why it was not supported. But with the latest dev-branch pulled today it works perfectly. Thanks a lot! However, the volume button "low" and "high" in the Z2M Frontend are swapped. Sending {"volume":"high"|"low"} works correctly.  Still the documentation and picture on zigbee2mqtt.io are missing. 
But thanks for the good work!

ririck0, 2022-02-02:
Hi,

I guess there is also a modification to do in file **node_modules/zigbee-herdsman-converters/converters/fromZigbee.js**

In section neo_alarm at line 2526
[image]

It should be

`                return {volume: {0: 'low', 1: 'medium', 2: 'high'}[value]};`