Koenkk/zigbee2mqtt #10350: [New device support]: TuYa  TS011F_plug_1 > zigbee _TZ3000_ew3ldmgx

FlexxFR, 2021-12-25:
### Link

https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1

### Database entry

{"id":129,"type":"Router","ieeeAddr":"0x540f57fffe8ebce9","nwkAddr":25381,"manufId":4098,"manufName":"_TZ3000_ew3ldmgx","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_ew3ldmgx","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cbf157","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cbf157","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cbf157","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1640417430454,"useImplicitCheckin":true}

### Comments

Device paired OK, interview completed successfully.
However the device only reports link quality & on/off switch that is working, the power measurements remain N/A.

After searching is it exactly the same issue as reported here:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3107

However, this time the Zigbee Manufacturer is **_TZ3000_ew3ldmgx**

Zigbee Model: TS011F
Zigbee Manufacturer : _TZ3000_ew3ldmgx
Description: Smart plug (with power monitoring)
Manufacturer: TuYa
Model: TS011F_plug_1

I also have similar looking TuYa sockets with subtype _TZ3000_cphmq0q7, these are recognized as TS011F_plug_3 and work correctly. 
Is my issue wrongly detected as a _plug_1 and should it be a _plug_3 (polling) or is is not completely detected right as a plug_1 due the a new zogbee model code?

I have both models so I can do some troubleshooting is required.

Koenkk, 2021-12-25:
`_TZ3000_ew3ldmgx` should already be detected in the latest dev branch (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

FlexxFR, 2021-12-28:
Yeah, seems to work,. it's now detected as _plug_3. 
However, in the configuration menu > reporting > there is still a difference. Doesn't seem to make a difference and maybe it's cause by earlier onboarding. But could also be a detection bug? 

_ew3ldmgx
[image]

_cphmq0q7
[image]

Koenkk, 2021-12-29:
This is expected, plug 3 doesn't support reporting while plug 1 does so this is expected. plug 3 relies on polling instead.

closed 2021-12-29 by Koenkk

rchiileea, 2022-01-28:
plug 1 is still what mine is detected as, but dont report the the values

error on configuration shows

'''
Failed to configure 'Kettle', attempt 2 (Error: Bind 0xa4c138fb649db287/1 haElectricalMeasurement from '0x00212effff0767e9/1' failed (Error: status: 130) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:753:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:522:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/tuya.js:1148:13) at Configure.configure (/app/lib/extension/configure.ts:115:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))
'''

jakobrosenberg, 2022-03-08:
I have a `TS011F_plug_1` which is showing:

```
power
Instantaneous measured power
0W
current
Instantaneous measured electrical current
0A
voltage
Measured electrical potential value
0V
```

My configuration looks like this
```yaml
devices:
  ...
  '0x847127fffe8f7abf':
    friendly_name: '0x847127fffe8f7abf'
    measurement_poll_interval: 60
```
and `zigbee2mqtt` is version `"1.24.0"`

Any feedback on what I can do to get a reading from it?