Koenkk/zigbee2mqtt #10352: [Problem]: OTA update hangs for E1810

Gandem, 2021-12-25:
### What happened?

When trying to upgrade E1810 from `2.3.014` to latest firmware, update hangs and never succeeds.

What I've tried already:
- Remove the battery and put it back
- Remove the device from the network and re-pair it
- Changing the position of the button to put it as close as possible to the adapter

### What did you expect to happen?

The OTA update should start eventually.

### How to reproduce it (minimal and precise)

Using latest `dev` branch and development adapter firmware `20211217` for adapter `SONOFF Zigbee 3.0 USB Dongle Plus`

1. Add to network a E1810 device, with firmware version 2.3.014
2. Request an OTA update by sending an MQTT message to `zigbee2mqtt/bridge/request/device/ota_update/update`
3. From `zigbee2mqtt` perspective, update hangs on:
```
Zigbee2MQTT:debug 2021-12-25 11:14:57: Updating to latest '0x14b457fffe7fc3b4' (TRADFRI remote control)
Zigbee2MQTT:debug 2021-12-25 11:14:57: Using endpoint '1'
```
4. A couple of seconds later, unhandled promise rejection from `zigbee2mqtt-herdsman` perspective:
```
(node:5747) UnhandledPromiseRejectionWarning: Error: Timeout - 41522 - 1 - null - 25 - 1 after 30000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
    at emitUnhandledRejectionWarning (internal/process/promises.js:168:15)
    at processPromiseRejections (internal/process/promises.js:247:11)
    at processTicksAndRejections (internal/process/task_queues.js:96:32)
(node:5747) Error: Timeout - 41522 - 1 - null - 25 - 1 after 30000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
(node:5747) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
    at emitDeprecationWarning (internal/process/promises.js:180:11)
    at processPromiseRejections (internal/process/promises.js:249:13)
    at processTicksAndRejections (internal/process/task_queues.js:96:32)
``` 

Notes:
- The device is paired successfully, and placed right next to the adapter
- Toggle/button clicks/hold are detected correctly and is usable during the whole process

### Zigbee2MQTT version

1.22.1-dev

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

[zigbee2mqtt-log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7775764/zigbee2mqtt-log.txt)

Stamper-repo, 2021-12-26:
Same issue as Gandem. I am using the same adapter. Diving a little deeper into herdsman, I also found these logs which are perhaps related:

```
zigbee2mqtt    | Zigbee2MQTT:debug 2021-12-26 14:24:25: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id": "0x14b457fffe7822f6"}'
zigbee2mqtt    | Zigbee2MQTT:info  2021-12-26 14:24:25: Checking if update available for '0x14b457fffe7822f6'
zigbee2mqtt    | Zigbee2MQTT:debug 2021-12-26 14:24:25: Check if update available for '0x14b457fffe7822f6' (TRADFRI remote control)
zigbee2mqtt    | Zigbee2MQTT:debug 2021-12-26 14:24:25: Using endpoint '1'
zigbee2mqtt    | 2021-12-26T13:24:25.959Z zigbee-herdsman:controller:endpoint CommandResponse 0x14b457fffe7822f6/1 genOta.imageNotify({"payloadType":0,"queryJitter":100}, {"sendWhenActive":true,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
zigbee2mqtt    | 2021-12-26T13:24:25.960Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x14b457fffe7822f6:57583/1 (0,0,1)
zigbee2mqtt    | 2021-12-26T13:24:25.961Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":57583,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":2,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[25,3,0,0,100]}}
zigbee2mqtt    | 2021-12-26T13:24:25.961Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,239,224,1,1,25,0,2,0,30,5,25,3,0,0,100,91]
zigbee2mqtt    | 2021-12-26T13:24:25.967Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt    | 2021-12-26T13:24:25.967Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt    | 2021-12-26T13:24:25.968Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2021-12-26T13:24:25.968Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2021-12-26T13:24:25.968Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2021-12-26T13:24:33.211Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,240,1,2,52]
zigbee2mqtt    | 2021-12-26T13:24:33.212Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,240,1,2,52]
zigbee2mqtt    | 2021-12-26T13:24:33.212Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [240,1,2] - 52
zigbee2mqtt    | 2021-12-26T13:24:33.212Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":240,"endpoint":1,"transid":2}
zigbee2mqtt    | 2021-12-26T13:24:33.212Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2021-12-26T13:24:33.212Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x14b457fffe7822f6:57583,240,0)
zigbee2mqtt    | 2021-12-26T13:24:33.212Z zigbee-herdsman:adapter:zStack:adapter Wait 2000ms
zigbee2mqtt    | 2021-12-26T13:24:35.215Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x14b457fffe7822f6:57583/1 (0,1,1)
zigbee2mqtt    | 2021-12-26T13:24:35.215Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":57583,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":3,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[25,3,0,0,100]}}
zigbee2mqtt    | 2021-12-26T13:24:35.216Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,239,224,1,1,25,0,3,0,30,5,25,3,0,0,100,90]
zigbee2mqtt    | 2021-12-26T13:24:35.222Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt    | 2021-12-26T13:24:35.222Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt    | 2021-12-26T13:24:35.222Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2021-12-26T13:24:35.222Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2021-12-26T13:24:35.222Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2021-12-26T13:24:41.600Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,240,1,3,53]
zigbee2mqtt    | 2021-12-26T13:24:41.601Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,240,1,3,53]
zigbee2mqtt    | 2021-12-26T13:24:41.601Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [240,1,3] - 53
zigbee2mqtt    | 2021-12-26T13:24:41.601Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":240,"endpoint":1,"transid":3}
zigbee2mqtt    | 2021-12-26T13:24:41.601Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2021-12-26T13:24:41.601Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x14b457fffe7822f6:57583,240,1)
zigbee2mqtt    | 2021-12-26T13:24:41.602Z zigbee-herdsman:adapter:zStack:znp:SREQ --> UTIL - assocGetWithAddress - {"extaddr":"0x14b457fffe7822f6","nwkaddr":57583}
zigbee2mqtt    | 2021-12-26T13:24:41.602Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,10,39,74,246,34,120,254,255,87,180,20,239,224,50]
zigbee2mqtt    | 2021-12-26T13:24:41.610Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,36,103,74,239,224,1,0,1,4,0,0,0,1,125,0,0,0,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,126]
[... cut ...]
zigbee2mqtt    |     at listOnTimeout (internal/timers.js:557:17)
zigbee2mqtt    |     at processTimers (internal/timers.js:500:7)
zigbee2mqtt    | (Use `node --trace-warnings ...` to show where the warning was created)
```

I am unsure what these logs mean, but the status 240 on the 'Data confirm error' stand out to me. I have flashed the SONOFF Zigbee adapter to the latest firmware: 
```
Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
```

Gandem, 2021-12-26:
Seeing the same thing as Stamper-repo, the OTA update is working on another device (I've succeded with a TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white). The problem seems to be linked to the E1810 button

rednerd, 2021-12-27:
I'm seeing the same thing with the exact same versions that Gandem posted in the initial report.

mcmellow, 2021-12-28:
When trying to upgrade E1810 from 2.3.014 to latest firmware, update hangs and never succeeds.  5 bulbs (LED1903C5/LED1835C6) are updated OK. They are on 2.3.087 now.  For what it is worth. The FW build date remains the same as before the update (20190312) Al this information is from the Z2M frontend page.

mcmellow, 2022-01-03:
Deleted the E1810 from my Zigbee network and Domoticz.  Changed battery to a 100% full one. re=paired the device to my Zigbee network. This works OK.
No battery value visible. Update failed on this device again. See log
`Debug Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"AB-01","transaction":"y3nv2-1"}'
Info Checking if update available for 'AB-01'
Debug Check if update available for '0x842e14fffe0d9641' (TRADFRI remote control)
Debug Using endpoint '1'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"AB-01"},"error":"Failed to check if update available for 'AB-01' (Device didn't respond to OTA request)","status":"error","transaction":"y3nv2-1"}'
Error Failed to check if update available for 'AB-01' (Device didn't respond to OTA request)
Debug Error: Device didn't respond to OTA request at requestOTA (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:196:15)
Debug Saving state to file /opt/zigbee2mqtt/data/state.json
`

markiceman, 2022-01-04:
I have similar problem with E1812 button. Battery is 100%. I send request for update, i see in log Updating nad thats all. Upgrade does not start.

If I try repeat it, then i see in log "Update or check for update already in progress", but it is not true.

Restart zigbee2mqtt have no effect.

AF360, 2022-01-11:
Same issue with an E1812 here. :-(

Koenkk, 2022-01-11:
This is the same issue as #10666 , currently working on a fix there.

Note that this is a bug on the z2m side, if the progress is 100 the device updated succesfully.
Current workaround:
- stop z2m
- remove the `update` state in `data/state.json`
- start z2m
- If your button doesn't send any events anymore, click the yellow reconfigure button in the frontend, just before that make sure to wakeup the device by pressing a button on it.

closed 2022-01-11 by Koenkk

graphicsxp, 2022-01-30:
I have a similar issue with the Tradfri Remote (5 buttons).

I am on firmware 2.3.014. I clicked Check for new updates button. The new firmware was found and it started installation. After almost 3 hours (!) it reached the 100% and displayed "26s remaining". I waited one more hour and had enough so I restarted zigbee2mqtt. Now the firmware is still on 2.3.014 and if I click on "Check for new updates" button I get an error :  
`Error Failed to check if update available for '0x5c0272fffe3d4e8b' (Device didn't respond to OTA request)]`

I deleted the device, removed the battery, repaired it. But when I click on the 'Check for new updatess' button I still get the error. I also tried to manually delete the update property of the device in state.json with no luck. 
What else can I try ?

DesertDwellerEv, 2022-04-17:
I know it's probably too late since the fix is in dev right now and this issue has been closed.
However, I stumbled on how to update E1812 without issue.

1. Press switch in back 4 times to get into pairing mode
2. Watch the logs for device_announce
3. Wait for glowing red led to stop
4. OTA tab and press Update device firmware
5. Rinse and repeat for other devices

exetico, 2022-12-21:
@Koenkk I ended in the update-hang too, after making to maintenance work and forgot all about the fact that a hue-lamp was in the OTA update progress.

I changed `state.json` after I stopped the zigbee2MQTT container.  Started the container again, and everything worked as expected.

A small `Yes, force back to normal "idle" update-state` could be useful .

boelle, 2023-03-05:
i also have the problem i think... i have about 49 ikea units and 17 of those are remotes

the bulb's that are updating do not move and have not done for a day... how do i "stop" it so i can start over?