Koenkk/zigbee2mqtt #10355: [Problem]: toggle does not work correctly on Philips HUE (8719514328242)

metlinux, 2021-12-25:
### What happened?

When I use the "set state toggle" command after I have used "set state off", my Philips Hue light does not turn on with the expected brightness (it is harldy showing any light)

When I use the "set state toggle" command after I have used "set state on" my Philips Hue light works as expected with the correct brightness. Even if I continue using the toggle command from there on the brightness remains correct when the state is toggled to on  !

But when I use "set state off" again and than use "set state toggle", the brightness is wrong again, even though the debug information show the correct brightness, just like when I check it with "get state". It states it is on brightness 163, but the light is hardly noticable and surely not 163

When i just use "set state on" the brightness is as expected

I am using Philips Hue model 8719514328242 with firmware version 1.88.2

### What did you expect to happen?

I expect when using "set state toggle" that the visible brightness remains the same, wether I use toggle from an off or on sitution. 
Just using toggle should not alter the visible brightness, but just exhange the state (from either on to off or off to on)
With visible I mean what I see as a human, not what is stated by mqtt

### How to reproduce it (minimal and precise)

start situation: Light is ON with brightness 163
- send message to zigbee2mqtt/ZITKAMER/set' with data '{"state":"OFF"}'
light turns off
- send message to zigbee2mqtt/ZITKAMER/set' with data '{"state":"TOGGLE"}'
light turns on, but does not have a brightness of 163 (for the human eye noticable because there is hardly any brighness)

extra while light is still on, after using TOGGLE
- send message to zigbee2mqtt/ZITKAMER/get' with data '{"state":""}'
It returns that the brightness is 163 even though it is not

### Zigbee2MQTT version

1.22.1 commit: c2b5229

### Adapter firmware version

20210708

### Adapter

Slaesh's CC2652RB stick (zStack3x0)

### Debug log

debug 2021-12-25 14:26:03: Received MQTT message on 'zigbee2mqtt/ZITKAMER/set' with data '{"state":"ON"}'
debug 2021-12-25 14:26:03: Publishing 'set' 'state' to 'ZITKAMER'
info  2021-12-25 14:26:03: MQTT publish: topic 'zigbee2mqtt/ZITKAMER', payload '{"brightness":163,"color_mode":"color_temp","color_temp":453,"color_temp_startup":453,"linkquality":99,"state":"ON","update":{"state":"idle"}}'
debug 2021-12-25 14:26:08: Received MQTT message on 'zigbee2mqtt/ZITKAMER/set' with data '{"state":"OFF"}'
debug 2021-12-25 14:26:08: Publishing 'set' 'state' to 'ZITKAMER'
info  2021-12-25 14:26:08: MQTT publish: topic 'zigbee2mqtt/ZITKAMER', payload '{"brightness":163,"color_mode":"color_temp","color_temp":453,"color_temp_startup":453,"linkquality":99,"state":"OFF","update":{"state":"idle"}}'
debug 2021-12-25 14:26:15: Received MQTT message on 'zigbee2mqtt/ZITKAMER/set' with data '{"state":"TOGGLE"}'
debug 2021-12-25 14:26:15: Publishing 'set' 'state' to 'ZITKAMER'
info  2021-12-25 14:26:16: MQTT publish: topic 'zigbee2mqtt/ZITKAMER', payload '{"brightness":163,"color_mode":"color_temp","color_temp":453,"color_temp_startup":453,"linkquality":99,"state":"ON","update":{"state":"idle"}}'
debug 2021-12-25 14:26:24: Received MQTT message on 'zigbee2mqtt/ZITKAMER/get' with data '{"state":""}'
debug 2021-12-25 14:26:24: Publishing get 'get' 'state' to 'ZITKAMER'
info  2021-12-25 14:26:24: MQTT publish: topic 'zigbee2mqtt/ZITKAMER', payload '{"brightness":163,"color_mode":"color_temp","color_temp":453,"color_temp_startup":453,"linkquality":99,"state":"ON","update":{"state":"idle"}}'

Koenkk, 2021-12-25:
Could you share your `configuration.yaml`? I'm interested wether you have a `transition` defined for this light.

metlinux, 2021-12-25:
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mosquitto:1883
  client_id: zigbee2mqtt
  keepalive: 60
  reject_unauthorized: true
  version: 4
  include_device_information: false
serial:
  port: /dev/ttyACM0
  adapter: auto
  disable_led: true
frontend:
  port: 8080
advanced:
  network_key:
<CENSORED>
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  ikea_ota_use_test_url: false
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  homeassistant_legacy_triggers: true
  cache_state: true
  cache_state_persistent: true
  cache_state_send_on_startup: true
  log_level: debug
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500
permit_join: false
device_options:
  legacy: false
devices:
  '0x00<CENSORED>':
    friendly_name: ZBMINI
  '0x00<CENSORED>':
    friendly_name: ZITKAMER
    transition: 2
    legacy: false
    optimistic: true
  '0x00<CENSORED>':
    friendly_name: EETKAMER
    transition: 2
  '0x00<CENSORED>':
    friendly_name: OPKAST
    transition: 2
  '0x00<CENSORED>':
    friendly_name: BANK
    transition: 2
groups: {}

I tried with disabling optimistic, that's why it is explicitly mentioned now.
And indeed I have transition set to 2

metlinux, 2021-12-25:
I tried with transition set to 0 (no transition) but the same result. Toggle alters the visible brightness when used after "set state off"

Koenkk, 2021-12-26:
Can you try without any transition specified (remove it from your `configuration.yaml`?)

metlinux, 2021-12-26:
I removed transition for all my devices and restarted zigbee2mqtt. That had a surprising result. When I now used "set state on" I did not get the brightness (visible) I expected and did get before. The result was now like I had with using "set state toggle". I verified in the frontend dashboard the setting for it, but brightness there still showed 163 as I had it set. Only when I moved the slider I got the expected brightness. So after changing this I have set it back to 163 and now toggle does seems to work as expected (at least for now ;-))

Does this make sense to you ?

Koenkk, 2021-12-27:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

metlinux, 2021-12-28:
I tried this version, but it seems to make it worse and I expect this to impact more ppl now.

What happens now:
If I use transition (e.g. 4) I can turn on the light correctly (when it is off as well), but I am not able to turn the light off anymore.
Even though the front-end also sees it is as a valid change and put the light to off in the portal, in reality the light remains on.
Without transition (even when I set it to 0 now) it works as expected and I can turn on and off the light (in reality as well)

I won't mind if you reverse this change and keep it as it were. Knowing I had to remove transition is/was OK for me.

metlinux, 2021-12-29:
I tried once more, but the update seems to make it worse and I expect this to impact more ppl now.

What happens after this update:
If I use transition (e.g. 4) I can turn on the light correctly (when it is off as well), but I am not able to turn the light off anymore.
Even though the front-end also sees it is as a valid change and put the light to off in the portal, in reality the light remains on.
So the frontend see the light turned off, while in realitu it remains on

Without transition (only if I delete it, zero is not enough) it works as expected and I can turn on and off the light (in reality as well)

I won't mind if you reverse this change and keep it as it were. Knowing I had to remove transition is/was OK for me.
But if you want me to do more tests, I am happy to assist

Koenkk, 2021-12-29:
I cannot reproduce this with my hue light, can you provide the herdsman debug logging of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

metlinux, 2021-12-30:
I don't have it with my other Hue lights either (model 9290022944 or 9290024406). Those seems to just ignore the transition setting completely. I only have this with the type as mentioned in the subject.

I have made the debug, but notice my networkkey, panid etc is in there. In what safe way can I share this with you without publishing it here....

Koenkk, 2021-12-30:
You can send it to me on telegram (@koenkk)

metlinux, 2021-12-30:
I don't have Telegram, sorry. But I think this info will be sufficient. Let me know when you have it so I can remove it again:

`Zigbee2MQTT:info  2021-12-30 09:49:24: Connected to MQTT server
Zigbee2MQTT:info  2021-12-30 09:49:24: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2021-12-30 09:49:24: MQTT publish: topic 'zigbee2mqtt/ZBMINI/availability', payload 'online'
Zigbee2MQTT:info  2021-12-30 09:49:24: MQTT publish: topic 'zigbee2mqtt/ZITKAMER/availability', payload 'online'
Zigbee2MQTT:info  2021-12-30 09:49:24: MQTT publish: topic 'zigbee2mqtt/EETKAMER/availability', payload 'online'
Zigbee2MQTT:info  2021-12-30 09:49:24: MQTT publish: topic 'zigbee2mqtt/OPKAST/availability', payload 'online'
Zigbee2MQTT:info  2021-12-30 09:49:24: MQTT publish: topic 'zigbee2mqtt/BANK/availability', payload 'online'
Zigbee2MQTT:info  2021-12-30 09:49:24: MQTT publish: topic 'zigbee2mqtt/REMOTE/availability', payload 'online'
Zigbee2MQTT:info  2021-12-30 09:49:25: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2021-12-30 09:49:25: MQTT publish: topic 'zigbee2mqtt/ZBMINI', payload '{"state":"OFF"}'
Zigbee2MQTT:info  2021-12-30 09:49:25: MQTT publish: topic 'zigbee2mqtt/kamer', payload '{"brightness":163,"color_mode":"color_temp","color_temp":453,"state":"OFF"}'
Zigbee2MQTT:info  2021-12-30 09:49:25: MQTT publish: topic 'zigbee2mqtt/ZITKAMER', payload '{"brightness":163,"color_mode":"color_temp","color_temp":453,"color_temp_startup":453,"state":"OFF","update":{"state":"idle"}}'
Zigbee2MQTT:info  2021-12-30 09:49:25: MQTT publish: topic 'zigbee2mqtt/kamer', payload '{"brightness":52,"color_mode":"color_temp","color_temp":453,"state":"OFF"}'
Zigbee2MQTT:info  2021-12-30 09:49:25: MQTT publish: topic 'zigbee2mqtt/EETKAMER', payload '{"brightness":52,"color_mode":"color_temp","color_temp":453,"state":"OFF","update":{"state":"idle"}}'
Zigbee2MQTT:info  2021-12-30 09:49:25: MQTT publish: topic 'zigbee2mqtt/kamer', payload '{"brightness":163,"color_mode":"color_temp","color_temp":453,"state":"OFF"}'
Zigbee2MQTT:info  2021-12-30 09:49:25: MQTT publish: topic 'zigbee2mqtt/OPKAST', payload '{"brightness":163,"color_mode":"color_temp","color_temp":453,"color_temp_startup":453,"state":"OFF","update":{"state":"idle"}}'
Zigbee2MQTT:info  2021-12-30 09:49:25: MQTT publish: topic 'zigbee2mqtt/BANK', payload '{"brightness":163,"color_mode":"color_temp","color_temp":453,"color_temp_startup":453,"state":"OFF","update":{"state":"idle"}}'
Zigbee2MQTT:info  2021-12-30 09:49:25: MQTT publish: topic 'zigbee2mqtt/REMOTE', payload '{"battery":96}'
2021-12-30T08:49:48.806Z zigbee-herdsman:controller:endpoint Command 0x001788010b9cb0ac/11 genLevelCtrl.moveToLevelWithOnOff({"level":163,"transtime":60}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-12-30T08:49:48.809Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x001788010b9cb0ac:91/11 (0,0,1)
2021-12-30T08:49:48.811Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":91,"destendpoint":11,"srcendpoint":1,"clusterid":8,"transid":2,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,3,4,163,60,0]}}
2021-12-30T08:49:48.812Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,16,36,1,91,0,11,1,8,0,2,0,30,6,1,3,4,163,60,0,239]
2021-12-30T08:49:48.826Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-12-30T08:49:48.826Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-12-30T08:49:48.827Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-12-30T08:49:48.827Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-12-30T08:49:48.828Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-30T08:49:48.829Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,2,196]
2021-12-30T08:49:48.830Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,2,196]
2021-12-30T08:49:48.830Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,2] - 196
2021-12-30T08:49:48.830Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":2}
2021-12-30T08:49:48.831Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-30T08:49:48.845Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,91,0,0,217]
2021-12-30T08:49:48.846Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,91,0,0,217]
2021-12-30T08:49:48.846Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [91,0,0] - 217
2021-12-30T08:49:48.847Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":91,"relaycount":0,"relaylist":[]}
2021-12-30T08:49:48.847Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-30T08:49:48.892Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,8,0,91,0,11,1,0,90,0,12,145,32,0,0,5,24,3,11,4,0,91,0,29,53]
2021-12-30T08:49:48.892Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,8,0,91,0,11,1,0,90,0,12,145,32,0,0,5,24,3,11,4,0,91,0,29,53]
[... cut ...] point":1}'
2021-12-30T08:50:03.780Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2021-12-30 09:50:03: MQTT publish: topic 'zigbee2mqtt/kamer', payload '{"brightness":163,"color_mode":"color_temp","color_temp":453,"state":"OFF"}'
Zigbee2MQTT:info  2021-12-30 09:50:03: MQTT publish: topic 'zigbee2mqtt/ZITKAMER', payload '{"brightness":163,"color_mode":"color_temp","color_temp":453,"color_temp_startup":453,"linkquality":90,"state":"OFF","update":{"state":"idle"}}'
`

Koenkk, 2021-12-30:
Based on the OP I would expect to see a toggle in this log. Can you do the following:
- start z2m, wait 10 seconds till logging is settled (to prevent noise in the log)
- start recording logs
- send `zigbee2mqtt/ZITKAMER/set` `{"state":"ON","brightness":163}`
- send `zigbee2mqtt/ZITKAMER/set` `{"state":"OFF"}`
- send `zigbee2mqtt/ZITKAMER/set` `{"state":"TOGGLE"}` (if I understand correctly the issue should appear now)
- provide that logging here.

metlinux, 2021-12-30:
Your understanding was correct for the initial problem. Intitially after those 3 commands the brightness was much less as expected if there was a value set for transition. There was hardly any brightness (not the expected 163)

After your proposed "fix" this changed, and all those 3 commands work flawlessly (even with a transition set), but if after that last command (when the light is on) you send another {"state":"TOGGLE"} (as 4th command) you would expect the light to go OFF again. But that does not happen (although the frontend will show the light as set off, in reality the light remains on, on the correct brightness)

So I have tested this in that "latest-dev" version. See the attached file for details

[log-4commands.txt](https://github.com/Koenkk/zigbee2mqtt/files/7793147/log-4commands.txt)

Koenkk, 2021-12-30:
Found the issue, made a typo. 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

metlinux, 2021-12-30:
I just tested it and now it seems to work perfectly !!! Great !

Koenkk, 2021-12-30:
great!

closed 2021-12-30 by Koenkk