Koenkk/zigbee2mqtt #10358: [New device support]:Tuya Smart Life ZigBee Garage Door Opener Controller Timer App Monitor Security Remote Control Supports Alexa Google Home

amgstone, 2021-12-25:
### Link

https://aliexpress.ru/item/1005003610757320.html?srcSns=sns_Copy&spreadType=socialShare&bizType=ProductDetail&social_params=20428159193&aff_fcid=92a2b9469b184b528e2c9427d5b02d78-1640457011599-04429-_mPK6Sle&tt=MG&aff_fsk=_mPK6Sle&aff_platform=default&sk=_mPK6Sle&aff_trace_key=92a2b9469b184b528e2c9427d5b02d78-1640457011599-04429-_mPK6Sle&shareId=20428159193&businessType=ProductDetail&platform=AE&terminal_id=8b579fd2e35b4783888812bca7b93a72

### Database entry

{"id":13,"type":"Router","ieeeAddr":"0xa4c13835f76fbfd7","nwkAddr":1671,"manufId":4417,"manufName":"_TZE200_nklqjk62","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0013,Z)i","65506":54,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_nklqjk62","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b00228133c8","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1640453155394,"useImplicitCheckin":true}

### Comments

просьба подключить

dmertsalov, 2022-01-11:
I have the same device and I have no idea how to connect it via an external converter. Please add support for this device.

Toma82, 2022-01-12:
Hi, I have the same problem, here is the extract from my database.db

{"id":29,"type":"Router","ieeeAddr":"0xa4c1387f74","nwkAddr":62627,"manufId":4417,"manufName":"_TZE200_nklqjk62","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\n\u0000\u0000\u0000i\u000b\u0000\u0000\u0000i\f\u0000\u0000\u0000i\f\u0000\u0000\u0000i\r\u0000\u0000\u0000i\r\u0000\u0000\u0000i\u001d[O)\u0013D\u0000\u0000\u0000e","65506":54,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_nklqjk62","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1641047739043,"useImplicitCheckin":true}

Scradam, 2022-01-18:
I have a very similar device. LoraTap zigbee garage door opener. The door sensor is RF rather than wired.

```
{"id":30,"type":"Router","ieeeAddr":"0xa4c138c835f55266","nwkAddr":1151,"manufId":4417,"manufName":"_TZE200_wfxuhoea","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u000f\u0000\u0000\u0000e\u0010\u0000\u0000\u0000\u0012\u001e\u0000\u0000\u0000e\u001f\u0000\u0000\u0000\u0012","65506":54,"65508":0,"appVersion":70,"modelId":"TS0601","manufacturerName":"_TZE200_wfxuhoea","powerSource":1,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1642461110340,"useImplicitCheckin":true}
```

glenoc, 2022-01-20:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: [
            'owvfni3\u0000', 'owvfni3', 'u1rkty3', 'aabybja', // Curtain motors
            'mcdj3aq', 'mcdj3aq\u0000', // Tubular motors
        ],
        fingerprint: [
            // Garage door controller:
            {modelID: 'TS0601', manufacturerName: '_TZE200_nklqjk62'},
        ],
        model: 'TS0601_cover',
        vendor: 'TuYa',
        description: 'Garage door controller',
        fromZigbee: [fz.tuya_cover, fz.ignore_basic_report],
        toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options],
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            exposes.composite('options', 'options')
                .withFeature(exposes.numeric('motor_speed', ea.STATE_SET)
                    .withValueMin(0)
                    .withValueMax(255)
                    .withDescription('Motor speed'))],
}

module.exports = definition;
```
I just tried write a custom device based on the tuya curtain motor. The open closed state is working, but i cannot send any command to open/close.

Koenkk, 2022-01-21:
Added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/3714

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-21 by Koenkk

thezjo, 2022-01-22:
I tried this morning after updating to dev branch, it reports an error in de log view

Info Device '0xa4c1381c4debf559' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381c4debf559","ieee_address":"0xa4c1381c4debf559"},"type":"device_joined"}'
Info Configuring '0xa4c1381c4debf559'
Info Starting interview of '0xa4c1381c4debf559'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381c4debf559","ieee_address":"0xa4c1381c4debf559","status":"started"},"type":"device_interview"}'
Debug Device '0xa4c1381c4debf559' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381c4debf559","ieee_address":"0xa4c1381c4debf559"},"type":"device_announce"}'
Debug Received Zigbee message from '0xa4c1381c4debf559', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0xa4c1381c4debf559', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0xa4c1381c4debf559', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Info Successfully interviewed '0xa4c1381c4debf559', device has successfully been paired
Info Device '0xa4c1381c4debf559' is supported, identified as: Tuya Garage door opener (TS0601)
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Garage door opener","exposes":[{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0601","options":[],"supports_ota":false,"vendor":"Tuya"},"friendly_name":"0xa4c1381c4debf559","ieee_address":"0xa4c1381c4debf559","status":"successful","supported":true},"type":"device_interview"}'
Debug Received Zigbee message from '0xa4c1381c4debf559', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":12}],"seq":15616}' from endpoint 1 with groupID 0
Info zigbee-herdsman-converters:tuya_data_point_dump: Received DP #12 from 0xa4c1381c4debf559 with raw data '{"dp":12,"datatype":4,"data":{"type":"Buffer","data":[2]}}': type='commandDataReport', datatype='enum', value='2', known DP# usage: ["haozeeChildLock","AM02Fault","garageDoorStatus"]
Debug Received Zigbee message from '0xa4c1381c4debf559', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,10],"type":"Buffer"},"datatype":2,"dp":4}],"seq":15872}' from endpoint 1 with groupID 0
Info zigbee-herdsman-converters:tuya_data_point_dump: Received DP #4 from 0xa4c1381c4debf559 with raw data '{"dp":4,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,10]}}': type='commandDataReport', datatype='value', value='10', known DP# usage: ["mode","moesSboostHeating","silvercrestSetColorTemp","haozeeBoostHeating","nousBattery","wlsBatteryPercentage","evanellHeatingSetpoint","AM02Mode"]
Debug Received Zigbee message from '0xa4c1381c4debf559', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":2}],"seq":16128}' from endpoint 1 with groupID 0
Info zigbee-herdsman-converters:tuya_data_point_dump: Received DP #2 from 0xa4c1381c4debf559 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}': type='commandDataReport', datatype='value', value='0', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","haozeeHeatingSetpoint","nousHumidity","tIlluminanceLux","evanellMode","AM02PercentControl"]
Debug Received Zigbee message from '0xa4c1381c4debf559', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,14,16],"type":"Buffer"},"datatype":2,"dp":5}],"seq":16384}' from endpoint 1 with groupID 0
Info zigbee-herdsman-converters:tuya_data_point_dump: Received DP #5 from 0xa4c1381c4debf559 with raw data '{"dp":5,"datatype":2,"data":{"type":"Buffer","data":[0,0,14,16]}}': type='commandDataReport', datatype='value', value='3600', known DP# usage: ["fanMode","motorDirection","config","moesSboostHeatingCountdown","neoAOVolume","silvercrestSetColor","lidlTimer","haozeeBoostHeatingCountdown","evanellLocalTemp","AM02ControlBackMode"]
Debug Received Zigbee message from '0xa4c1381c4debf559', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":11}],"seq":16640}' from endpoint 1 with groupID 0
Info zigbee-herdsman-converters:tuya_data_point_dump: Received DP #11 from 0xa4c1381c4debf559 with raw data '{"dp":11,"datatype":1,"data":{"type":"Buffer","data":[0]}}': type='commandDataReport', datatype='bool', value='false', known DP# usage: ["nousMinTemp","AM02SituationSet"]
Debug Received Zigbee message from '0xa4c1381c4debf559', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":3}],"seq":16896}' from endpoint 1 with groupID 0
Info zigbee-herdsman-converters:tuya_data_point_dump: Received DP #3 from 0xa4c1381c4debf559 with raw data '{"dp":3,"datatype":1,"data":{"type":"Buffer","data":[1]}}': type='commandDataReport', datatype= [... cut ...]
Debug Received Zigbee message from '0xa4c1381c4debf559', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Error Failed to configure '0xa4c1381c4debf559', attempt 2 (Error: Bind 0xa4c1381c4debf559/1 genBasic from '0x00124b0022a539a9/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:557:17) at processTimers (node:internal/timers:500:7))

pakkie1985, 2022-02-08:
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = [...]

pakkie1985, 2022-02-08:
I can only open the door. Closing is not support. Do i something wrong?

Cossey, 2022-02-10:
Does z2m also support the garage door sensor for this device? Looking at picking one up myself.

Is everyone also having troubles with closing the door, like pakkie?

Cheers.

dmertsalov, 2022-02-10:
> Does z2m also support the garage door sensor for this device? Looking at picking one up myself.
> 
> Is everyone also having troubles with closing the door, like pakkie?
> 
> Cheers.

Everything works fine. I use Node-RED for the creation logic for HomeKit Garage Door Opener.

st4ro, 2022-02-10:
> Does z2m also support the garage door sensor for this device? Looking at picking one up myself.
> 
> Is everyone also having troubles with closing the door, like pakkie?
> 
> Cheers.

i played with my device - when door sensor state is TRUE it can be triggered to open the door. When sensor state is FALSE it will not trigger it again so door remains open... 

if anyone managed to close the door while sensor state is FALSE please let us know

Regards

st4ro, 2022-02-10:
> 

> > > Does z2m also support the garage door sensor for this device? Looking at picking one up myself.
> > > Is everyone also having troubles with closing the door, like pakkie?
> > > Cheers.
> > 
> > 
> > i played with my device - when door sensor state is TRUE it can be triggered to open the door. When sensor state is FALSE it will not trigger it again so door remains open...
> > if anyone managed to close the door while sensor state is FALSE please let us know
> > Regards
> 
> What device do you have? If the same as the beginning of this issue, then maybe it is due to a faulty device? I bought the device from the link below. The gate opens/closes regardless of the sensor status. I use the sensor only for logic in HomeKit, flow for Node-RED is attached below. [https://aliexpress.ru/item/1005003610757320.html?spm=a2g2w.productlist.0.0.786a7bf3UUefIS&sku_id=12000026494616761](url)
> 
> All additional checks for matching statuses were only in order not to spam the sensor status in the HomeKit node every 5-10 seconds. [flows.json.zip](https://github.com/Koenkk/zigbee2mqtt/files/8041293/flows.json.zip)

Hi @dmertsalov 

device identification: _TZE200_nklqjk62
device model: PJ-ZGD01

Don't know if it's broken device or standard behavior... I have no logic associated with sensor status (yet). My garage door only needs to "connect the circuit" (I connected the sensor in parallel to the wall switch) - it works when the door is closed. in the opposite direction the circuit closes only when the sensor state is TRUE  (I added another magnet to the hall probe).. 

Regards
st4ro

pakkie1985, 2022-02-10:
It worked with the original Tuya hub and app. Is gives the state of the door en I can open en close it. So i don’t know why it stays open when I use it with zigbee2qmtt

dmertsalov, 2022-02-10:
> It worked with the original Tuya hub and app. Is gives the state of the door en I can open en close it. So i don’t know why it stays open when I use it with zigbee2qmtt

Oddly enough, this also works for me in Tuya hub and in z2m. Perhaps there are some differences in the firmware. But at least a few other people are also working fine. I wouldn't be surprised if there is a difference in the hardware of the device.

st4ro, 2022-02-10:
> It worked with the original Tuya hub and app. Is gives the state of the door en I can open en close it. So i don’t know why it stays open when I use it with zigbee2qmtt

do you still have tuya bridge? Maybe there is a different DP on the HW / board revision ..?

there are information how to obtain DP, maybee it is only SW problem: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

st4ro, 2022-02-10:
> Is it possible that the door closing sensor is connected to the wrong connectors? Could it be that the gate drive does not have the open/close mode turned on without an intermediate stop? I had a wi-fi device there before buying a zigbee device, the connection scheme was exactly the same. Maybe you have a problem with connecting the wires? It seems that this device does not have the ability to connect multiple closing sensors.

well, the wiring is fine ... The condition of the garage door is reported correctly. One direction si working fine.. 
the secondary hall probe can be "quick hack", the fastest was a magnet ..: D

pakkie1985, 2022-02-13:
> do you still have tuya bridge? Maybe there is a different DP on the HW / board revision ..?
>
> there are information how to obtain DP, maybee it is only SW problem: [...]

I still have the hub. But id dont now how to trouble shoot it. It worked now via the tuya app. But i prefer do it through the zigbee2qmtt

kabakakao, 2022-02-16:
I have the same issue. It only triggers, when the reed contact is closed. So I first of all bridged the contact, that the trigger is working. But it would be cool to have the door status ;)
I do not have the Tuya Hub to troubleshoot

Pakum11, 2022-02-17:
i have same problem. when contact is closed, door opening. if contact is open door not close.

st4ro, 2022-02-17:
strange, there must be difference between device owned by dmertsalov  and mine. 

is there any possibility to check FW version on this device ?

H3buss, 2022-02-19:
I'm having the same issue, the device won't activate the contactor command if the closed status is false for some reason...

H3buss, 2022-02-19:
I've fixed it on my side (but had to remove the 'trigger' action), see https://github.com/Koenkk/zigbee-herdsman-converters/pull/3897

pakkie1985, 2022-02-19:
> I've fixed it on my side (but had to remove the 'trigger' action), see [Koenkk/zigbee-herdsman-converters#3897](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3897)

Can you please tell me how I can change it? I’m still learning and don’t know much about zigbee2qmtt enough.

H3buss, 2022-02-19:
You can either wait for it to join the stable branch or switch to dev branch once it is merged as explained here: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

pakkie1985, 2022-02-19:
> You can either wait for it to join the stable branch or switch to dev branch once it is merged as explained here: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

I have already the edge version. So I have to wait for an update? Of is that something happened automatic. Sorry for my questions

H3buss, 2022-02-19:
You need to update manually https://www.zigbee2mqtt.io/guide/installation/01_linux.html#for-later-update-zigbee2mqtt-to-the-latest-version

My pull request has been merged, but I don't know if it is available for npm yet.

dmertsalov, 2022-02-19:
> strange, there must be difference between device owned by dmertsalov and mine, pakum11's and others
> 
> is there any possibility to check FW version on this device ?

I connected the device to the Usb hub and looked at the firmware versions.
zigbee module: V1.0.6
MCU module: V1.0.0

After the purchase, the firmware was updated only once.

pakkie1985, 2022-02-19:
> You need to update manually https://www.zigbee2mqtt.io/guide/installation/01_linux.html#for-later-update-zigbee2mqtt-to-the-latest-version
> 
> My pull request has been merged, but I don't know if it is available for npm yet.

Again sorry for the questions. But is there any way I can see which version converter I already have installed. I use supervisor ha and installs the edge version. Do I have to delete this and install again and that’s it? I can’t see anything changed

H3buss, 2022-02-19:
I misread and did not see you were running edge. According to [the docs](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/tree/master/zigbee2mqtt-edge#updating-the-edge-add-on), you need to remove then re-install the plugin.

st4ro, 2022-02-20:
> I've fixed it on my side (but had to remove the 'trigger' action), see [Koenkk/zigbee-herdsman-converters#3897](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3897)

I can confirm - your solution Is working , thanks

st4ro, 2022-02-20:
one more little thing (maybe just cosmetics) - the binary "garage_door_contact" will not be classified as "device_class: door" (https://www.home-assistant.io/integrations/binary_sensor/) in HA (MQTT Discovery - https://www.home-assistant.io/docs/mqtt/discovery/ ) 

so I added device_class: garage_door into customize section of HA and changed

`exposes.binary ('garage_door_contact', ea.STATE,  true, false)`

into 

`exposes.binary ('garage_door_contact', ea.STATE, false, true)`

becouse garage_door: **on** means **open**, **off** means **closed**

pakkie1985, 2022-02-21:
> one more little thing (maybe just cosmetics) - the binary "garage_door_contact" will not be classified as "device_class: door" (https://www.home-assistant.io/integrations/binary_sensor/) in HA (MQTT [...]

I can confirm ist working now indeed. I changed the device class in HA but can someone tell me how to change the exposes.binary  ? Im almost there where i wanted, but nog the doorcontact said open when its closes and closed when its open.

st4ro, 2022-02-22:
There should be a better way for class_class: garage_door - instead of using customizations in HomeAssistant I would like to use MQTT discovery from Z2M.

I found that _lib/extension/homeassistant.ts_ should be enriched as follows:
`garage_door_contact: {device_class: 'garage_door'}`

but it doesn't work (I restarted Z2M and HASS) - maybe I missed something

inverted meaning of door status should be done in _tuya.js_

pakkie1985, 2022-02-22:
Sorry for asking but change Tuya.js where can I find that file.

pakkie1985, 2022-02-24:
Can anybody please tell me how to change the state of the door. The door perfectly closed now and perfectly open but the door contact tells me something else Is there a possibility that it will fixed in the next rellease . Or how can i change it.

st4ro, 2022-03-01:
check https://github.com/st4ro/zigbee-herdsman-converters/commit/1fbfd3c822a9b475b968dd682adf7015ec098c6b